JeffyCN/meta-rockchip #3: conf  能否增加支持  px30 evb 32bit

hanqingwu, 2019-10-26:
conf  能否增加支持  px30 evb 32bit

hanqingwu, 2019-10-27:
我试下个 rockchip-rk3326-evb-32.conf  ，编译出错了。
| qc: no such file or directory
| [391/565] Linking CXX static library lib/linux/libclang_rt.stats_client-armhf.a
| FAILED: lib/linux/libclang_rt.stats_client-armhf.a
| : && /home/wuhq/work2/chromium-32/build/tmp/work/armv7vet2hf-neon-pokymllib32-linux-gnueabi/lib32-compiler-rt/7.1.0-r0/recipe-sysroot-native/usr/bin/cmake -E remove lib/linux/libclang_rt.stats_client-armhf.a && /home/wuhq/work2/chromium-32/build/tmp/work/armv7vet2hf-neon-pokymllib32-linux-gnueabi/lib32-compiler-rt/7.1.0-r0/recipe-sysroot-native/usr/bin/arm-pokymllib32-linux-gnueabi/arm-pokymllib32-linux-gnueabi-llvm-ar qc lib/linux/libclang_rt.stats_client-armhf.a  lib/stats/CMakeFiles/clang_rt.stats_client-armhf.dir/stats_client.cc.o && /home/wuhq/work2/chromium-32/build/tmp/work/armv7vet2hf-neon-pokymllib32-linux-gnueabi/lib32-compiler-rt/7.1.0-r0/recipe-sysroot-native/usr/bin/arm-pokymllib32-linux-gnueabi/arm-pokymllib32-linux-gnueabi-llvm-ranlib lib/linux/libclang_rt.stats_client-armhf.a && :
| qc: no such file or directory
| [392/565] Building CXX object lib/lsan/CMakeFiles/RTLSanCommon.armhf.dir/lsan_common.cc.o
| ninja: build stopped: subcommand failed.
| WARNING: exit code 1 from a shell command.

hanqingwu, 2019-10-27:
build/tmp/work/armv7vet2hf-neon-pokymllib32-linux-gnueabi/lib32-compiler-rt/7.1.0-r0/recipe-sysroot-native/usr/bin/arm-pokymllib32-linux-gnueabi/arm-pokymllib32-linux-gnueabi-llvm-ar   这个不支持 qc 参数。

JeffyCN, 2019-11-21:
> conf 能否增加支持 px30 evb 32bit

I think there are 3 ways to build a 32bit rootfs for px30 evb:
1/ Create a machine conf based on meta-rockchip/conf/machine/rockchip-rk3326-evb-32.conf
or
2/ Do in in the yocto's standard way:
https://www.yoctoproject.org/docs/current/dev-manual/dev-manual.html#combining-multiple-versions-library-files-into-one-image
And "bitbake lib32-core-image-minimal"
or
3/ Use other 32bit machine conf to build rootfs.img(for example 3326-32/3288/px3se), and use it with the px30's kernel and loader(u-boot)
(Might need to replace mali/alsa conf/wifi bt firmware in the rootfs)

For compiling issues, please try:
https://github.com/JeffyCN/yocto-manifests/tree/release-1.1.1_20191030_thud

JeffyCN, 2019-11-22:
Hmm, i can repro the llvm-ar issue, please check https://github.com/kraj/meta-clang/issues/127 for more details.

So it would only repro in multilib case when the meta-clang version newer than thud

hanqingwu, 2019-11-22:
thanks reply.

JeffyCN, 2019-12-02:
closing this, since we are going to trace it in meta-clang issues.

closed 2019-12-02 by JeffyCN