JeffyCN/meta-rockchip #7: u-boot failed to build

firexx, 2021-03-10:
last messages from the build log
|   aarch64-poky-linux-objcopy --dump-section .rodata.default_environment=u-boot-initial-env env/common.o; sed --in-place -e 's/\x00/\x0A/g' u-boot-initial-env
| make[1]: Leaving directory '/work/ak/orangepi/rk-yocto-bsp/build/tmp/work/orangepi_4-poky-linux/u-boot/1_2017.09+gitAUTOINC+f06413e433-r0/build'
| make: Leaving directory '/work/ak/orangepi/rk-yocto-bsp/build/tmp/work/orangepi_4-poky-linux/u-boot/1_2017.09+gitAUTOINC+f06413e433-r0/git'
| ERROR: No ../rkbin repository
| WARNING: exit code 1 from a shell command.

it looks line the rkbin is not checked out 
I found in the git/make.sh:159 it checks on existin directory of ../rkbin/tools but this path does not exists

JeffyCN, 2021-03-10:
rkbin should be installed from:
https://github.com/JeffyCN/meta-rockchip/blob/master/recipes-bsp/u-boot/u-boot_%25.bbappend#L17

firexx, 2021-03-10:
yes. it was my failt. the old mirror.conf did overwrite the SRC_URI. so only the one link to u-boot branch stay in SRC_URI :-)

JeffyCN, 2021-03-10:
well, the mirror conf should not be needed now, we are using mirror sites by default :)

closed 2021-03-10 by JeffyCN