JeffyCN/meta-rockchip #8: Gitchy graphics with xserver

zlemonpie, 2021-06-30:
Hi @JeffyCN, I have a custom board that uses rk3399pro (rockpi-n10) and I'm glueing a yocto build that uses parts of meta-rockchip (this repo) and meta-radxa. Without these graphics recipes, xserver displays fine, but rendering 2d graphics is extremely slow so I thought of baking in the graphic recipes from this layer.

Now, I'm having a weird issue with the xserver I was hoping you can help me with:
[image]

X11 logs seems to be complaining about rockchip_dri.so driver missing:
```
root@rockpi-n10-rk3399pro:~# cat /var/log/Xorg.0.log
X.Org X Server 1.20.8
X Protocol Version 11, Revision 0
[     9.751] Build Operating System: Linux 
[     9.751] Current Operating System: Linux rockpi-n10-rk3399pro 4.4.167 #1 SMP Mon Apr 26 13:14:17 UTC 2021 aarch64
[     9.751] Kernel command line: console=tty1 console=ttyS2,1500000n8 rw root=PARTUUID=b921b045-1df0-41c3-af44-4c6f280d3fae rootfstype=ext4 init=/sbin/init rootwait cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
[     9.752] Build Date: 15 March 2021  02:06:25PM
[     9.752]  
[     9.753] Current version of pixman: 0.38.4
[     9.753]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[     9.753] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[     9.754] (==) Log file: "/var/log/Xorg.0.log", Time: Sun Sep 20 10:44:02 2020
[     9.755] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[     9.756] (==) No Layout section.  Using the first Screen section.
[     9.756] (**) |-->Screen "Default Screen" (0)
[     9.756] (**) |   |-->Monitor "Default Monitor"
[     9.757] (**) |   |-->Device "Rockchip Graphics"
[     9.757] (==) Automatically adding devices
[     9.757] (==) Automatically enabling devices
[     9.757] (==) Automatically adding GPU devices
[     9.757] (==) Max clients allowed: 256, resource mask: 0x1fffff
[     9.757] (WW) The directory "/usr/share/fonts/X11/misc/" does not exist.
[     9.757]    Entry deleted from font path.
[     9.757] (WW) The directory "/usr/share/fonts/X11/TTF/" does not exist.
[     9.757]    Entry deleted from font path.
[     9.757] (WW) The directory "/usr/share/fonts/X11/OTF/" does not exist.
[     9.757]    Entry deleted from font path.
[     9.757] (WW) The directory "/usr/share/fonts/X11/Type1/" does not exist.
[     9.757]    Entry deleted from font path.
[     9.757] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[     9.757]    Entry deleted from font path.
[     9.757] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[     9.757]    Entry deleted from font path.
[     9.757] (==) FontPath set to:
[     9.757] (==) ModulePath set to "/usr/lib/xorg/modules"
[     9.757] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[     9.757] (II) Loader magic: 0x55690eecd0
[     9.757] (II) Module ABI versions:
[     9.757]    X.Org ANSI C Emulation: 0.4
[     9.757]    X.Org Video Driver: 24.1
[     9.757]    X.Org XInput driver : 24.1
[     9.757]    X.Org Server Extension : 10.0
[     9.760] (--) using VT number 2
[     9.760] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[     9.762] (II) xfree86: Adding drm device (/dev/dri/card0)
[     9.770] (II) no primary bus or device found
[     9.770]    falling back to /sys/devices/platform/display-subsystem/drm/card0
[     9.770] (II) LoadModule: "glx"
[     9.771] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[     9.778] (II) Module glx: vendor="X.Org Foundation"
[     9.778]    compiled for 1.20.8, module version = 1.0.0
[     9.778]    ABI class: X.Org Server Extension, version 10.0
[     9.778] (II) LoadModule: "modesetting"
[     9.779] (II) Loading /usr/lib/xorg/modules/drivers/modesetting_drv.so
[     9.780] (II) Module modesetting: vendor="X.Org Foundation"
[     9.780]    compiled for 1.20.8, module version = 1.20.8
[     9.780]    Module class: X.Org Video Driver
[     9.780]    ABI class: X.Org Video Driver, version 24.1
[     9.780] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[... 117 lines cut ...]
[    10.126] (II) event0  - rk8xx_pwrkey: is tagged by udev as: Keyboard
[    10.126] (II) event0  - rk8xx_pwrkey: device is a keyboard
[    10.127] (II) event0  - rk8xx_pwrkey: device removed
[    10.148] (**) Option "config_info" "udev:/sys/devices/platform/ff3c0000.i2c/i2c-0/0-0020/input/input0/event0"
[    10.148] (II) XINPUT: Adding extended input device "rk8xx_pwrkey" (type: KEYBOARD, id 7)
[    10.153] (II) event0  - rk8xx_pwrkey: is tagged by udev as: Keyboard
[    10.153] (II) event0  - rk8xx_pwrkey: device is a keyboard
[    10.155] (II) config/udev: Adding input device gpio-keys (/dev/input/event2)
[    10.155] (**) gpio-keys: Applying InputClass "evdev keyboard catchall"
[    10.155] (**) gpio-keys: Applying InputClass "libinput keyboard catchall"
[    10.155] (II) Using input driver 'libinput' for 'gpio-keys'
[    10.155] (**) gpio-keys: always reports core events
[    10.155] (**) Option "Device" "/dev/input/event2"
[    10.155] (**) Option "_source" "server/udev"
[    10.157] (II) event2  - gpio-keys: is tagged by udev as: Keyboard
[    10.157] (II) event2  - gpio-keys: device is a keyboard
[    10.157] (II) event2  - gpio-keys: device removed
[    10.224] (**) Option "config_info" "udev:/sys/devices/platform/gpio-keys/input/input2/event2"
[    10.224] (II) XINPUT: Adding extended input device "gpio-keys" (type: KEYBOARD, id 8)
[    10.230] (II) event2  - gpio-keys: is tagged by udev as: Keyboard
[    10.230] (II) event2  - gpio-keys: device is a keyboard
[    10.267] (WW) modeset(0): flip queue failed: Invalid argument
[    10.267] (WW) modeset(0): Page flip failed: Invalid argument
[    10.588] (II) modeset(0): Disabling kernel dirty updates, not required.
```

JeffyCN, 2021-06-30:
Hi,

By "a weird issue with the xserver", did you mean the "missing rockchip_dri.so" warning?
If so, that is expected, and it will fallback to swrast.

We only provide GBM/EGL/GLESv2/GLESv1/OPENCL with mali, so other implements(e.g. glx) will fallback to mesa(software opengl)

And according to the log, you are using RGA(hw accel module) based exa accel and flipfb(double-buffer rendering):
[     9.852] (**) modeset(0): Option "AccelMethod" "exa"
[     9.852] (**) modeset(0): Option "FlipFB" "always"
[     9.861] (II) modeset(0): Using RGA EXA

There's no other error logs, so guessing it should work.

zlemonpie, 2021-06-30:
Hi Jeffy,
By weird issue, I mean rendering of the graphics onto the screen (like the image I posted). Top half seems to have multiple renderings of the terminal box and the bottom half has the calibration screen. (I was expecting just a single terminal box).

What settings/drivers would you recommend to use on the rk3399pro boards to use for smooth experiences? Like watching youtube

zlemonpie, 2021-06-30:
For reference, this is what I expected (I built this against poky's xserver's recipe)
[image]

JeffyCN, 2021-06-30:
well, that seems like something wrong in the kernel side(rga driver).
maybe try to modify /etc/X11/xorg*/20-mod*.conf, change exa to glamor

JeffyCN, 2021-07-01:
by the way, if possible, please upload your rootfs and kernel image, maybe i can try to debug it.
and which version of this repo did you use?

zlemonpie, 2021-07-01:
thanks for reaching out a helping hand, I appreciate it. 
I added the openbox WM and the problem seem to go away (weird). i think previously it was using matchbox.. so maybe something that with matchbox that it doesn't like.

I also tried using glamor, but rendering of GUI is extremely slow (on xfce deskop).. is this expected?

zlemonpie, 2021-07-01:
Jeffy, I'll go ahead and close this issue. It also turns out that mali drivers weren't loading properly (kernel expecting r14, but r18 was installed)

closed 2021-07-01 by zlemonpie

JeffyCN, 2021-07-02:
> Jeffy, I'll go ahead and close this issue. It also turns out that mali drivers weren't loading properly (kernel expecting r14, but r18 was installed)

So the kernel is old(with out-dated mali driver), and there're lots of fixes about RGA in recent kernel.
The matchbox+lxterminal works well on my side, could related to kernel version.