JeffyCN/meta-rockchip #9: Board 5 volt problem with rk3288 processor, rk808 pmu

TYeniyayla, 2022-02-14:
Hi,

My board with rk808 PMU and rk3288 CPU does not output 5 volts from USBs. I'm sure the problem is due to Device Tree because; There is no such problem when I use the dtb file provided by the manufacturer of the board, whose contents I cannot see. When I use that dtb file, I cannot activate other features. How can I activate the 5v feature on the card?

Thank you in advance for your help. Attached you can see the contents of the dts files that I am currently using
[rk3288-evb-rk808-linux.zip](https://github.com/JeffyCN/meta-rockchip/files/8060182/rk3288-evb-rk808-linux.zip)
.

JeffyCN, 2022-02-14:
maybe try to decode those dtb into dts, and compare.

TYeniyayla, 2022-02-15:
Hi,

I compared the dts file I took from the system I created with Yocto and the dts files I took from the company's os. I examined the differences and changed some places, but I could not get a result. I am sharing the original (unmodified) files in the attachments, is it possible for you to have a look? Poky is your dts file, manufacturer is manufacturer's dts files.

[dts files.zip](https://github.com/JeffyCN/meta-rockchip/files/8067206/dts.files.zip)

JeffyCN, 2022-02-15:
the dts file should match the hardware(for hardware modules/regulators/pin configs, etc.) so the dts in this meta-rockchip is for the specific evb board only...

according to the diff, these might related:
otg-vbus-drv
usb@ff580000's dr_mode and vbus-supply
other regulator settings

anyway, i don't know your hardware, so might not be able to help  :(

JeffyCN, 2022-02-15:
maybe just use the manufacturer's kernel and dtb instead of meta-rockchip's?

TYeniyayla, 2022-02-15:
Let me take a closer look at the parts you mentioned. The manufacturer's dtb works, but when I try to add features such as lvds and touch, I cannot use it because it is dtb. When I convert it to dts and edit it, for some reason it gets stuck on "Starting Kernel..." after uboot. Everything I edited in your dts works fine, but usbs do not give 5v.

JeffyCN, 2022-02-15:
>> When I convert it to dts and edit it, for some reason it gets stuck on "Starting Kernel..."

TYeniyayla, 2022-02-15:
> that might because there're some phandle in that decoded dts, would might be changed after edit.

I checked the Phandle parts, there is no problem. Now I'm trying to change the usb and regulator parts. I tried 3 times, still no results.

JeffyCN, 2022-02-16:
i think there would be 2 ways for it:
1/ use fdtdump/fdtget/fdtput to manipulate the manufacturer.dtb
or
2/ try to port from the decoded manufacturer.dts(using dtc/fdtget) to the new dts(i) 
a) make sure /pinctrl/ matched(mostly otg-vbus-drv and host-vbus-drv)
b) make sure usb nodes's dr_mode and status matched
c) fixup new dts's otg-vbus-regulator from manufacturer.dts's usb-otg-regulator (the manufacturer's dts is using it as a standalone always on regulator)

TYeniyayla, 2022-02-16:
I was getting the dts file with the command: `dtc -I fs /proc/device-tree -O dts -o a.dts`

How can I do it with the method you say? is there any source?

JeffyCN, 2022-02-16:
for the first way, fdtput is a tool to modify  manufacturer.dtb directly, you can google for it's usage. 

for the second way, you can try to understand those fdt nodes in the decoded manufacturer.dts and modify the poky kernel's dts(check the readme.md for using local custom kernel)

TYeniyayla, 2022-02-18:
Hi, I've been working on the problem for the last 2 days. I haven't been able to solve it yet. I checked the places you mentioned. No problem with Phandles. dr_mode and status are the same. I tried fdtput but it was too slow and cumbersome so I went ahead with manual editing. I saved the changes as a patch and created an image from scratch but without success. I don't know what else to do, what could I be missing?

TYeniyayla, 2022-02-28:
Hello, I could not get a result from what they said, but as a result of long talks with the manufacturer, I was able to get the dts file. It's running at 5 volts now, but it gets stuck on "Starting kernel..." at uboot startup. I have attached the dts file I received
[rk3288-lc-linux.zip](https://github.com/JeffyCN/meta-rockchip/files/8153732/rk3288-lc-linux.zip)
.

JeffyCN, 2022-02-28:
maybe you can try to enable earlycon in the dts's bootargs(check other 3288 dts in kernel for how to set it), that should give you a few more early boot logs.

maybe you can try to decode the new dtb and compare with the old decoded manufacturer.dts

maybe there are some difference in the dtsi(which they didn't provide in that zip)

TYeniyayla, 2022-02-28:
Does this help in solving the problem?:
```
arch=arm
autoload=no
baudrate=115200
board=evb_rk3288
board_name=evb_rk3288
boot_a_script=load ${devtype} ${devnum}:${distro_bootpart} ${scriptaddr} ${prefix}${script}; source ${scriptaddr}
boot_extlinux=sysboot ${devtype} ${devnum}:${distro_bootpart} any ${scriptaddr} ${prefix}extlinux/extlinux.conf
boot_net_usb_start=usb start
boot_prefixes=/ /boot/
boot_script_dhcp=boot.scr.uimg
boot_scripts=boot.scr.uimg boot.scr
boot_targets=mmc1 mmc0 usb0 pxe dhcp
bootargs=storagemedia=emmc androidboot.storagemedia=emmc androidboot.mode=normal
bootcmd=boot_android ${devtype} ${devnum};boot_fit;bootrkp;run distro_bootcmd;
bootcmd_dhcp=run boot_net_usb_start; if dhcp ${scriptaddr} ${boot_script_dhcp}; then source ${scriptaddr}; fi;
bootcmd_mmc0=setenv devnum 0; run mmc_boot
bootcmd_mmc1=setenv devnum 1; run mmc_boot
bootcmd_pxe=run boot_net_usb_start; dhcp; if pxe get; then pxe boot; fi
bootcmd_usb0=setenv devnum 0; run usb_boot
bootdelay=0
cpu=armv7
devnum=0
devtype=mmc
distro_bootcmd=for target in ${boot_targets}; do run bootcmd_${target}; done
eth1addr=2e:a5:ef:15:4d:5b
ethaddr=2a:a5:ef:15:4d:5b
fdt_addr_r=0x08300000
kernel_addr_r=0x02008000
mmc_boot=if mmc dev ${devnum}; then setenv devtype mmc; run scan_dev_for_boot_part; fi
partitions=uuid_disk=${uuid_gpt_disk};name=loader1,start=32K,size=4000K,uuid=${uuid_gpt_loader1};name=loader2,start=8MB,size=4MB,uuid=${uuid_gpt_loader2};name=trust,size=4M,uuid=${uuid_gpt_atf};name=boot,size=112M,bootable,uuid=${uuid_gpt_boot};name=rootfs,size=-,uuid=69DAD710-2CE4-4E3C-B16C-21A1D49ABED3;
pxefile_addr_r=0x00100000
ramdisk_addr_r=0x0a200000
reset_reason=POR
rkimg_bootdev=if mmc dev 1 && rkimgtest mmc 1; then setenv devtype mmc; setenv devnum 1; echo Boot from SDcard;elif mmc dev 0; then setenv devtype mmc; setenv devnum 0;elif mtd_blk dev 0; then setenv devtype mtd; setenv devnum 0;elif mtd_blk dev 1; then setenv devtype mtd; setenv devnum 1;elif mtd_blk dev 2; then setenv devtype mtd; setenv devnum 2;elif rknand dev 0; then setenv devtype rknand; setenv devnum 0;elif rksfc dev 0; then setenv devtype spinand; setenv devnum 0;elif rksfc dev 1; then setenv devtype spinor; setenv devnum 1;elsesetenv devtype ramdisk; setenv devnum 0;fi;
scan_dev_for_boot=echo Scanning ${devtype} ${devnum}:${distro_bootpart}...; for prefix in ${boot_prefixes}; do run scan_dev_for_extlinux; run scan_dev_for_scripts; done;
scan_dev_for_boot_part=part list ${devtype} ${devnum} -bootable devplist; env exists devplist || setenv devplist 1; for distro_bootpart in ${devplist}; do if fstype ${devtype} ${devnum}:${distro_bootpart} bootfstype; then run scan_dev_for_boot; fi; done
scan_dev_for_extlinux=if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}extlinux/extlinux.conf; then echo Found ${prefix}extlinux/extlinux.conf; run boot_extlinux; echo SCRIPT FAILED: continuing...; fi
scan_dev_for_scripts=for script in ${boot_scripts}; do if test -e ${devtype} ${devnum}:${distro_bootpart} ${prefix}${script}; then echo Found U-Boot script ${prefix}${script}; run boot_a_script; echo SCRIPT FAILED: continuing...; fi; done
scriptaddr=0x00000000
serial#=c3d9b8674f4b94f6
soc=rockchip
stderr=serial,vidconsole
stdout=serial,vidconsole
usb_boot=usb start; if usb dev ${devnum}; then setenv devtype usb; run scan_dev_for_boot_part; fi
vendor=rockchip
```

Or this:
```
@DDR Version 1.11 20210818
In
Channel a: DDR3 400MHz
Bus Width=32 Col=10 Bank=8 Row=15 CS=1 Die Bus-Width=16 Size=1024MB
Channel b: DDR3 400MHz
Bus Width=32 Col=10 Bank=8 Row=15 CS=1 Die Bus-Width=16 Size=1024MB
Memory OK
Memory OK
OUT
Boot1 Release Time: Jul 22 2021 09:08:57, version: 2.63
ChipType = 0x8, 236
mmc2:cmd19,100
SdmmcInit=2 0
BootCapSize=2000
UserCapSize=7456MB
FwPartOffset=2000 , 2000
mmc0:cmd8,400
mmc0:cmd8,400
mmc0:cmd8,400
[... 79 lines cut ...]
optee api revision: 2.0
TEEC: Waring: Could not find security partition
Not AVB images, AVB skip
Booting ZIMAGE kernel at 0x02008000(Uncompress to 0) with fdt at 0x08300000...

Fdt Ramdisk skip relocation
No misc partition
## Booting Android Image at 0x02007800 ...
Kernel: 0x02008000 - 0x026b4ff0 (6836 KiB)
## Flattened Device Tree blob at 0x08300000
   Booting using the fdt blob at 0x08300000
   XIP Kernel Image from 0x02008000 to 0x02008000 ... OK
   kernel loaded at 0x02008000, end = 0x026b4ff0
  'reserved-memory' dma-unusable@fe000000: addr=fe000000 size=1000000
   Using Device Tree in place at 08300000, end 083163dd
can't found rockchip,drm-logo, use rockchip,fb-logo
WARNING: could not set reg FDT_ERR_BADOFFSET.
failed to reserve fb-loader-logo memory
Adding bank: 0x00000000 - 0x08400000 (size: 0x08400000)
Adding bank: 0x09310000 - 0x80000000 (size: 0x76cf0000)
Total: 514.186 ms

Starting kernel ...

```

JeffyCN, 2022-03-01:
1/ according to your previous comment, the manufacturer.dtb(manufacturer's dts + manufacturer's dtsi) works, the manufacturer's dts + poky(your current kernel)'s dtsi stucks. so guessing the problem might due to different dtsi, you can try to get the dtsi too, or check the difference from the decoded dtb.

2/ you can try to enable early boot logs(in kernel dts) to get more logs, one way to do it would be using earlycon:
```
+++ b/arch/arm/boot/dts/rk3288-linux.dtsi
@@ -8,7 +8,7 @@

 / {
        chosen {
-               bootargs = "earlyprintk console=ttyFIQ0 rw root=PARTUUID=614e0000-0000 rootfstype=ext4 rootwait";
+               bootargs = "earlycon=uart8250,mmio32,0xff690000 keep_bootcon console=ttyFIQ0 rw root=PARTUUID=614e0000-0000 rootfstype=ext4 rootwait";
        };
```

TYeniyayla, 2022-03-01:
1) I'm still looking into it, so far I haven't seen a reason.
2) I did what you said and created the image. The output was as follows:
```
U-Boot 2017.09 (Jan 10 2022 - 01:38:43 +0000)

Model: Evb-RK3288
PreSerial: 2, raw, 0xff690000
DRAM:  2 GiB
Sysmem: init
Relocation Offset: 00000000
Relocation fdt: 7bdd0948 - 7bdd1ddd
CR: M/C/I
Using default environment

DM: v1
dwmmc@ff0c0000: 1, dwmmc@ff0f0000: 0
Bootdev(atags): mmc 0
MMC0: High Speed, 52Mhz
PartType: EFI
No misc partition
boot mode: normal
Found DTB in boot part
DTB: rk-kernel.dtb
HASH(c): OK
I2c0 speed: 400000Hz
PMIC:  RK808
vdd_arm 1100000 uV
vdd_gpu 1100000 uV
Can't get crtc id, default set to id = 0
Can't get crtc id, default set to id = 0
Could not find baseparameter partition
Rockchip UBOOT DRM driver version: v1.0.1
Using display timing dts
lvds:  detailed mode clock 51200 kHz, flags[a]
    H: 1024 1184 1188 1344
    V: 0600 0612 0615 0635
bus_format: 100e
failed to set phy mode: -19
hdmi@ff980000 disconnected
CLK: (uboot. arm: enter 1000000 KHz, init 1000000 KHz, kernel 0N/A)
  apll 1000000 KHz
  dpll 396000 KHz
  cpll 24000 KHz
  gpll 594000 KHz
  npll 24000 KHz
  aclk_bus 198000 KHz
No misc partition
Net:   Net Initialization Skipped
No ethernet found.
Hit key to stop autoboot('CTRL+C'):  0
Could not find misc partition
ANDROID: reboot reason: "(none)"
optee api revision: 2.0
TEEC: Waring: Could not find security partition
Not AVB images, AVB skip
Booting ZIMAGE kernel at 0x02008000(Uncompress to 0) with fdt at 0x08300000...

Fdt Ramdisk skip relocation
No misc partition
## Booting Android Image at 0x02007800 ...
Kernel: 0x02008000 - 0x026b4ff0 (6836 KiB)
## Flattened Device Tree blob at 0x08300000
   Booting using the fdt blob at 0x08300000
   XIP Kernel Image from 0x02008000 to 0x02008000 ... OK
   kernel loaded at 0x02008000, end = 0x026b4ff0
  'reserved-memory' dma-unusable@fe000000: addr=fe000000 size=1000000
   Using Device Tree in place at 08300000, end 08316405
can't found rockchip,drm-logo, use rockchip,fb-logo
WARNING: could not set reg FDT_ERR_BADOFFSET.
failed to reserve fb-loader-logo memory
[... 138 lines cut ...]
[    0.997996] io scheduler noop registered
[    1.002317] io scheduler deadline registered
[    1.007156] io scheduler cfq registered (default)
[    1.019614] backlight supply power not found, using dummy regulator
[    1.027616] rk-vcodec ff9a0000.vpu-service: no regulator for vcodec
[    1.034941] rk-vcodec ff9a0000.vpu-service: probe device
[    1.041262] rk-vcodec ff9a0000.vpu-service: drm allocator with mmu enabled
[    1.050141] rk-vcodec ff9a0000.vpu-service: could not find power_model node
[    1.057849] rk-vcodec ff9a0000.vpu-service: init success
[    1.064286] rk-vcodec ff9c0000.hevc-service: no regulator for vcodec
[    1.071819] rk-vcodec ff9c0000.hevc-service: probe device
[    1.078197] rk-vcodec ff9c0000.hevc-service: drm allocator with mmu enabled
[    1.086765] rk-vcodec ff9c0000.hevc-service: could not find power_model node
[    1.094511] rk-vcodec ff9c0000.hevc-service: init success
[    1.104172] dma-pl330 ff250000.dma-controller: Loaded driver for PL330 DMAC-241330
[    1.112562] dma-pl330 ff250000.dma-controller:       DBUFF-128x8bytes Num_Chans-8 Num_Peri-20 Num_Events-16
[    1.124134] dma-pl330 ff600000.dma-controller: Loaded driver for PL330 DMAC-241330
[    1.132506] dma-pl330 ff600000.dma-controller:       DBUFF-64x8bytes Num_Chans-5 Num_Peri-6 Num_Events-10
[    1.144479] rockchip-system-monitor rockchip-system-monitor: system monitor probe
[    1.153769] Serial: 8250/16550 driver, 5 ports, IRQ sharing disabled
[    1.161801] ff180000.serial: ttyS0 at MMIO 0xff180000 (irq = 37, base_baud = 1500000) is a 16550A
[    1.172601] ff190000.serial: ttyS1 at MMIO 0xff190000 (irq = 38, base_baud = 1500000) is a 16550A
[    1.183433] ff690000.serial: ttyS2 at MMIO 0xff690000 (irq = 39, base_baud = 1500000) is a 16550A

```

JeffyCN, 2022-03-01:
try to disable uart2 in that dts, remove this:
```
&uart2 {
        status = "okay";
};
```

TYeniyayla, 2022-03-01:
I just deleted it. The result is like this:
```
U-Boot 2017.09 (Jan 10 2022 - 01:38:43 +0000)

Model: Evb-RK3288
PreSerial: 2, raw, 0xff690000
DRAM:  2 GiB
Sysmem: init
Relocation Offset: 00000000
Relocation fdt: 7bdd0948 - 7bdd1ddd
CR: M/C/I
Using default environment

DM: v1
dwmmc@ff0c0000: 1, dwmmc@ff0f0000: 0
Bootdev(atags): mmc 0
MMC0: High Speed, 52Mhz
PartType: EFI
No misc partition
boot mode: normal
Found DTB in boot part
DTB: rk-kernel.dtb
HASH(c): OK
I2c0 speed: 400000Hz
PMIC:  RK808
vdd_arm 1100000 uV
vdd_gpu 1100000 uV
Can't get crtc id, default set to id = 0
Can't get crtc id, default set to id = 0
Could not find baseparameter partition
Rockchip UBOOT DRM driver version: v1.0.1
Using display timing dts
lvds:  detailed mode clock 51200 kHz, flags[a]
    H: 1024 1184 1188 1344
    V: 0600 0612 0615 0635
bus_format: 100e
failed to set phy mode: -19
hdmi@ff980000 disconnected
CLK: (uboot. arm: enter 1000000 KHz, init 1000000 KHz, kernel 0N/A)
  apll 1000000 KHz
  dpll 396000 KHz
  cpll 24000 KHz
  gpll 594000 KHz
  npll 24000 KHz
  aclk_bus 198000 KHz
No misc partition
Net:   Net Initialization Skipped
No ethernet found.
Hit key to stop autoboot('CTRL+C'):  0
Could not find misc partition
ANDROID: reboot reason: "(none)"
optee api revision: 2.0
TEEC: Waring: Could not find security partition
Not AVB images, AVB skip
Booting ZIMAGE kernel at 0x02008000(Uncompress to 0) with fdt at 0x08300000...

Fdt Ramdisk skip relocation
No misc partition
## Booting Android Image at 0x02007800 ...
Kernel: 0x02008000 - 0x026b4ff0 (6836 KiB)
## Flattened Device Tree blob at 0x08300000
   Booting using the fdt blob at 0x08300000
   XIP Kernel Image from 0x02008000 to 0x02008000 ... OK
   kernel loaded at 0x02008000, end = 0x026b4ff0
  'reserved-memory' dma-unusable@fe000000: addr=fe000000 size=1000000
   Using Device Tree in place at 08300000, end 08316405
can't found rockchip,drm-logo, use rockchip,fb-logo
WARNING: could not set reg FDT_ERR_BADOFFSET.
failed to reserve fb-loader-logo memory
Adding bank: 0x00000000 - 0x08400000 (size: 0x08400000)
[... 137 lines cut ...]
[    1.007899] io scheduler noop registered
[    1.012220] io scheduler deadline registered
[    1.017059] io scheduler cfq registered (default)
[    1.029505] backlight supply power not found, using dummy regulator
[    1.037512] rk-vcodec ff9a0000.vpu-service: no regulator for vcodec
[    1.044838] rk-vcodec ff9a0000.vpu-service: probe device
[    1.051163] rk-vcodec ff9a0000.vpu-service: drm allocator with mmu enabled
[    1.060026] rk-vcodec ff9a0000.vpu-service: could not find power_model node
[    1.067736] rk-vcodec ff9a0000.vpu-service: init success
[    1.074172] rk-vcodec ff9c0000.hevc-service: no regulator for vcodec
[    1.081702] rk-vcodec ff9c0000.hevc-service: probe device
[    1.088079] rk-vcodec ff9c0000.hevc-service: drm allocator with mmu enabled
[    1.096652] rk-vcodec ff9c0000.hevc-service: could not find power_model node
[    1.104398] rk-vcodec ff9c0000.hevc-service: init success
[    1.114048] dma-pl330 ff250000.dma-controller: Loaded driver for PL330 DMAC-241330
[    1.122438] dma-pl330 ff250000.dma-controller:       DBUFF-128x8bytes Num_Chans-8 Num_Peri-20 Num_Events-16
[    1.134009] dma-pl330 ff600000.dma-controller: Loaded driver for PL330 DMAC-241330
[    1.142380] dma-pl330 ff600000.dma-controller:       DBUFF-64x8bytes Num_Chans-5 Num_Peri-6 Num_Events-10
[    1.154347] rockchip-system-monitor rockchip-system-monitor: system monitor probe
[    1.163648] Serial: 8250/16550 driver, 5 ports, IRQ sharing disabled
[    1.171679] ff180000.serial: ttyS0 at MMIO 0xff180000 (irq = 37, base_baud = 1500000) is a 16550A
[    1.182472] ff190000.serial: ttyS1 at MMIO 0xff190000 (irq = 38, base_baud = 1500000) is a 16550A
[    1.193304] ff690000.serial: ttyS2 at MMIO 0xff690000 (irq = 39, base_baud = 1500000) is a 16550A

```

JeffyCN, 2022-03-01:
[    0.497624] uart2 is enabled, please disable it
[    0.502667] rk-fiq-debugger: probe of fiq-debugger failed with error -22

this log means it's still enabled... maybe try to:
```
&uart2 {
        status = "disabled";
};
```

TYeniyayla, 2022-03-01:
I disabled it but the uboot output is still the same. The dts file looks like this:
```
/*
 * This file is dual-licensed: you can use it either under the terms
 * of the GPL or the X11 license, at your option. Note that this dual
 * licensing only applies to this file, and not this project as a
 * whole.
 *
 *  a) This file is free software; you can redistribute it and/or
 *     modify it under the terms of the GNU General Public License as
 *     published by the Free Software Foundation; either version 2 of the
 *     License, or (at your option) any later version.
 *
 *     This file is distributed in the hope that it will be useful,
 *     but WITHOUT ANY WARRANTY; without even the implied warranty of
 *     MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *     GNU General Public License for more details.
 *
 * Or, alternatively,
 *
 *  b) Permission is hereby granted, free of charge, to any person
 *     obtaining a copy of this software and associated documentation
 *     files (the "Software"), to deal in the Software without
 *     restriction, including without limitation the rights to use,
 *     copy, modify, merge, publish, distribute, sublicense, and/or
 *     sell copies of the Software, and to permit persons to whom the
 *     Software is furnished to do so, subject to the following
 *     conditions:
 *
 *     The above copyright notice and this permission notice shall be
 *     included in all copies or substantial portions of the Software.
 *
 *     THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND,
 *     EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES
 *     OF MERCHANTABILITY, FITNESS FOR A PARTICULAR PURPOSE AND
 *     NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT
 *     HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER LIABILITY,
 *     WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
 *     FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR
 *     OTHER DEALINGS IN THE SOFTWARE.
 */

/dts-v1/;
#include <dt-bindings/clock/rockchip,rk808.h>
#include <dt-bindings/pwm/pwm.h>
#include <dt-bindings/input/input.h>
#include "rk3288.dtsi"
#include "rk3288-rkisp1.dtsi"
#include "rk3288-linux.dtsi"

/ {
	compatible = "rockchip,rk3288-miniarm", "rockchip,rk3288";

	backlight: backlight {
		compatible = "pwm-backlight";
		brightness-levels = <
			255 190 189 188 187 186 185 184
			183 182 181 180	179 178 177 176
			175 174 173 172 171 170 169 168
			167 166 165 164 163 162 161 160
			159 158 157 156 155 154 153 152
			151 150 149 148 147 146 145 144
			143 142 141 140	139 138 137 136
			135 134 133 132 131 130 129 128
			127 126 125 124 123 122 121 120
			119 118 117 116 115 114 113 112
			111 110 109 108 107 106 105 104
			103 102 101 100	99	98	97	96
			95	94	93	92	91	90	89	88
			87	86	85	84	83	82	81	80
[... 981 lines cut ...]
		otg_vbus_drv: otg-vbus-drv {
			rockchip,pins = <0 12 RK_FUNC_GPIO &pcfg_pull_none>;
		};
	};

	wireless-bluetooth {
		uart0_gpios: uart0-gpios {
			rockchip,pins = <4 19 RK_FUNC_GPIO &pcfg_pull_none>;
		};
	};

    sound {
        hpmic_det: hpmic-det {
			rockchip,pins = <0 14 RK_FUNC_GPIO &pcfg_pull_down>;
        };
    };

	// cam_pins {
	// 	cam_pwr: cam-pwr {
	// 		rockchip,pins = <2 0 RK_FUNC_GPIO &pcfg_pull_none>;
	// 	};
	// };
};

```

JeffyCN, 2022-03-01:
that log is very clear, the uart2 is enabled in dtb. so guessing wrong dtb been used...

TYeniyayla, 2022-03-01:
Yes, that's what happened when uart2 was turned off, I guess the system still won't boot. staying here:
```
U-Boot 2017.09 (Jan 10 2022 - 01:38:43 +0000)

Model: Evb-RK3288
PreSerial: 2, raw, 0xff690000
DRAM:  2 GiB
Sysmem: init
Relocation Offset: 00000000
Relocation fdt: 7bdd0948 - 7bdd1ddd
CR: M/C/I
Using default environment

DM: v1
dwmmc@ff0c0000: 1, dwmmc@ff0f0000: 0
Bootdev(atags): mmc 0
MMC0: High Speed, 52Mhz
PartType: EFI
No misc partition
boot mode: normal
Found DTB in boot part
DTB: rk-kernel.dtb
HASH(c): OK
I2c0 speed: 400000Hz
PMIC:  RK808
vdd_arm 1100000 uV
vdd_gpu 1100000 uV
Can't get crtc id, default set to id = 0
Can't get crtc id, default set to id = 0
Could not find baseparameter partition
Rockchip UBOOT DRM driver version: v1.0.1
Using display timing dts
lvds:  detailed mode clock 51200 kHz, flags[a]
    H: 1024 1184 1188 1344
    V: 0600 0612 0615 0635
bus_format: 100e
failed to set phy mode: -19
hdmi@ff980000 disconnected
CLK: (uboot. arm: enter 1000000 KHz, init 1000000 KHz, kernel 0N/A)
  apll 1000000 KHz
  dpll 396000 KHz
  cpll 24000 KHz
  gpll 594000 KHz
  npll 24000 KHz
  aclk_bus 198000 KHz
No misc partition
Net:   Net Initialization Skipped
No ethernet found.
Hit key to stop autoboot('CTRL+C'):  0
Could not find misc partition
ANDROID: reboot reason: "(none)"
optee api revision: 2.0
TEEC: Waring: Could not find security partition
Not AVB images, AVB skip
Booting ZIMAGE kernel at 0x02008000(Uncompress to 0) with fdt at 0x08300000...

Fdt Ramdisk skip relocation
No misc partition
## Booting Android Image at 0x02007800 ...
Kernel: 0x02008000 - 0x026b4ff8 (6836 KiB)
## Flattened Device Tree blob at 0x08300000
   Booting using the fdt blob at 0x08300000
   XIP Kernel Image from 0x02008000 to 0x02008000 ... OK
   kernel loaded at 0x02008000, end = 0x026b4ff8
  'reserved-memory' dma-unusable@fe000000: addr=fe000000 size=1000000
   Using Device Tree in place at 08300000, end 08316409
can't found rockchip,drm-logo, use rockchip,fb-logo
WARNING: could not set reg FDT_ERR_BADOFFSET.
failed to reserve fb-loader-logo memory
Adding bank: 0x00000000 - 0x08400000 (size: 0x08400000)
[... 666 lines cut ...]
[    7.096260] mmc_host mmc2: Bus speed (slot 0) = 49500000Hz (slot req 50000000Hz, actual 49500000HZ div = 0)
[    7.097657] mmc2: new high speed SDIO card at address 0001
[    7.097657] mmc2: new high speed SDIO card at address 0001
[    7.135798] I : [File] : /media/tunahan/f1cceee6-ac24-4956-9e80-7ce648ac31e4/yoctoproject/build/tmp/work-shared/mepsan-pcr172/kernel-source/drivers/gpu/arm/mali400/mali/linux/mali_kernel_linux.c; [Line] : 413; [Func] : mali_module_init(); svn_rev_string_from_arm of this mali_ko is '', rk_ko_ver is '5', built at '03:48:33', on 'Jan  7 2022'.
[    7.135798] I : [File] : /media/tunahan/f1cceee6-ac24-4956-9e80-7ce648ac31e4/yoctoproject/build/tmp/work-shared/mepsan-pcr172/kernel-source/drivers/gpu/arm/mali400/mali/linux/mali_kernel_linux.c; [Line] : 413; [Func] : mali_module_init(); svn_rev_string_from_arm of this mali_ko is '', rk_ko_ver is '5', built at '03:48:33', on 'Jan  7 2022'.
[    7.197114] Mali: [    7.197114] Mali: Mali device driver loaded
Mali device driver loaded
[    7.205604] rkisp1 ff910000.isp: clear unready subdev num: 1
[    7.205604] rkisp1 ff910000.isp: clear unready subdev num: 1
[    7.217217] rkisp1: update sensor failed
[    7.217217] rkisp1: update sensor failed
[    7.225386] ALSA device list:
[    7.225386] ALSA device list:
[    7.231405]   No soundcards found.
[    7.231405]   No soundcards found.
[    7.241095] EXT4-fs (mmcblk1p4): mounted filesystem with ordered data mode. Opts: (null)
[    7.241095] EXT4-fs (mmcblk1p4): mounted filesystem with ordered data mode. Opts: (null)
[    7.257638] VFS: Mounted root (ext4 filesystem) on device 179:4.
[    7.257638] VFS: Mounted root (ext4 filesystem) on device 179:4.
[    7.270169] devtmpfs: mounted
[    7.270169] devtmpfs: mounted
[    7.277049] Freeing unused kernel memory: 1024K
[    7.281577] Unable to handle kernel paging request at virtual address c0f2c4b8

```

JeffyCN, 2022-03-01:
so it seems like kernel booted, but crashed somewhere.

you can try to revert the bootargs changes(earlycon=uart8250,mmio32,0xff690000 keep_bootcon), and start to disable devices in the dts to figure out which driver caused that crash

TYeniyayla, 2022-03-01:
After restoring the Bootargs variables, the system booted. It works at 5 volts. Thanks for the help.