JeffyCN/meta-rockchip #11: Trouble with Chromium 98.0.4758.102-r0

MastaG, 2022-03-11:
I'm trying to build Chromium 98.0.4758.102-r0.
The patches do apply, but it fails with:
```
error: undefined symbol: media::V4L2VideoDecoder::Create(std::__1::unique_ptr<media::MediaLog, std::__1::default_delete<media::MediaLog> >, scoped_refptr<base::SequencedTaskRunner>, base::WeakPtr<media::VideoDecoderMixin::Client>)
| >>> referenced by video_decoder_pipeline.cc:0 (./../../media/gpu/chromeos/video_decoder_pipeline.cc:0)
```

It seems media::V4L2VideoDecoder::Create is called in: ```/media/gpu/chromeos/video_decoder_pipeline.cc```
```
 #if BUILDFLAG(USE_VAAPI)
      base::BindOnce(&VaapiVideoDecoder::Create);
#elif BUILDFLAG(USE_V4L2_CODEC)
      base::BindOnce(&V4L2VideoDecoder::Create);
#endif
```

Which is included because ```/media/gpu/BUILD.gn``` has:
```
  # TODO(crbug.com/1006266): consider using |use_chromeos_video_acceleration|.
  if (use_v4l2_codec || use_vaapi) {
    public_deps += [
      ":video_frame_mapper",
      "//media/gpu/chromeos",
    ]
  }
```

I'm I supposed to build  ```//media/gpu/chromeos``` even on the ozone platform for linux?

MastaG, 2022-03-11:
It seems ```V4L2VideoDecoder::Create``` is defined in ```/media/gpu/v4l2/v4l2_video_decoder.cc``` which is guarded by ```if (!use_linux_v4l2_only)```.

So it seems they did some refactoring.
I can try to remove the "use_linux_v4l2_only" flag so these files get built as well.

MastaG, 2022-03-11:
@JeffyCN 
It seems the following is required (on top of the existing patches) to make Chromium 98 build again:

Fix for undefined symbol: ```media::V4L2VideoDecoder::Create``` in /media/gpu/chromeos/video_decoder_pipeline.cc:
```
--- a/media/gpu/v4l2/BUILD.gn	2022-03-11 10:31:31.649092450 +0100
+++ b/media/gpu/v4l2/BUILD.gn	2022-03-11 13:38:00.834024785 +0100
@@ -51,6 +51,8 @@
     "v4l2_video_decoder_backend_stateful.h",
     "v4l2_video_encode_accelerator.cc",
     "v4l2_video_encode_accelerator.h",
+    "v4l2_video_decoder.cc",
+    "v4l2_video_decoder.h",
   ]
 
   if (!use_linux_v4l2_only) {
@@ -60,8 +62,6 @@
       "v4l2_decode_surface_handler.h",
       "v4l2_slice_video_decode_accelerator.cc",
       "v4l2_slice_video_decode_accelerator.h",
-      "v4l2_video_decoder.cc",
-      "v4l2_video_decoder.h",
       "v4l2_video_decoder_backend_stateless.cc",
       "v4l2_video_decoder_backend_stateless.h",
       "v4l2_video_decoder_delegate_h264.cc",
```
Fix for undefined symbol: ```base::SequencedTaskRunnerHandle::Get``` in media/gpu/v4l2/v4l2_video_decoder.cc:
```
--- a/media/gpu/v4l2/v4l2_video_decoder.cc	2022-03-11 10:31:31.433091249 +0100
+++ b/media/gpu/v4l2/v4l2_video_decoder.cc	2022-03-11 13:45:04.486423379 +0100
@@ -11,6 +11,7 @@
 #include "base/logging.h"
 #include "base/memory/ptr_util.h"
 #include "base/task/post_task.h"
+#include "base/threading/sequenced_task_runner_handle.h"
 #include "base/trace_event/trace_event.h"
 #include "media/base/limits.h"
 #include "media/base/media_log.h"
```

Correct me if I'm wrong but this makes it build again with:
```
use_v4l2_codec=true
use_v4lplugin=true
use_linux_v4l2_only=true
```

JeffyCN, 2022-03-14:
yes, you are right.

patches uploaded(https://github.com/JeffyCN/meta-rockchip/commit/2345140c7505b78148d27deb1f0bdd8cab6d3637), which is much the same as what you did

MastaG, 2022-04-12:
Hi @JeffyCN 
If you happen to have some spare time, could you please rebase on chromium 100 ?
They just pushed it to meta-browser.

Thanks in advance!

JeffyCN, 2022-04-13:
i've been busy these days, should be able to port it before the end of the next week...

JeffyCN, 2022-04-18:
@MastaG , i've ported the patches:[wip/chromium_100](https://github.com/JeffyCN/meta-rockchip/tree/wip/chromium_100)
but haven't got time to test it

MastaG, 2022-04-18:
@JeffyCN thanks a lot bro!
I'll build it tomorrow and hope to run it later this week 👍

JeffyCN, 2022-04-20:
patches uploaded