JeffyCN/meta-rockchip #12: LVDS or HDMI display not working

TYeniyayla, 2022-03-22:
Hello,
I'm building OS using Yocto, at boot time I can only see splashscreen on HDMI, but then the screen goes black and HSMI goes into no signal. In LVDS, only the backlight works, but no image is displayed.

I am using `bitbake core-image-x11` when compiling. When I run a qt interface application over ssh or debug port, I get an error like this:
```
qt.qpa.xcb: could not connect to display :0.0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

aborted
```
Kernel was used independently of Yocto. LVDS and HDMI worked flawlessly.

I can send it if you need extra information to solve the problem. Thanks.

JeffyCN, 2022-03-22:
try to upload the /var/log

TYeniyayla, 2022-03-23:
/var/log file is here: [log.zip](https://github.com/JeffyCN/meta-rockchip/files/8330739/log.zip)

JeffyCN, 2022-03-23:
[ 38024.475] (II) LoadModule: "modesetting"
[ 38024.475] (WW) Warning, couldn't open module modesetting
[ 38024.475] (EE) Failed to load module "modesetting" (module does not exist, 0)

so the modesetting is missing.
i was using this manifest:
https://github.com/JeffyCN/yocto-manifests

the xf86-video-modesetting enabled at here:
https://github.com/JeffyCN/yocto-conf/blob/master/include/display.conf#L10

TYeniyayla, 2022-03-23:
I added the following line inside the local.conf file:
`IMAGE_INSTALL_append = " xf86-video-modesetting xserver-xorg-module-exa"`

The screen still didn't come up and the Qt app still didn't work. There was no apparent change.

New /var/log file: [log.zip](https://github.com/JeffyCN/meta-rockchip/files/8331156/log.zip)

JeffyCN, 2022-03-23:
[   158.252] (WW) Falling back to old probe method for modesetting
[   158.252] (EE) No devices detected.
[   158.252] (EE)
Fatal server error:
[   158.252] (EE) no screens found(EE)
[   158.252] (EE)

so no devices detected, something is wrong on the kernel(drm driver) or device node permission.
try to add this udev rules to fixup device node permissions:
https://github.com/JeffyCN/yocto-conf/blob/master/include/common.conf#L292

and use modetest tool(drm-tests) to check the drm driver:
https://github.com/JeffyCN/yocto-conf/blob/master/include/debug.conf#L2

TYeniyayla, 2022-03-23:
When compiling, I get an error like this, what's missing?

```ERROR: Nothing RPROVIDES 'udev-conf-rockchip' (but /media/tunahan/f1cceee6-ac24-4956-9e80-7ce648ac31e4/yoctoproject/poky/meta/recipes-graphics/images/core-image-x11.bb RDEPENDS on or otherwise requires it)
NOTE: Runtime target 'udev-conf-rockchip' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['udev-conf-rockchip']
ERROR: Required build target 'core-image-x11' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-x11', 'udev-conf-rockchip']```

JeffyCN, 2022-03-23:
maybe your meta-rockchip is old, that package was added at Feb 10:
https://github.com/JeffyCN/meta-rockchip/commit/3b60aa073d57e24893e5f4cc6ae6f579e929b555

JeffyCN, 2022-03-23:
same error, so there must be something wrong in the kernel side.

with libdrm-tests installed, there would be a modetest tool, you can run it to check the information about drm driver.

and according to boot logs, lvds might not be probed(failed to get phy)

TYeniyayla, 2022-03-23:
How can I run drm tests?

The phy error was always writing, I guess because I gave the power to lvds from outside, not from the card, but it was working.

How can we detect if there is an error on the kernel side, and I was using this kernel alone. Before yocto, I was flashing and using DevTool such as `MiniLoaderAll.bin`, `parameter` and `zboot.img`. LVDS and HDMI could be used.

I am attaching 3 files that I use with devtool without Yocto.

[devtoolfiles.zip](https://github.com/JeffyCN/meta-rockchip/files/8333510/devtoolfiles.zip)

TYeniyayla, 2022-03-23:
Did I do it right? It gave a result like this:

```
usage: modetest [-acDdefMPpsCvw]

 Query options:

        -c      list connectors
        -e      list encoders
        -f      list framebuffers
        -p      list CRTCs and planes (pipes)

 Test options:

        -P <plane_id>@<crtc_id>:<w>x<h>[+<x>+<y>][*<scale>][@<format>]  set a plane
        -s <connector_id>[,<connector_id>][@<crtc_id>]:[#<mode index>]<mode>[-<vrefresh>][@<format>]    set a mode
        -C      test hw cursor
        -v      test vsynced page flipping
        -w <obj_id>:<prop_name>:<value> set property
        -a      use atomic API
        -F pattern1,pattern2    specify fill patterns

 Generic options:

        -d      drop master after mode set
        -M module       use the given driver
        -D device       use the given device

        Default is to dump all info.

root@mepsan-pcr172:~# modetest -c
trying to open device 'i915'...failed
trying to open device 'amdgpu'...failed
trying to open device 'radeon'...failed
trying to open device 'nouveau'...failed
trying to open device 'vmwgfx'...failed
trying to open device 'omapdrm'...failed
trying to open device 'exynos'...failed
trying to open device 'tilcdc'...failed
trying to open device 'msm'...failed
trying to open device 'sti'...failed
trying to open device 'tegra'...failed
trying to open device 'imx-drm'...failed
trying to open device 'rockchip'...failed
trying to open device 'atmel-hlcdc'...failed
trying to open device 'fsl-dcu-drm'...failed
trying to open device 'vc4'...failed
trying to open device 'virtio_gpu'...failed
trying to open device 'mediatek'...failed
trying to open device 'meson'...failed
trying to open device 'pl111'...failed
trying to open device 'stm'...failed
trying to open device 'sun4i-drm'...failed
trying to open device 'armada-drm'...failed
no device found

root@mepsan-pcr172:~# modetest -e
trying to open device 'i915'...failed
trying to open device 'amdgpu'...failed
trying to open device 'radeon'...failed
trying to open device 'nouveau'...failed
trying to open device 'vmwgfx'...failed
trying to open device 'omapdrm'...failed
trying to open device 'exynos'...failed
trying to open device 'tilcdc'...failed
trying to open device 'msm'...failed
trying to open device 'sti'...failed
trying to open device 'tegra'...failed
trying to open device 'imx-drm'...failed
trying to open device 'rockchip'...failed
trying to open device 'atmel-hlcdc'...failed
trying to open device 'fsl-dcu-drm'...failed
trying to open device 'vc4'...failed
trying to open device 'virtio_gpu'...failed
trying to open device 'mediatek'...failed
trying to open device 'meson'...failed
trying to open device 'pl111'...failed
trying to open device 'stm'...failed
trying to open device 'sun4i-drm'...failed
trying to open device 'armada-drm'...failed
no device found

root@mepsan-pcr172:~# modetest -f
trying to open device 'i915'...failed
trying to open device 'amdgpu'...failed
trying to open device 'radeon'...failed
trying to open device 'nouveau'...failed
trying to open device 'vmwgfx'...failed
trying to open device 'omapdrm'...failed
trying to open device 'exynos'...failed
trying to open device 'tilcdc'...failed
trying to open device 'msm'...failed
trying to open device 'sti'...failed
trying to open device 'tegra'...failed
trying to open device 'imx-drm'...failed
trying to open device 'rockchip'...failed
trying to open device 'atmel-hlcdc'...failed
trying to open device 'fsl-dcu-drm'...failed
trying to open device 'vc4'...failed
trying to open device 'virtio_gpu'...failed
trying to open device 'mediatek'...failed
trying to open device 'meson'...failed
trying to open device 'pl111'...failed
trying to open device 'stm'...failed
trying to open device 'sun4i-drm'...failed
trying to open device 'armada-drm'...failed
no device found

root@mepsan-pcr172:~# modetest -p
trying to open device 'i915'...failed
trying to open device 'amdgpu'...failed
trying to open device 'radeon'...failed
trying to open device 'nouveau'...failed
trying to open device 'vmwgfx'...failed
trying to open device 'omapdrm'...failed
trying to open device 'exynos'...failed
trying to open device 'tilcdc'...failed
trying to open device 'msm'...failed
trying to open device 'sti'...failed
trying to open device 'tegra'...failed
trying to open device 'imx-drm'...failed
trying to open device 'rockchip'...failed
trying to open device 'atmel-hlcdc'...failed
trying to open device 'fsl-dcu-drm'...failed
trying to open device 'vc4'...failed
trying to open device 'virtio_gpu'...failed
trying to open device 'mediatek'...failed
trying to open device 'meson'...failed
trying to open device 'pl111'...failed
trying to open device 'stm'...failed
trying to open device 'sun4i-drm'...failed
trying to open device 'armada-drm'...failed
no device found
```

JeffyCN, 2022-03-23:
the modetest shows the drm device is not available, you can check /dev/dri/card0 manually

TYeniyayla, 2022-03-23:
/dev/ dump, card0 not showing up here:

```
root@mepsan-pcr172:~# ls /dev/
ashmem              i2c-2               loop6               net                 tty                 tty21               tty35               tty49               tty62               usbmon3             vcsa5
block               i2c-3               loop7               network_latency     tty0                tty22               tty36               tty5                tty63               v4l                 vcsa6
bus                 i2c-4               mali0               network_throughput  tty1                tty23               tty37               tty50               tty7                v4l-subdev0         vendor_storage
char                iio:device0         mapper              null                tty10               tty24               tty38               tty51               tty8                vcs                 vhci
console             initctl             media0              ppp                 tty11               tty25               tty39               tty52               tty9                vcs1                video0
cpu_dma_latency     input               mem                 ptmx                tty12               tty26               tty4                tty53               ttyFIQ0             vcs2                video1
disk                kmsg                memory_bandwidth    pts                 tty13               tty27               tty40               tty54               ttyS0               vcs3                video2
dri                 log                 mmcblk1             ram0                tty14               tty28               tty41               tty55               ttyS1               vcs4                video3
fd                  loop-control        mmcblk1boot0        random              tty15               tty29               tty42               tty56               ttyS3               vcs5                video4
full                loop0               mmcblk1boot1        rfkill              tty16               tty3                tty43               tty57               uhid                vcs6                vpu-service
fuse                loop1               mmcblk1p1           shm                 tty17               tty30               tty44               tty58               uinput              vcsa                zero
hevc-service        loop2               mmcblk1p2           snd                 tty18               tty31               tty45               tty59               urandom             vcsa1               zram0
hwrng               loop3               mmcblk1p3           stderr              tty19               tty32               tty46               tty6                usbmon0             vcsa2
i2c-0               loop4               mmcblk1p4           stdin               tty2                tty33               tty47               tty60               usbmon1             vcsa3
i2c-1               loop5               mmcblk1rpmb         stdout              tty20               tty34               tty48               tty61               usbmon2             vcsa4
```

strace is not installed, I guess, how is it installed? Shall I set up?
```
root@mepsan-pcr172:~# strace -f X
-sh: strace: command not found
```

TYeniyayla, 2022-03-23:
How should I proceed? I am very unfamiliar with these subjects.

JeffyCN, 2022-03-23:
anyway, it's a kernel issue that the drm driver failed to bring up lvds, likely due to missing phy(something wrong in the clk/regulator or config/dts)

JeffyCN, 2022-03-23:
you can try to add logs in the kernel drm driver for example lvds driver's bind function to make sure it success.

or you can try to disable lvds in kernel dts, and use hdmi only

JeffyCN, 2022-03-23:
for drm, it would try to register all components specified in the dts(vop/lvds/hdmi and so on), and try to bring them up by calling their bind callback.

and it would keep retrying if any of those callback failed. so you can either fix the broken component or disable it

TYeniyayla, 2022-03-23:
I've been looking for hours, I can't see a wrong place. I changed the vopl and vopb parts, I tried again and again, but I did not get any results, I tried to turn off lvds, but I did not get any results. I wonder if you could examine at an appropriate time. I have attached the 4 dts files that I think are used. As far as I understand there is a time difference between us and you are awake while I am sleeping, there is a short time when we two are awake. So I'd be happy if you could take a look.

[dtsfiles.zip](https://github.com/JeffyCN/meta-rockchip/files/8336534/dtsfiles.zip)

JeffyCN, 2022-03-24:
1/
quick check with your dts, the lvds's phy(video_phy) is disabled(that is why your lvds failed to probe), you can try to enable it:
```
+++ b/rk3288.dtsi
@@ -1690,7 +1690,7 @@
                reset-names = "rst";
                power-domains = <&power RK3288_PD_VIO>;
                #phy-cells = <0>;
-               status = "disabled";
+               status = "okay";
        };
```

but weird thing is that it is enabled in the dumped dts from your previous provided zboot.img:
```
    video-phy@ff96c000 {
        compatible = "rockchip,rk3288-video-phy";
...
        status = "okay";
        phandle = < 0x67 >;
    };
```

so guessing something wrong during your porting.

2/
for disabling lvds, try:
```
+++ b/rk3288-lc-linux.dts
@@ -339,7 +339,7 @@
 };

 &lvds {
-       status = "okay";
+       status = "disabled";
        ports {
                lvds_out: port@1 {
                        reg = <1>;
@@ -432,7 +432,7 @@
 };

 &route_lvds {
-       status = "okay";
+       status = "disabled";
 };
```

3/
for further debugging, try attached debug patch
[drm.txt](https://github.com/JeffyCN/meta-rockchip/files/8337699/drm.txt), which would enable more logs:
sh-5.1# dmesg|grep -E "display-subsystem|compo"
[    0.924043] dwhdmi-rockchip ff940000.hdmi: adding component (ops 0xffffff8008afb188)
[    0.926015] rockchip-dp ff970000.edp: adding component (ops 0xffffff8008afd8c8)
[    0.928370] rockchip-vop ff8f0000.vop: adding component (ops 0xffffff8008b06e30)
[    0.929250] rockchip-vop ff900000.vop: adding component (ops 0xffffff8008b06e30)
[    0.930841] rockchip-drm display-subsystem: component_match_add /vop@ff900000
[    0.931501] rockchip-drm display-subsystem: component_match_add /vop@ff8f0000
[    0.932158] rockchip-drm display-subsystem: component_match_add /edp@ff970000
[    0.932825] rockchip-drm display-subsystem: component_match_add /hdmi@ff940000
[    0.933486] rockchip-drm display-subsystem: component_match_add /dp@fec00000
[    0.934168] rockchip-drm display-subsystem: component_match_add /edp@ff970000
[    0.934820] rockchip-drm display-subsystem: component_match_add /hdmi@ff940000
[    0.935474] rockchip-drm display-subsystem: component_match_add /dp@fec00000
[    1.517118] cdn-dp fec00000.dp: adding component (ops 0xffffff8008afc218)
[    1.517829] rockchip-drm display-subsystem: defer getting devfreq
[    1.518559] rockchip-drm display-subsystem: binding ff900000.vop (ops 0xffffff8008b06e30)
[    1.520084] rockchip-drm display-subsystem: bound ff900000.vop (ops 0xffffff8008b06e30)
[    1.520806] rockchip-drm display-subsystem: binding ff8f0000.vop (ops 0xffffff8008b06e30)
[    1.522238] rockchip-drm display-subsystem: bound ff8f0000.vop (ops 0xffffff8008b06e30)
[    1.522967] rockchip-drm display-subsystem: binding ff970000.edp (ops 0xffffff8008afd8c8)
[    1.525530] rockchip-drm display-subsystem: bound ff970000.edp (ops 0xffffff8008afd8c8)
[    1.526234] rockchip-drm display-subsystem: binding ff940000.hdmi (ops 0xffffff8008afb188)
[    1.530717] rockchip-drm display-subsystem: bound ff940000.hdmi (ops 0xffffff8008afb188)
[    1.531445] rockchip-drm display-subsystem: binding fec00000.dp (ops 0xffffff8008afc218)
[    1.533373] rockchip-drm display-subsystem: bound fec00000.dp (ops 0xffffff8008afc218)

"component_match_add"  means drm driver expected those components
"adding component" means the component driver registered it(usually means success probed)
"binding" means drm driver try to bind that compoment
"bound" means success to bind

so from those logs you will know which driver failed(probe or bind), and add more logs to it's probe()/bind() to debug

TYeniyayla, 2022-03-24:
1/
Normally, I used to open video_phy in rk3288-lc-linux.dts, but I did not open it, I added the following patch file, but there was no change.
```
--- a/arch/arm/boot/dts/rk3288-lc-linux.dts
+++ b/arch/arm/boot/dts/rk3288-lc-linux.dts
@@ -435,6 +435,10 @@
 	status = "okay";
 };
 
+&video_phy {
+	status = "okay";
+};
+
 &route_hdmi {
 	status = "okay";
 };
```

2/
Since what I wrote above was not enough, I added the patch file below, but this time, unlike my previous attempts, the splash screen did not appear on the HDMI.
```
--- a/arch/arm/boot/dts/rk3288-lc-linux.dts
+++ b/arch/arm/boot/dts/rk3288-lc-linux.dts
@@ -339,7 +339,7 @@
 };
 
 &lvds {
-	status = "okay";
+	status = "disabled";
 	ports {
 		lvds_out: port@1 {
 			reg = <1>;
@@ -364,7 +364,7 @@
 	rockchip,data-width = <24>;
 	pinctrl-0 = <&lcd_cs>;
 	power-supply = <&vcc_lcd>;
-	status = "okay";
+	status = "disabled";
 
 	display-timings {
 		native-mode = <&timing0>;
@@ -412,7 +412,7 @@
 		// };
 
 		route_lvds: route-lvds {
-		 	status = "okay";
+		 	status = "disabled";
 		 	logo,uboot = "logo.bmp";
 		 	logo,kernel = "logo_kernel.bmp";
 		 	logo,mode = "center";
```

3/
I added the patch file below, but when I use the `dmesg|grep -E "display-subsystem|compo"` command, no response is returned.
```
diff --git a/drivers/base/component.c b/drivers/base/component.c
index f748430bb654..6564168e5513 100644
--- a/drivers/base/component.c
+++ b/drivers/base/component.c
@@ -186,7 +186,7 @@ static int try_to_bring_up_master(struct master *master,
 	ret = master->ops->bind(master->dev);
 	if (ret < 0) {
 		devres_release_group(master->dev, NULL);
-		dev_info(master->dev, "master bind failed: %d\n", ret);
+		dev_err(master->dev, "master bind failed: %d\n", ret);
 		goto out;
 	}
 
@@ -401,7 +401,7 @@ static int component_bind(struct component *component, struct master *master,
 		return -ENOMEM;
 	}
 
-	dev_dbg(master->dev, "binding %s (ops %ps)\n",
+	dev_err(master->dev, "binding %s (ops %ps)\n",
 		dev_name(component->dev), component->ops);
 
 	ret = component->ops->bind(component->dev, master->dev, data);
@@ -417,7 +417,7 @@ static int component_bind(struct component *component, struct master *master,
 		devres_close_group(component->dev, NULL);
 		devres_remove_group(master->dev, NULL);
 
-		dev_info(master->dev, "bound %s (ops %ps)\n",
+		dev_err(master->dev, "bound %s (ops %ps)\n",
 			 dev_name(component->dev), component->ops);
 	} else {
 		devres_release_group(component->dev, NULL);
@@ -470,7 +470,7 @@ int component_add(struct device *dev, const struct component_ops *ops)
 	component->ops = ops;
 	component->dev = dev;
 
-	dev_dbg(dev, "adding component (ops %ps)\n", ops);
+	dev_err(dev, "adding component (ops %ps)\n", ops);
 
 	mutex_lock(&component_mutex);
 	list_add_tail(&component->node, &component_list);
	 
diff --git a/drivers/gpu/drm/rockchip/rockchip_drm_drv.c b/drivers/gpu/drm/rockchip/rockchip_drm_drv.c
index 0520bde44e5a..7ce52522821f 100644
--- a/drivers/gpu/drm/rockchip/rockchip_drm_drv.c
+++ b/drivers/gpu/drm/rockchip/rockchip_drm_drv.c
@@ -1832,6 +1832,7 @@ static void rockchip_add_endpoints(struct device *dev,
 			continue;
 		}
 
+		dev_err(dev, "component_match_add %s\n", remote->full_name);
 		component_match_add(dev, match, compare_of, remote);
 		of_node_put(remote);
 	}
@@ -1881,6 +1882,7 @@ static int rockchip_drm_platform_probe(struct platform_device *pdev)
 			is_support_iommu = false;
 		}
 
+		dev_err(dev, "component_match_add %s\n", port->parent->full_name);
 		component_match_add(dev, &match, compare_of, port->parent);
 		of_node_put(port);
 	}
@@ -1914,6 +1916,7 @@ static int rockchip_drm_platform_probe(struct platform_device *pdev)
 
 	port = of_parse_phandle(np, "backlight", 0);
 	if (port && of_device_is_available(port)) {
+		dev_err(dev, "component_match_add %s\n", port->full_name);
 		component_match_add(dev, &match, compare_of, port);
 		of_node_put(port);
 	}
```

I added it as code, but I am also adding the files: [changes.zip](https://github.com/JeffyCN/meta-rockchip/files/8341361/changes.zip)

JeffyCN, 2022-03-24:
since it's a kernel issue, i might not be able to provide more information...you might need to add more logs to the drm driver to figure it out(likely in the probe or bind).

or if you believe the old kernel works, maybe you can try the old zboot.img and make sure /dev/dri/card0 exist, then flash the yocto's rootfs.img.
for example:
upgrade_tool di -b zboot.img
upgrade_tool di -rootfs rootfs.img

TYeniyayla, 2022-03-24:
The old kernel is now working on other boards, no problem. When I throw the old kernel, it gives Kernel Panic. I think the line is this: `[    6.609305] ---[ end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,4)`

Here is the whole boot log:
```
In
Channel a: DDR3 400MHz
Bus Width=32 Col=10 Bank=8 Row=15 CS=1 Die Bus-Width=16 Size=1024MB
Channel b: DDR3 400MHz
Bus Width=32 Col=10 Bank=8 Row=15 CS=1 Die Bus-Width=16 Size=1024MB
Memory OK
Memory OK
OUT
Boot1 Release Time: Apr 11 2018 10:32:58, version: 2.36
ChipType = 0x8, 226
mmc2:cmd19,256
SdmmcInit=2 0
BootCapSize=2000
UserCapSize=7456MB
FwPartOffset=2000 , 2000
mmc0:cmd8,1024
mmc0:cmd8,1024
mmc0:cmd8,1024
SdmmcInit=0 1
StorageInit ok = 30928
LOADER Check OK! 0x200000, 66980
TOS    Check OK! 0x8400000, 101649
Enter Trust OS
INF [0x0] TEE-CORE:init_primary_helper:385: Core0 Initializing (1.1.0-432-gc99a453d #4 Mon Aug 17 09:16:49 UTC 2020 arm)

INF [0x0] TEE-CORE:init_primary_helper:386: Release version: 1.3

INF [0x0] TEE-CORE:init_primary_helper:387: Next entry point address: 0x200000

INF [0x0] TEE-CORE:init_teecore:83: teecore inits done
[    0.000000] Booting Linux on physical CPU 0x500
[    0.000000] Initializing cgroup subsys cpuset
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.4.194 (root@linaro-alip) (gcc version 6.3.0 20170516 (Debian 6.3.0-18+deb9u1) ) #108 SMP Mon Mar 7 15:20:40 +03 2022
[    0.000000] CPU: ARMv7 Processor [410fc0d1] revision 1 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: rockchip,rk3288w
[    0.000000] cma: Reserved 16 MiB at 0x7f000000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv65535.65535 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] PERCPU: Embedded 14 pages/cpu @eef79000 s24728 r8192 d24424 u57344
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 518896
[    0.000000] Kernel command line: storagemedia=emmc androidboot.storagemedia=emmc androidboot.mode=normal  androidboot.verifiedbootstate=orange androidboot.serialno=c3d9b8674f4b94f6  earlyprintk rw rootwait console=ttyFIQ0 root=PARTUUID=614e0000-0000 rootfstype=ext4
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 2030272K/2081728K available (10240K kernel code, 806K rwdata, 2892K rodata, 1024K init, 1189K bss, 35072K reserved, 16384K cma-reserved, 1294336K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc0b00000   (11232 kB)
[    0.000000]       .init : 0xc0f00000 - 0xc1000000   (1024 kB)
[    0.000000]       .data : 0xc1000000 - 0xc10c99a4   ( 807 kB)
[    0.000000]        .bss : 0xc10cb000 - 0xc11f4550   (1190 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[... 566 lines cut ...]
[   24.404592] [<c018a720>] (hrtimer_interrupt) from [<c0752af4>] (arch_timer_handler_phys+0x38/0x48)
[   24.413550] [<c0752af4>] (arch_timer_handler_phys) from [<c017ab70>] (handle_percpu_devid_irq+0xd4/0x1a8)
[   24.423111] [<c017ab70>] (handle_percpu_devid_irq) from [<c0176424>] (generic_handle_irq+0x28/0x38)
[   24.432153] [<c0176424>] (generic_handle_irq) from [<c0176780>] (__handle_domain_irq+0xb0/0xc4)
[   24.440840] [<c0176780>] (__handle_domain_irq) from [<c0101478>] (gic_handle_irq+0x60/0xa4)
[   24.449189] [<c0101478>] (gic_handle_irq) from [<c010c114>] (__irq_svc+0x54/0x70)
[   24.456656] Exception stack(0xee8d3dd8 to 0xee8d3e20)
[   24.461701] 3dc0:                                                       ee8d3e64 00000000
[   24.469867] 3de0: fd640800 c0752a44 ee8d3e64 00005dbf 00004bc8 00004c90 0000000d c10523d4
[   24.478035] 3e00: 066665b0 ee8d3e34 ee8d3e38 ee8d3e28 c010ee40 c0752a54 a0000013 ffffffff
[   24.486204] [<c010c114>] (__irq_svc) from [<c0752a54>] (arch_counter_get_cntpct+0x10/0x18)
[   24.494462] [<c0752a54>] (arch_counter_get_cntpct) from [<c010ee40>] (arch_timer_read_counter_long+0x20/0x28)
[   24.504370] [<c010ee40>] (arch_timer_read_counter_long) from [<c040a120>] (read_current_timer+0x30/0x48)
[   24.513846] [<c040a120>] (read_current_timer) from [<c040a170>] (__timer_delay+0x38/0x5c)
[   24.522018] [<c040a170>] (__timer_delay) from [<c040a1c0>] (__timer_const_udelay+0x2c/0x34)
[   24.530367] [<c040a1c0>] (__timer_const_udelay) from [<c01ea458>] (panic+0x1cc/0x200)
[   24.538199] [<c01ea458>] (panic) from [<c0f013c4>] (mount_block_root+0x1ec/0x288)
[   24.545672] [<c0f013c4>] (mount_block_root) from [<c0f015a8>] (mount_root+0x60/0x78)
[   24.553411] [<c0f015a8>] (mount_root) from [<c0f01744>] (prepare_namespace+0x184/0x1c4)
[   24.561406] [<c0f01744>] (prepare_namespace) from [<c0f00f7c>] (kernel_init_freeable+0x1f0/0x264)
[   24.570268] [<c0f00f7c>] (kernel_init_freeable) from [<c0a82fd0>] (kernel_init+0x18/0x118)
[   24.578524] [<c0a82fd0>] (kernel_init) from [<c01074e8>] (ret_from_fork+0x14/0x2c)
[   24.586087] ---[ end trace 359b25e9b726603b ]---
debug>
```

JeffyCN, 2022-03-24:
this log has drm driver bounded:
[    3.762761] rockchip-drm display-subsystem: bound ff940000.vop (ops 0xc0b65fb8)
[    3.770499] rockchip-drm display-subsystem: bound ff930000.vop (ops 0xc0b65fb8)
[    3.796916] rockchip-drm display-subsystem: bound ff980000.hdmi (ops 0xc0b5cea0)
[    3.804378] rockchip-drm display-subsystem: bound ff770000.syscon:lvds (ops 0xc0b5e64c)

and the panic means it failed to mount rootfs(partition 4) as ext4 fs.
so you can try to reflash the rootfs.img

JeffyCN, 2022-03-25:
ok

JeffyCN, 2022-03-25:
one more thing, since you are using rk3288w(according to the loader log), you might need this in the local config:
https://github.com/JeffyCN/yocto-conf/blob/master/rockchip-rk3288w-evb.conf#L3

closed 2022-03-25 by JeffyCN

JeffyCN, 2022-03-29:
first on all, you need to check the rockchip-drm display-subsystem: bound log to make sure it's a kernel issue.

if it's a kernel issue, there must be something different in kernel, then you'll have to learn how to debug kernel(usually by reading code and adding log)

if you want me to help checking it, please upload the booting log/image/.config/dts of those 2 kernel(prebuilt zboot.img and yocto kernel boot.img)

JeffyCN, 2022-03-29:
the image/.config/dts would be found in path like:
```
~/yocto/build/tmp/work/rockchip_rk3399_sapphire_excavator-poky-linux/linux-rockchip/4.4-r0/linux-rockchip_rk3399_sapphire_excavator-standard-build$ find arch/ -name "*dts.tmp"
arch/arm64/boot/dts/rockchip/.rk3399-sapphire-excavator-linux.dtb.dts.tmp
~/yocto/build/tmp/work/rockchip_rk3399_sapphire_excavator-poky-linux/linux-rockchip/4.4-r0/linux-rockchip_rk3399_sapphire_excavator-standard-build$ ls .config
.config
~/yocto/build/tmp/deploy/images/rockchip-rk3399-sapphire-excavator$ ls boot.img
boot.img
```

TYeniyayla, 2022-04-04:
Sorry, I'm currently working on usb not automounting issue. If you want, let's close it for now as we provide a temporary solution. Then when I come back to this problem we can turn it back on. Since the deathline of my project was approaching, I had to continue temporarily and change the order of priority.