JeffyCN/meta-rockchip #13: Mali DDK inconsisten on rk3568

jdica, 2022-03-31:
I'm trying to get hardware video acceleration for the rk3568 eval board. I can build an image just fine, but X11 is pretty sluggish as is video playback which also shows >80% CPU usage (using gstreamer to play 1080p at 30 FPS mp4 video).

After doing some digging I suspect the issue is with the Mali DDK versions in kernel and userspace:

Running `dmesg | grep mali` i get the following:
```
[    0.256150] mali fde60000.gpu: Kernel DDK version g7p1-01bet0
[    0.256240] mali fde60000.gpu: Looking up mali-supply from device tree
[    0.256274] mali fde60000.gpu: Device initialization Deferred
[    1.700872] mali fde60000.gpu: Kernel DDK version g7p1-01bet0
[    1.701048] mali fde60000.gpu: Looking up mali-supply from device tree
[    1.701171] mali fde60000.gpu: Linked as a consumer to regulator.18
[    1.701207] mali fde60000.gpu: Looking up shadercores-supply from device tree
[    1.701227] mali fde60000.gpu: Looking up shadercores-supply property in node /gpu@fde60000 failed
[    1.701340] mali fde60000.gpu: Looking up mali-supply from device tree
[    1.701412] mali fde60000.gpu: Looking up shadercores-supply from device tree
[    1.701427] mali fde60000.gpu: Looking up shadercores-supply property in node /gpu@fde60000 failed
[    1.701465] mali fde60000.gpu: dev_pm_opp_set_regulators: no regulator (shadercores) found: -19
[    1.701633] mali fde60000.gpu: leakage=10
[    1.702570] mali fde60000.gpu: avs=0
[    1.702615] W : [File] : drivers/gpu/arm/bifrost/platform/rk/mali_kbase_config_rk.c; [Line] : 112; [Func] : kbase_platform_rk_init(); power-off-delay-ms not available.
[    1.703303] mali fde60000.gpu: GPU identified as 0x2 arch 7.4.0 r1p0 status 0
[    1.703592] mali fde60000.gpu: No priority control manager is configured
[    1.703630] mali fde60000.gpu: No memory group manager is configured
[    1.704540] mali fde60000.gpu: l=-2147483648 h=2147483647 hyst=0 l_limit=0 h_limit=0 h_table=0
[    1.705957] mali fde60000.gpu: Probed as mali0
[    1.726487] I : [File] : drivers/gpu/arm/mali400/mali/linux/mali_kernel_linux.c; [Line] : 413; [Func] : mali_module_init(); svn_rev_string_from_arm of this mali_ko is '', rk_ko_ver is '5', built at '08:58:33', on 'Jan  6 2022'.
```
So it looks like the kernel is loading g7p1, however in userspace we have `/usr/lib/libmali-bifrost-g52-g2p0-x11.so` and when I run `eglinfo`:

```
EGL client extensions string:
    EGL_EXT_client_extensions EGL_EXT_platform_base
    EGL_KHR_client_get_all_proc_addresses EGL_KHR_platform_x11
    EGL_EXT_platform_x11 EGL_KHR_platform_gbm

GBM platform:
arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '4'.
eglinfo: eglInitialize failed

X11 platform:
EGL API version: 1.4
EGL vendor string: ARM
EGL version string: 1.4 Bifrost-"g2p0-01eac0"
EGL client APIs: OpenGL_ES
EGL extensions string:
    EGL_KHR_image_pixmap EGL_KHR_partial_update
    EGL_EXT_image_dma_buf_import EGL_KHR_config_attribs EGL_KHR_image
    EGL_KHR_image_base EGL_KHR_fence_sync EGL_KHR_wait_sync
    EGL_KHR_gl_colorspace EGL_KHR_get_all_proc_addresses
    EGL_IMG_context_priority EGL_KHR_no_config_context
    EGL_EXT_image_dma_buf_import EGL_EXT_image_dma_buf_import_modifiers
    EGL_EXT_yuv_surface EGL_EXT_pixel_format_float
    EGL_ARM_pixmap_multisample_discard EGL_ARM_implicit_external_sync
    EGL_KHR_gl_texture_2D_image EGL_KHR_gl_renderbuffer_image
    EGL_KHR_create_context EGL_KHR_surfaceless_context
    EGL_KHR_gl_texture_cubemap_image EGL_EXT_image_gl_colorspace
    EGL_EXT_create_context_robustness
Configurations:
     bf lv colorbuffer dp st  ms    vis   cav bi  renderable  supported
  id sz  l  r  g  b  a th cl ns b    id   eat nd gl es es2 vg surfaces 
---------------------------------------------------------------------
0x01 32  0  8  8  8  8  0  0  0 0 0x95TC      a     y  y     win,pb,pix
0x02 32  0  8  8  8  8 24  0  0 0 0x95TC      a     y  y     win,pb,pix
0x03 32  0  8  8  8  8 24  8  0 0 0x95TC      a     y  y     win,pb,pix
0x04 32  0  8  8  8  8 24  8  4 1 0x95TC      a     y  y     win,pb,pix
0x05 16  0  5  6  5  0  0  0  0 0 0x00--      y     y  y     pb,pix
0x06 16  0  5  6  5  0 24  0  0 0 0x00--      y     y  y     pb,pix
0x07 16  0  5  6  5  0 24  8  0 0 0x00--      y     y  y     pb,pix
0x08 16  0  5  6  5  0 24  8  4 1 0x00--      y     y  y     pb,pix
0x09 24  0  8  8  8  0  0  0  0 0 0x21TC      y     y  y     win,pb,pix
0x0a 24  0  8  8  8  0 24  8  0 0 0x21TC      y     y  y     win,pb,pix
0x0b 24  0  8  8  8  0  0  0  4 1 0x21TC      y     y  y     win,pb,pix
0x0c 24  0  8  8  8  0 24  8  4 1 0x21TC      y     y  y     win,pb,pix
0x0d 16  0  5  5  5  1 24  8  0 0 0x00--      a     y  y     pb,pix
0x0e 16  0  5  5  5  1 24  8  4 1 0x00--      a     y  y     pb,pix
0x0f 16  0  4  4  4  4 24  8  0 0 0x00--      a     y  y     pb,pix
0x10 16  0  4  4  4  4 24  8  4 1 0x00--      a     y  y     pb,pix
0x11 32  0  8  8  8  8 24  8  8 1 0x95TC      a     y  y     win,pb,pix
0x12 16  0  5  6  5  0 24  8  8 1 0x00--      y     y  y     pb,pix
0x13 24  0  8  8  8  0 24  8  8 1 0x21TC      y     y  y     win,pb,pix
0x14 32  0  8  8  8  8 24  8 16 1 0x95TC      a     y  y     win,pb,pix
0x15 16  0  5  6  5  0 24  8 16 1 0x00--      y     y  y     pb,pix
0x16 24  0  8  8  8  0 24  8 16 1 0x21TC      y     y  y     win,pb,pix
0x17 24  0  8  8  8  0  0  0  0 0 0x21TC            y  y     win,pb,pix
0x18 64  0 16 16 16 16 24  8  0 0 0x00--            y  y     pb
0x19 32  0 10 10 10  2 24  8  0 0 0x00--      a     y  y     pb,pix
0x1a 24  0  8  8  8  0  0  0  0 0 0x21TC      y     y  y     win,pb,pix
```
I am building with rockchip-rk3568-evb machine and Poky distro on Honister.
I have also added the following to my image:
```
DISTRO_FEATURES:remove:rockchip-rk3568-evb = "wayland"
DISTRO_FEATURES:append:rockchip-rk3568-evb = " x11 opengl"
IMAGE_FEATURES:append:rockchip-rk3568-evb = " x11-base ssh-server-openssh"

PACKAGECONFIG:append:pn-qtbase:rockchip-rk3568-evb = " examples"
PACKAGECONFIG:append:pn-gstreamer1.0-plugins-bad:rockchip-rk3568-evb = " kms"
PACKAGECONFIG:append:pn-glmark2:rockchip-rk3568-evb = " x11-gles2 drm-gles2"

CORE_IMAGE_EXTRA_INSTALL:append:rockchip-rk3568-evb = " \
    xf86-video-modesetting xserver-xorg-module-exa xserver-xorg-extension-glx udev-conf-rockchip \
    gstreamer1.0-rockchip gstreamer1.0-libav gstreamer1.0-omx \
    ${@' rockchip-rkisp-server rockchip-rkisp-iqfiles' if d.getVar('RK_ISP_VERSION') == '1' else ''} \
    ${@oe.utils.version_less_or_equal('RK_ISP_VERSION', '1', '', ' rockchip-rkaiq-server rockchip-rkaiq-iqfiles', d)} \
    libdrm-tests strace gdb e2fsprogs e2fsprogs-resize2fs io perf openssh-sftp-server mesa-demos \
    glmark2 \
    iw wpa-supplicant bluez5 \
    alsa-utils rockchip-alsa-config pulseaudio-server alsa-plugins-pulseaudio-conf \
    ${MACHINE_EXTRA_RRECOMMENDS} \
"
```

I am not familiar with Mali GPU so I may be missing something here, any info that can point me in the right direction would be great.

JeffyCN, 2022-03-31:
try gstreamer with kmssink/rkximagesink/xvimagesink

JeffyCN, 2022-03-31:
you can use glmark2-es2 to test gpu, but your log looks fine to me.

for video playback, the most important things would be hw decoding and reducing buf-copy(related to videosink).

jdica, 2022-03-31:
Thanks for the quick response. 

It looks like the issue isn't with the GPU after all. I ran glmark2-es2 and it looks fine. However, X and video playback is still slow. I was unable to run gstreamer as above, so I'll continue to dig into that and look more into X11.

So the difference in DDK versions shouldn't be an issue?

JeffyCN, 2022-03-31:
the driver's version is not bind to ddk, unless having api compatible issue, which might need some extra version checks in ddk(or driver).

for 1080p30fps, kmssink/rkximagesink/xvimagesink should be enough, and if you want 60fps, may need this to enable async commit(for kernel > 4.4):
https://github.com/JeffyCN/rockchip_mirrors/blob/buildroot/board/rockchip/common/base/etc/init.d/S10async_commit.sh#L12

maybe i can try to make a patch to force enabling it in yocto kernels

JeffyCN, 2022-03-31:
hmm, it looks like there are 2 issues here:
1/ the gst version bumped recently, so the patches are not included(patches are for 1.18.4).that's why xv is slow and kms could not open
2/ the kernel driver is using cluster hw plane as the first overlay plane, but it cannot support normal linear video formats, that's why rkximage failed to show

so i'll:
1/ bump the patches so xv should work
2/ hack the kernel to hide cluster planes

JeffyCN, 2022-04-01:
patches uploaded

jdica, 2022-04-01:
Thanks!

Running gstreamer with kmssink runs pretty smooth and low cpu usage with 1080p at 30fps.

closed 2022-04-01 by jdica

linxzhang, 2022-07-11:
i got the same issue, how did you fix it?