JeffyCN/meta-rockchip #15: [px30-evb-board] Problem with core minimal image from Yocto

hntu1995, 2022-04-14:
Hi, I am new in this field

So I cloned your repo and try to setup the enviroment as in readme.md (the only different is that I cannot clone meta-oe branch kirkstone and have to use the master branch) for yocto to create a minimal image to test it on my custom board. The build went smoothly but when I load the update.img to the board. Nothing happen on the uart debug port. So I do not know which part I did wrong and I just want to ask how do you proceed the build for the px30-evb-board. Is there anything special to config before or any more file to flash in to the board  beside update.img

Please help,

Best regards,

JeffyCN, 2022-04-14:
px30-evb-board is only for that px30 evb board.

if you are using other boards, check the last part of readme.md:
https://github.com/JeffyCN/meta-rockchip#v-supporting-new-machine

hntu1995, 2022-04-15:
Thank you for your answer. So I made a custom board from this board [px30-jd4-firefly](https://en.t-firefly.com/product/coreboard/core_px30_jd4) but I thought it was the px30-evb-board [px30-evb-board](https://www.alibaba.com/product-detail/motherboard-development-board-EVB-DEV-KIT_1600188127770.html). I will try to support the new board. Thank you very much.

JeffyCN, 2022-04-15:
the evb boards are rockchip's official reference boards. firefly is a custom of rockchip and providing their design of boards.

if you cannot get the source of loader(uboot) or kernel, you might still be able to use their prebuilt loader and kernel images(u-boot.img and boot.img) with the rest parts of this repo(mainly rootfs.img)

the way to flash a standalone image would be
upgrade_tool di -<part name> part.img
for example
upgrade_tool di -boot boot.img

closed 2022-04-15 by JeffyCN