JeffyCN/meta-rockchip #16: sdk image failed build package which depeend mali

SquallATF, 2022-05-25:
>/opt/poky/4.0.1/sysroots/x86_64-pokysdk-linux/usr/bin/aarch64-poky-linux/aarch64-poky-linux-ld: cannot find -lmali_hook_injector: No such file or directory

`libmali_hook_injector.a` packaged in `rockchip-libmali-staticdev`,  `rockchip-libmali-dev` required it but does not depend on it.

JeffyCN, 2022-05-25:
try:

+++ b/recipes-graphics/rockchip-libmali/rockchip-libmali.bb
@@ -96,5 +96,6 @@ FILES:${PN}:append = " \

 FILES:${PN}-dev = " \
        ${includedir} \
+       ${libdir}/lib*.a \
        ${libdir}/pkgconfig \
 "

SquallATF, 2022-05-26:
fixed https://github.com/JeffyCN/meta-rockchip/commit/c79677624ef1b0b4b8778e67e9be2c261e99e388

SquallATF, 2022-05-30:
After rebuild, I found that this patch doesn't work, `libmali_hook_injector. still packaged in `rockchip-libmali-staticdev`.
may be should add `TOOLCHAIN_TARGET_TASK:append = " rockchip-libmali-staticdev"` in config file.

JeffyCN, 2022-05-30:
hmm, that is because it(-staticdev) has the higher list order than -dev in the default PACKAGES list, that is not what i expected. (so it's out-dated somehow in https://www.yoctoproject.org/docs/1.1/poky-ref-manual/poky-ref-manual.html)

anyway, i've re-wrote the fix, please try.

the reason i don't want to use -staticdev here is that i think it's designed for static building, so instead of adding exceptions to every shared/static building logic(may not only the sdk building), i'd prefer this way ;)

SquallATF, 2022-05-30:
Thanks， new fix worked.

closed 2022-05-30 by SquallATF