JeffyCN/meta-rockchip #17: weston cannot resume after idle

SquallATF, 2022-05-27:
After weston resume from the idle black screen, only the background image is displayed, and the newly launched application is not displayed. but if set `idle-time` to 0, it seems to work fine all the time.

JeffyCN, 2022-05-27:
couldn't repro, it works well on my side(3588)

JeffyCN, 2022-05-27:
the weston supports using drm atomic api and legacy api(default is atomic), so maybe try legacy api:
export WESTON_DISABLE_ATOMIC=1

and maybe also check the kernel's dmesg for error logs and check /sys/kernel/debug/dri/0/summary for drm driver status

JeffyCN, 2022-05-30:
any luck? i found there might be a problem for 356x and 3588 when using atomic mode with kernel drm fbdev emulation enabled:
https://github.com/JeffyCN/meta-rockchip/commit/bd9b31e11b4f508464d124c88a8c789e3ca296d8

SquallATF, 2022-05-30:
> any luck? i found there might be a problem for 356x and 3588 when using atomic mode with kernel drm fbdev emulation enabled: [bd9b31e](https://github.com/JeffyCN/meta-rockchip/commit/bd9b31e11b4f508464d124c88a8c789e3ca296d8)
Unlucky
`export WESTON_DISABLE_ATOMIC=1` didn't help to solve my problem, and new patch apply failed.
```
ERROR: weston-10.0.0-r0 do_patch: Applying patch '0005-backend-drm-Bind-Nth-primary-plane-to-Nth-crtc.patch' on target directory '/home/squallatf/yocto/build/tmp/work/armv8a-poky-linux/weston/10.0.0-r0/weston-10.0.0'
CmdError('quilt --quiltrc /home/squallatf/yocto/build/tmp/work/armv8a-poky-linux/weston/10.0.0-r0/recipe-sysroot-native/etc/quiltrc push', 0, 'stdout: Applying patch 0005-backend-drm-Bind-Nth-primary-plane-to-Nth-crtc.patch
patching file libweston/backend-drm/drm.c
Hunk #1 FAILED at 842.
1 out of 1 hunk FAILED -- rejects in file libweston/backend-drm/drm.c
Patch 0005-backend-drm-Bind-Nth-primary-plane-to-Nth-crtc.patch does not apply (enforce with -f)

stderr: ')
ERROR: Logfile of failure stored in: /home/squallatf/yocto/build/tmp/work/armv8a-poky-linux/weston/10.0.0-r0/temp/log.do_patch.55784
ERROR: Task (/home/squallatf/yocto/build/../poky/meta/recipes-graphics/wayland/weston_10.0.0.bb:do_patch) failed with exit code '1'
```

JeffyCN, 2022-05-30:
patches re-uploaded.

did the app crashed? or just not shown?

SquallATF, 2022-05-30:
app will not shown and not crash, run `glmark2-es2-wayland` will show score in ssh terminal. and If restart weston, nothing will be displayed on screen.

dmesg 
```
[  813.641099] dw-mipi-dsi fe060000.dsi: [drm:dw_mipi_dsi_encoder_enable] final DSI-Link bandwidth: 420 x 4 Mbps
[  813.749044] rockchip-vop2 fe040000.vop: [drm:vop2_crtc_atomic_disable] Crtc atomic disable vp1
[  813.829554] rockchip-vop2 fe040000.vop: [drm:vop2_crtc_atomic_enable] Update mode to 800x1280p56, type: 16 for vp1
[  813.829964] dw-mipi-dsi fe060000.dsi: [drm:dw_mipi_dsi_encoder_enable] final DSI-Link bandwidth: 420 x 4 Mbps
```

JeffyCN, 2022-05-30:
nothing displayed or just background image? maybe try to check screenshot:
weston --debug --idle-time=2&
then:
weston-screenshooter

and if you are using kernel 4.19 or 5.10, it might be old(i've been warned not to sync the newest kernel versions)

SquallATF, 2022-05-30:
nothing displayed, and weston-screenshooter not return

my kernel based on 
```
commit 5bebb638902ab6ebe6a73a6628adb703a819e077
Author: Chen Shunqing <csq@rock-chips.com>
Date:   Thu Apr 14 09:04:15 2022 +0000
```

JeffyCN, 2022-05-30:
it looks like weston hang somewhere...maybe you can try to use gdb to check it, or "echo t > /proc/sysrq-trigger" to dump the stacks in kernel's dmesg

SquallATF, 2022-05-30:
gdb dump
```
Using host libthread_db library "/lib/libthread_db.so.1".
0x0000007f9f647644 in epoll_pwait () from /lib/libc.so.6
(gdb) info thread
  Id   Target Id                                       Frame
* 1    Thread 0x7f9f7a7460 (LWP 309) "weston"          0x0000007f9f647644 in epoll_pwait () from /lib/libc.so.6
  2    Thread 0x7f9b898f40 (LWP 311) "mali-mem-purge"  0x0000007f9f5dc74c in ?? () from /lib/libc.so.6
  3    Thread 0x7f9b088f40 (LWP 312) "mali-utility-wo" 0x0000007f9f5dc74c in ?? () from /lib/libc.so.6
  4    Thread 0x7f9a878f40 (LWP 313) "mali-utility-wo" 0x0000007f9f5dc74c in ?? () from /lib/libc.so.6
  5    Thread 0x7f9a068f40 (LWP 314) "mali-utility-wo" 0x0000007f9f5dc74c in ?? () from /lib/libc.so.6
  6    Thread 0x7f99858f40 (LWP 315) "mali-utility-wo" 0x0000007f9f5dc74c in ?? () from /lib/libc.so.6
  7    Thread 0x7f99048f40 (LWP 316) "mali-cmar-backe" 0x0000007f9f63da20 in poll () from /lib/libc.so.6
  8    Thread 0x7f986d2f40 (LWP 317) "weston"          0x0000007f9f5dc74c in ?? () from /lib/libc.so.6
(gdb) thread 3
[Switching to thread 3 (Thread 0x7f9b088f40 (LWP 312))]
#0  0x0000007f9f5dc74c in ?? () from /lib/libc.so.6
(gdb) bt
#0  0x0000007f9f5dc74c in ?? () from /lib/libc.so.6
#1  0x0000007f9f5e8310 in ?? () from /lib/libc.so.6
#2  0x0000007f9cffab0c in gles_vertexp_bb_neon_transform_and_produce_clip_bits () from /usr/lib/libmali.so.1
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) thread 4
[Switching to thread 4 (Thread 0x7f9a878f40 (LWP 313))]
#0  0x0000007f9f5dc74c in ?? () from /lib/libc.so.6
(gdb) bt
#0  0x0000007f9f5dc74c in ?? () from /lib/libc.so.6
#1  0x0000007f9f5e8310 in ?? () from /lib/libc.so.6
#2  0x0000007f9cffab0c in gles_vertexp_bb_neon_transform_and_produce_clip_bits () from /usr/lib/libmali.so.1
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) thread 5
[Switching to thread 5 (Thread 0x7f9a068f40 (LWP 314))]
#0  0x0000007f9f5dc74c in ?? () from /lib/libc.so.6
(gdb) bt
#0  0x0000007f9f5dc74c in ?? () from /lib/libc.so.6
#1  0x0000007f9f5e8310 in ?? () from /lib/libc.so.6
#2  0x0000007f9cffab0c in gles_vertexp_bb_neon_transform_and_produce_clip_bits () from /usr/lib/libmali.so.1
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) thread 6
[Switching to thread 6 (Thread 0x7f99858f40 (LWP 315))]
#0  0x0000007f9f5dc74c in ?? () from /lib/libc.so.6
(gdb) bt
#0  0x0000007f9f5dc74c in ?? () from /lib/libc.so.6
#1  0x0000007f9f5e8310 in ?? () from /lib/libc.so.6
#2  0x0000007f9cffab0c in gles_vertexp_bb_neon_transform_and_produce_clip_bits () from /usr/lib/libmali.so.1
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) thread 7
[Switching to thread 7 (Thread 0x7f99048f40 (LWP 316))]
#0  0x0000007f9f63da20 in poll () from /lib/libc.so.6
(gdb) bt
#0  0x0000007f9f63da20 in poll () from /lib/libc.so.6
#1  0x0000007f9cffad30 in gles_vertexp_bb_neon_transform_and_produce_clip_bits () from /usr/lib/libmali.so.1
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) thread 8
[Switching to thread 8 (Thread 0x7f986d2f40 (LWP 317))]
#0  0x0000007f9f5dc74c in ?? () from /lib/libc.so.6
(gdb) bt
#0  0x0000007f9f5dc74c in ?? () from /lib/libc.so.6
#1  0x0000007f9f5df308 in pthread_cond_wait () from /lib/libc.so.6
#2  0x0000007f9d046af0 in ?? () from /usr/lib/libmali.so.1
#3  0x0000007fe2dbecb8 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) thread 1
[Switching to thread 1 (Thread 0x7f9f7a7460 (LWP 309))]
#0  0x0000007f9f647644 in epoll_pwait () from /lib/libc.so.6
(gdb) bt
#0  0x0000007f9f647644 in epoll_pwait () from /lib/libc.so.6
#1  0x0000007f9c79d82c in wl_event_loop_dispatch () from /usr/lib/libwayland-server.so.0
#2  0x0000007f9c79ae98 in wl_display_run () from /usr/lib/libwayland-server.so.0
#3  0x0000007f9f72bf44 in wet_main (argc=<optimized out>, argv=0x7fe2dc1b98, test_data=0x0) at ../weston-10.0.0/compositor/main.c:3615
#4  0x0000007f9f58b230 in ?? () from /lib/libc.so.6
#5  0x0000007f9f58b30c in __libc_start_main () from /lib/libc.so.6
#6  0x000000556c1d7770 in ?? ()
(gdb)
```
kernel dump
[stack.txt](https://github.com/JeffyCN/meta-rockchip/files/8797234/stack.txt)

JeffyCN, 2022-05-30:
could be something wrong in the mali. maybe try to run without gpu:

weston --use-pixman --idle-time=2

SquallATF, 2022-05-31:
Same result without gpu.
[first time.log](https://github.com/JeffyCN/meta-rockchip/files/8804136/first.time.log)
at `[4264897.792] wl_callback@24.done(43764)` screen black and `[4277943.126]  -> wl_seat@11.capabilities(3)` mouse move

[second time.log](https://github.com/JeffyCN/meta-rockchip/files/8804140/second.time.log)

JeffyCN, 2022-05-31:
those logs look fine to me. so the weston seems fine(not sure why sreenshot hang).

maybe something wrong to the DSI driver or hardware? could you try to use HDMI?

root@RK3588:/# echo off > /sys/class/drm/card0-DSI-1/status
and you can dump drm driver status here:
root@RK3588:/# cat /sys/kernel/debug/dri/0/summary
root@RK3588:/# cat /sys/kernel/debug/dri/0/state

SquallATF, 2022-06-01:
If DSI is turned off, the screen connected to hdmi always black.
/sys/kernel/debug/dri/0/summary
```
Video Port0: ACTIVE
    Connector: HDMI-A-1
        bus_format[100a]: RGB888_1X24
        overlay_mode[0] output_mode[f] color_space[0]
    Display mode: 1920x1080p60
        clk[148500] real_clk[148500] type[48] flag[5]
        H: 1920 2008 2052 2200
        V: 1080 1084 1089 1125
    Smart1-win0: ACTIVE
        win_id: 1
        format: XR24 little-endian (0x34325258) SDR[0] color_space[0] glb_alpha[0xff]
        rotate: xmirror: 0 ymirror: 0 rotate_90: 0 rotate_270: 0
        csc: y2r[0] r2y[0] csc mode[0]
        zpos: 1
        src: pos[0, 0] rect[1920 x 1080]
        dst: pos[0, 0] rect[1920 x 1080]
        buf[0]: addr: 0x0000000001149000 pitch: 7680 offset: 0
Video Port1: ACTIVE
    Connector: DSI-1
        bus_format[100a]: RGB888_1X24
        overlay_mode[0] output_mode[0] color_space[0]
    Display mode: 800x1280p56
        clk[68000] real_clk[68000] type[48] flag[a]
        H: 800 872 892 912
        V: 1280 1310 1314 1326
    Smart0-win0: ACTIVE
        win_id: 0
        format: RG16 little-endian (0x36314752) SDR[0] color_space[0] glb_alpha[0xff]
        rotate: xmirror: 0 ymirror: 0 rotate_90: 0 rotate_270: 0
        csc: y2r[0] r2y[0] csc mode[0]
        zpos: 0
        src: pos[0, 0] rect[654 x 270]
        dst: pos[73, 505] rect[654 x 270]
        buf[0]: addr: 0x000000007df00000 pitch: 1308 offset: 397312
```
/sys/kernel/debug/dri/0/state
```
plane[57]: Smart1-win0
        crtc=video_port0
        fb=167
                allocated by = weston
                refcount=2
                format=XR24 little-endian (0x34325258)
                modifier=0x0
                size=1920x1080
                layers:
                        size[0]=1920x1080
                        pitch[0]=7680
                        offset[0]=0
                        obj[0]:(null)
        crtc-pos=1920x1080+0+0
        src-pos=1920.000000x1080.000000+0.000000+0.000000
        rotation=1
        normalized-zpos=0
        color-encoding=ITU-R BT.601 YCbCr
        color-range=YCbCr limited range
plane[73]: Smart0-win0
        crtc=video_port1
        fb=164
                allocated by = kworker/2:2
                refcount=1
                format=RG16 little-endian (0x36314752)
                modifier=0x0
                size=654x270
                layers:
                        size[0]=654x270
                        pitch[0]=1308
                        offset[0]=397312
                        obj[0]:(null)
        crtc-pos=654x270+73+505
        src-pos=654.000000x270.000000+0.000000+0.000000
        rotation=1
        normalized-zpos=0
        color-encoding=ITU-R BT.601 YCbCr
        color-range=YCbCr limited range
plane[89]: Esmart1-win0
        crtc=(null)
        fb=0
        crtc-pos=0x0+0+0
        src-pos=0.000000x0.000000+0.000000+0.000000
        rotation=1
        normalized-zpos=0
        color-encoding=ITU-R BT.601 YCbCr
        color-range=YCbCr limited range
plane[103]: Esmart0-win0
        crtc=(null)
        fb=0
        crtc-pos=0x0+0+0
        src-pos=0.000000x0.000000+0.000000+0.000000
        rotation=1
        normalized-zpos=0
        color-encoding=ITU-R BT.601 YCbCr
        color-range=YCbCr limited range
plane[117]: Cluster0-win0
        crtc=(null)
        fb=0
        crtc-pos=0x0+0+0
        src-pos=0.000000x0.000000+0.000000+0.000000
        rotation=1
        normalized-zpos=0
        color-encoding=ITU-R BT.601 YCbCr
        color-range=YCbCr limited range
plane[131]: Cluster1-win0
        crtc=(null)
        fb=0
        crtc-pos=0x0+0+0
        src-pos=0.000000x0.000000+0.000000+0.000000
        rotation=1
        normalized-zpos=0
        color-encoding=ITU-R BT.601 YCbCr
        color-range=YCbCr limited range
crtc[71]: video_port0
        enable=1
        active=1
        planes_changed=1
        mode_changed=0
        active_changed=0
        connectors_changed=0
        color_mgmt_changed=0
        plane_mask=1
        connector_mask=2
        encoder_mask=2
        mode: 0:"1920x1080" 60 148500 1920 2008 2052 2200 1080 1084 1089 1125 0x48 0x5
crtc[87]: video_port1
        enable=1
        active=1
        planes_changed=1
        mode_changed=0
        active_changed=0
        connectors_changed=0
        color_mgmt_changed=0
        plane_mask=2
        connector_mask=4
        encoder_mask=4
        mode: 0:"800x1280" 56 68000 800 872 892 912 1280 1310 1314 1326 0x48 0xa
connector[150]: Writeback-1
        crtc=(null)
connector[152]: HDMI-A-1
        crtc=video_port0
connector[163]: DSI-1
        crtc=video_port1
```

If connected bouth HDMI and DSI, the HDMI display is incorrect
[image]
wayland-screenshot before idle
[image]
wayland-screenshot after resume from idle (when hdmi connected, DSI screen keep black,but HDMI screen will resume like before.)
[image]

when connected HDMI, stop weston and restart, some time work but most time not work.

JeffyCN, 2022-06-01:
so it looks like a driver or hardware issue. (the drm driver state looks well)

[kmsgrab.zip](https://github.com/JeffyCN/meta-rockchip/files/8810576/kmsgrab.zip)

JeffyCN, 2022-06-01:
after disabling DSI and kernel logo(delete logo related nodes in the dts), the /sys/kernel/debug/dri/0/summary should be clean(only HDMI actived)
if the dumped HDMI content is correct too, but the display is still abnormal, then it's a hardware/driver issue

JeffyCN, 2022-06-01:
and try to disable the kernel's CONFIG_DRM_FBDEV_EMULATION too

SquallATF, 2022-06-02:
running `weston-screenshooter` does not hang with the latest version, but screen still black after idle. I think maybe there is some wrong with the screen initialization, because if run `echo 1 > /sys/class/graphics/fb0/blank` sleep screen before run weston has the same effect.

JeffyCN, 2022-06-02:
that's why i think this is a hardware/driver issue.
a few tests to confirm would be:
1/ disable fbdev emulation, to get a pure drm enviroment
2/ disable kernel logo, this logic is rockchip's custom hack, might cause lots of unexpected behaviors
3/ disable DSI to rule out DSI hardware/driver issue
4/ dump HDMI screen content and drm summary/state to confirm that userspace(weston) is doing nothing wrong

anyway, everything works well on my side(5.10 kernel+3588 evb)

SquallATF, 2022-06-02:
Thank you for your suggestion. In the next period of time, I will transfer the development board to other colleagues for application development. I cannot debug this problem for the time being, so I will close this issue first, and I will open it later if necessary.

closed 2022-06-02 by SquallATF

JeffyCN, 2022-06-02:
ok.

and if you can access to rockchip's redmineissue system, i would sugguest to report this issue there...because i don't know much about dsi driver or hardware