JeffyCN/meta-rockchip #18: Wrong rkwifibt firmware install path

SquallATF, 2022-06-01:
https://github.com/JeffyCN/mirrors/blob/82957dba3977fd50d4c013e0d359f3203072a0f2/drivers/net/wireless/rockchip_wlan/rkwifi/Kconfig#L39-L51
kernel default config firmware path is `/vendor/etc/firmware/`
but rkwifibt-firmware recipe installed to `/system/etc/firmware`
https://github.com/JeffyCN/meta-rockchip/blob/60b0b464bf936707fa31a16ad4e7f15fbe630da4/recipes-kernel/rkwifibt-firmware/rkwifibt-firmware.bb#L46-L113

JeffyCN, 2022-06-01:
the rkwifibt drivers are actually for android, so they used to use system/etc/firmware, but switched to vendor/etc/firmware recently.

so both of those path are needed, maybe try to add a symlink:

+++ b/recipes-kernel/rkwifibt-firmware/rkwifibt-firmware.bb
@@ -18,6 +18,9 @@ inherit allarch deploy

 do_install() {
        install -d ${D}/system/etc/firmware/
+       install -d ${D}/vendor/etc/
+       ln -sf /system/etc/firmware/ ${D}/vendor/etc/

JeffyCN, 2022-06-02:
patches uploaded

SquallATF, 2022-06-02:
fixed ddd47d6b57d035655bd78d405e3c7035a48837f9

closed 2022-06-02 by SquallATF