JeffyCN/meta-rockchip #19: Kirkstone meta-rockchip bitbake error

TYeniyayla, 2022-06-01:
Hi, I'm new to the kirkstone version. I have updated all the files but when I call `bitbake-layers show-layers` I get this error. What is the reason?
 
`ERROR: Variable BB_BASEHASH_IGNORE_VARS_append file: /media/tunahan/f1cceee6-ac24-4956-9e80-7ce648ac31e4/yoctoproject-kirkstone/meta-rockchip/conf/layer.conf line: 22 contains an operation using the old override syntax. Please convert this layer/metadata before attempting to use with a newer bitbake.`

The contents of the file mentioned in the error are as follows:
```# Copyright (c) 2019, Fuzhou Rockchip Electronics Co., Ltd
# Released under the MIT license (see COPYING.MIT for the terms)

# We have a conf and classes directory, add to BBPATH
BBPATH .= ":${LAYERDIR}"

# We have recipes-* directories, add to BBFILES
BBFILES += "${LAYERDIR}/recipes-*/*/*.bb \
	    ${LAYERDIR}/recipes-*/*/*.bbappend"

BBFILES_DYNAMIC += " \
	browser-layer:${LAYERDIR}/dynamic-layers/recipes-browser/*/*.bbappend \
	chromium-browser-layer:${LAYERDIR}/dynamic-layers/recipes-browser/*/*.bbappend \
	qt5-layer:${LAYERDIR}/dynamic-layers/recipes-qt/*/*.bbappend \
"

BBFILE_COLLECTIONS += "rockchip"
BBFILE_PATTERN_rockchip := "^${LAYERDIR}/"
BBFILE_PRIORITY_rockchip = "9"

RKBASE = '${@os.path.normpath("${LAYERDIR}")}'
BB_BASEHASH_IGNORE_VARS:append = " RKBASE"

# Additional license directories.
LICENSE_PATH += "${LAYERDIR}/licenses"

LAYERDEPENDS_rockchip = "core openembedded-layer"

# This should only be incremented on significant changes that will
# cause compatibility issues with other layers
LAYERVERSION_rockchip = "1"
LAYERSERIES_COMPAT_rockchip = "kirkstone"
```

JeffyCN, 2022-06-01:
i know nothing about "visual studio code", it looks like the bitbake tool still seeing the old version of meta-rockchip(have you done the git-checkout after git-fetch?).

BTW, this is the manifest i used for kirkstone: https://github.com/JeffyCN/yocto-manifests/tree/kirkstone-next

JeffyCN, 2022-06-01:
> How can I get bitbake to see the new version of meta-rockchip?
i have no idea about that, it was saying '/media/tunahan/f1cceee6-ac24-4956-9e80-7ce648ac31e4/yoctoproject-kirkstone/meta-rockchip/conf/layer.conf' is wrong, but you said it's not.

well, the wrong content must be stored somewhere, maybe you can try to search BB_BASEHASH_IGNORE_VARS_append in your disk

JeffyCN, 2022-06-01:
anyway, it doesn't look like to be a meta-rockchip issue, maybe you can try to use my manifest as a code base, and add your repositories

closed 2022-06-01 by JeffyCN