JeffyCN/meta-rockchip #20: qt 5.15 camera example get RGA_BLIT fail: Bad address

SquallATF, 2022-06-02:
official example https://doc.qt.io/qt-5/qtmultimedia-multimediawidgets-camera-example.html
when click `Capture Photo`  the console will display the following log, is this normal?
```
arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '3'.rga_api version 1.3.1_[11] (RGA is compiling with meson base: $PRODUCT_BASE)
10000 is unsupport format now,pilese fix.
10000 is unsupport format now,pilese fix.
10000 is unsupport format now,pilese fix.
10000 is unsupport format now,pilese fix.
10000 is unsupport format now,pilese fix.
10000 is unsupport format now,pilese fix.
 RgaBlit(1356) RGA_BLIT fail: Bad address RgaBlit(1357) RGA_BLIT fail: Bad address
fd-vir-phy-hnd-format[28, (nil), (nil), (nil), 0]
rect[0, 0, 1920, 1080, 3840, 1080, 65536, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
fd-vir-phy-hnd-format[38, (nil), (nil), (nil), 0]
rect[0, 0, 1920, 1080, 1920, 1088, 2560, 0]
f-blend-size-rotation-col-log-mmu[0, 0, 0, 0, 0, 0, 1]
This output the user patamaters when rga call blit fail
```

SquallATF, 2022-06-02:
It seems work after switching to the old version.

```
arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '3'.Rga built version:274b345f976a
librga:RGA_GET_VERSION:3.02,3.020000
ctx=0x7f98006df0,ctx->rgaFd=45
```

JoshuaJakowlew, 2023-08-11:
I have the same problem on 3568. Error log sin console, but everything seems to work right. Is it broken on 3568 too? Is there any instructions how to downgrade this lib?

JeffyCN, 2023-08-11:
don't know much about rga, please mail and ask the rga maintainer.

JeffyCN, 2025-05-27:
> > RGA 在最近的芯片中有点坏。在您的情况下，MPP Video Encode 插件可能涉及它。
> > 所以也许： 1/ 尝试将 librga 降级（从 IM2D 分支降级到它的 master 分支）或 2/ 在 gstreamer-rockchip 的 BB 中禁用 RGA（它可能会回退到使用慢软件转换）
> 
> 请问怎么在 gstreamer-rockchip 中禁用 RGA？

See the config at:
https://github.com/JeffyCN/meta-rockchip/blob/master/recipes-multimedia/gstreamer-rockchip/gstreamer1.0-rockchip.bb#L31