JeffyCN/meta-rockchip #23: Using own local u-boot and kernel sources

bojidartonchev, 2022-07-12:
Hello,

I am trying to build an image for ASUS TinkerBoard 2 (RK3399). So first I cloned the official [u-boot](https://github.com/TinkerBoard2/u-boot/tree/linux4.19-rk3399-debian10) and [kernel](https://github.com/TinkerBoard2/kernel/tree/linux4.19-rk3399-debian10) and I placed in the root poky folder (next to meta-rockchip). I created a new machine config in ./meta-rockchip/conf/machines, using the device tree and uboot from the new custom sources.

tinker-board-2.conf
```
#@TYPE: Machine
#@NAME: RK3399 Tinker Board 2

require conf/machine/include/rk3399.inc

KERNEL_DEVICETREE = "rockchip/rk3399-tinker_board_2.dtb"

UBOOT_MACHINE = "tinker_board_2_defconfig"

RK_WIFIBT_FIRMWARES = " \
        rkwifibt-firmware-ap6354-wifi \
        rkwifibt-firmware-ap6354-bt \
        brcm-tools \
"

MACHINE_EXTRA_RRECOMMENDS:append = " \
        linux-firmware-rk-cdndp \
"
```

Finally, I have added the override in my ./build/conf/local.conf

```
MACHINE ??= "tinker-board-2"
CORE_IMAGE_EXTRA_INSTALL += "kernel-modules tzdata"
IMAGE_FSTYPES = "wic.gz"
DISTRO_FEATURES += " systemd opengl x11"
DISTRO_FEATURES_BACKFILL_CONSIDERED += "sysvinit"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = "systemd-compat-units"

SRC_URI_pn-linux-rockchip = " \
        git://${TOPDIR}/../kernel;protocol=file;usehead=1 \
        file://cgroups.cfg \
"
SRCREV_pn-linux-rockchip = "${AUTOREV}"
KBRANCH = "HEAD"

SRC_URI_pn-u-boot = " \
        git://${TOPDIR}/../u-boot;protocol=file;usehead=1 \
"
SRCREV_pn-u-boot = "${AUTOREV}"
```

The result is that `bitbake core-image-minimal` fails with the following errors:

`make[3]: *** No rule to make target 'arch/arm64/boot/dts/rockchip/rk3399-tinker_board_2.dtb'.  Stop.`
`| *** Can't find default configuration "arch/../configs/tinker_board_2_defconfig"!`

Any hint on what am I doing wrong would be appreciated.

Best regards!

JeffyCN, 2022-07-13:
u-boot has multiple packges, not just 'u-boot'.

you can try to use dummy u-boot package

bojidartonchev, 2022-07-13:
I managed to get it working by:

1. .bbappend for the u-boot, appending the `tinker_board_2_defconfig` from official TinkerBoard2 [u-boot](https://github.com/TinkerBoard2/u-boot/tree/linux4.19-rk3399-debian10)
2. .bbappend for the kernel, appending the `rk3399-tinker_board_2.dts` and `rk3399-tinker_board_2.dtsi` from official TinkerBoard2 [kernel](https://github.com/TinkerBoard2/kernel/tree/linux4.19-rk3399-debian10)
3. Adding the new device tree to the build config, so a .dtb is generated
`KERNEL_DEVICETREE += "rockchip/rk3399-tinker_board_2.dtb"`

closed 2022-07-13 by bojidartonchev