JeffyCN/meta-rockchip #29: Flashing a wic image question

buelowp, 2022-08-19:
Hello, I had a lot of luck getting a Kirkstone image to build, but am having some issues flashing that image. I'm new to RK devices, so I may be missing something simple. As a side note, this isn't a board supported in your meta layers, but for the upgrade_tool operation, I'm not sure if that matters.

```
upgrade_tool -v
Not found config.ini
Program Data in /home/pete/bin
Upgrade Tool v1.43
```
Scan output
```
sudo ~/bin/upgrade_tool
Not found config.ini
Program Data in /home/pete/bin
List of rockusb connected
DevNo=1	Vid=0x2207,Pid=0x330c,LocationID=301	Mode=Maskrom
Found 1 rockusb,Select input DevNo,Rescan press <R>,Quit press <Q>:
```
But, when I try to flash the image
```
sudo ~/bin/upgrade_tool wl 0 core-image-minimal-rockchip-rk3399-icg-200-20220809000634.rootfs.wic
Not found config.ini
Program Data in /home/pete/bin
Write LBA failed!
```

I can't find any reference to the config.ini, but could get the Windows copy if that's important from the AndroidTool binary.

Any thoughts? I realize this is a bit outside your offering, and will understand if this isn't something you can support, but any help would be appreciated. Thank you.

JeffyCN, 2022-08-19:
please check:
https://github.com/JeffyCN/meta-rockchip#iii-booting-your-device

the tools said it's maskrom mode, so you may need to flash(db) a loader to get it into rockusb(download) mode.

JeffyCN, 2022-08-19:
and the tool with config.ini would be found:
https://github.com/JeffyCN/rockchip_mirrors/tree/tools/linux/Linux_Upgrade_Tool/Linux_Upgrade_Tool

normally it will automatically install that config.ini into ~/.config/upgrade_tool/

buelowp, 2022-08-19:
Ah yes, that worked. Thank you, the bootloader step was missing.

closed 2022-08-19 by buelowp