JeffyCN/meta-rockchip #30: Issue on core-image-minimal build do_populate_sysroot

B21227916, 2022-08-21:
While building the image I'm getting this error every time. Dependencies are installed and the issue is exists with use of both the supported and custom machines.

```
DEBUG: Executing python function sstate_task_prefunc
DEBUG: Python function sstate_task_prefunc finished
DEBUG: Executing python function do_populate_sysroot
DEBUG: Executing shell function sysroot_stage_all
0 blocks
DEBUG: Shell function sysroot_stage_all finished
DEBUG: Executing python function sysroot_strip
NOTE: Skipping file /home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/recipe-sysroot-native/usr/bin/boot_merger from rk-binary-native for already-stripped QA test
DEBUG: runstrip: ['strip', '--remove-section=.comment', '--remove-section=.note', '/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/sysroot-destdir/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/recipe-sysroot-native/usr/bin/trust_merger']
DEBUG: runstrip: ['strip', '--remove-section=.comment', '--remove-section=.note', '/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/sysroot-destdir/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/recipe-sysroot-native/usr/bin/loaderimage']
DEBUG: runstrip: ['strip', '--remove-section=.comment', '--remove-section=.note', '/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/sysroot-destdir/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/recipe-sysroot-native/usr/bin/rkImageMaker']
DEBUG: runstrip: ['strip', '--remove-section=.comment', '--remove-section=.note', '/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/sysroot-destdir/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/recipe-sysroot-native/usr/bin/afptool']
ERROR: Fatal errors occurred in subprocesses:
Command '['strip', '--remove-section=.comment', '--remove-section=.note', '/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/sysroot-destdir/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/recipe-sysroot-native/usr/bin/trust_merger']' returned non-zero exit status 1.
Subprocess output:strip: Unable to recognise the format of the input file `/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/sysroot-destdir/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/recipe-sysroot-native/usr/bin/trust_merger'
Command '['strip', '--remove-section=.comment', '--remove-section=.note', '/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/sysroot-destdir/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/recipe-sysroot-native/usr/bin/loaderimage']' returned non-zero exit status 1.
Subprocess output:strip: Unable to recognise the format of the input file `/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/sysroot-destdir/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/recipe-sysroot-native/usr/bin/loaderimage'
Command '['strip', '--remove-section=.comment', '--remove-section=.note', '/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/sysroot-destdir/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/recipe-sysroot-native/usr/bin/rkImageMaker']' returned non-zero exit status 1.
Subprocess output:strip: Unable to recognise the format of the input file `/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/sysroot-destdir/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/recipe-sysroot-native/usr/bin/rkImageMaker'
Command '['strip', '--remove-section=.comment', '--remove-section=.note', '/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/sysroot-destdir/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/recipe-sysroot-native/usr/bin/afptool']' returned non-zero exit status 1.
Subprocess output:strip: Unable to recognise the format of the input file `/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/sysroot-destdir/home/alperenacer/Desktop/poky/build/tmp/work/aarch64-linux/rk-binary-native/1.0-r0/recipe-sysroot-native/usr/bin/afptool'

DEBUG: Python function sysroot_strip finished
DEBUG: Python function do_populate_sysroot finished
```

JeffyCN, 2022-08-21:
well, it looks like your host's strip tool could not handle those prebuilt x86-64 static binaries.

if so, maybe try to remove:
https://github.com/JeffyCN/meta-rockchip/blob/master/conf/machine/include/rockchip-common.inc#L161
to
https://github.com/JeffyCN/meta-rockchip/blob/master/conf/machine/include/rockchip-common.inc#L216

then it would not try to generate rockchip style firmware(use wic image only)

B21227916, 2022-08-21:
Oh, okay. My host machine is M1 arm based, so this caused the issue. I’am going to try your suggestion. Thanks for such quick response.

JeffyCN, 2022-08-21:
well, sadly there's no arm64 version of those tools...

and there is an arm64 version of the flash tool(was used for rk3399pro to download firmware to it's npu(1808)):
https://github.com/JeffyCN/rockchip_mirrors/blob/rknpu-fw/bin/upgrade_tool

closed 2022-09-06 by JeffyCN