JeffyCN/meta-rockchip #31: Missing libgav1 ?

MastaG, 2022-08-23:
When trying to build chromium-ozone-wayland using your patchset, it seems to error out during configure stage:
```
ERROR Unresolved dependencies.
//media/gpu/v4l2:v4l2_stateless_decoder(//build/toolchain/yocto:yocto_target)
  needs //third_party/libgav1:libgav1(//build/toolchain/yocto:yocto_target)
```
I'm building with: ```use_v4l2_codec=true use_v4lplugin=true use_linux_v4l2_only=true```

Any ideas?

MastaG, 2022-08-23:
I'm trying to build version 103.0.5060.129-r0

JeffyCN, 2022-08-23:
didn't have this issue, maybe due to different configs, is vaapi enabled?

MastaG, 2022-08-23:
No it's not.
This is what I have:
```
use_cups=true
use_custom_libcxx=true
ffmpeg_branding="Chrome"
proprietary_codecs=true
use_v4l2_codec=true
use_v4lplugin=true
use_linux_v4l2_only=true
use_vaapi=false
is_component_build=false
use_gnome_keyring=false
use_kerberos=false
use_pulseaudio=true
use_system_freetype=false
enable_js_type_check=false
host_pkg_config="pkg-config-native"
is_debug=false
is_official_build=true
use_lld=true
use_gold=false
symbol_level=0
enable_remoting=false
enable_nacl=false
use_sysroot=false
treat_warnings_as_errors=false
is_cfi=false
disable_fieldtrial_testing_config=true
chrome_pgo_phase=0
google_api_key="x"
google_default_client_id="x"
google_default_client_secret="x"
custom_toolchain="//build/toolchain/yocto:yocto_target"
gold_path=""
host_toolchain="//build/toolchain/yocto:yocto_native"
is_clang=true
clang_base_path="x/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/103.0.5060.129-r0/recipe-sysroot-native/usr/bin"
clang_use_chrome_plugins=false
target_cpu="arm" 
v8_snapshot_toolchain="//build/toolchain/yocto:yocto_target"
max_jobs_per_link="32"
use_cups=true
use_custom_libcxx=true
ffmpeg_branding="Chrome"
proprietary_codecs=true
use_v4l2_codec=true
use_v4lplugin=true
use_linux_v4l2_only=true
use_vaapi=false
use_ozone=true
ozone_auto_platforms=false
ozone_platform_headless=true
ozone_platform_wayland=true
ozone_platform_x11=false
system_wayland_scanner_path="x/tmp/work/cortexa15t2hf-neon-vfpv4-oe-linux-gnueabi/chromium-ozone-wayland/103.0.5060.129-r0/recipe-sysroot-native/usr/bin/wayland-scanner"
use_system_wayland_scanner=true
use_xkbcommon=true
use_system_libwayland=true
use_system_minigbm=true
use_system_libdrm=true
arm_float_abi="hard"
arm_fpu="neon-vfpv4"
arm_tune="cortex-a15"
arm_version=7
enable_hangout_services_extension=true
enable_widevine=true
enable_mdns=true
arm_use_thumb=false
arm_use_neon=true
arm_optionally_use_neon=false
use_gtk=true
use_glib=true
use_gio=true
use_dbus=true
use_libpci=true
link_pulseaudio=true
rtc_use_pipewire=true
```

MastaG, 2022-08-23:
Yeah I think I will.
BUILD.gn states:
```
    if (use_v4l2_codec || use_vaapi) {
      data_deps += [
        "//components/chromeos_camera:jpeg_encode_accelerator_unittest",
        "//media/gpu:video_decode_accelerator_perf_tests",
        "//media/gpu:video_decode_accelerator_tests",
        "//media/gpu:video_encode_accelerator_perf_tests",
        "//media/gpu:video_encode_accelerator_tests",
        "//media/gpu/chromeos:image_processor_test",
      ]
      if (use_vaapi) {
        data_deps += [
          "//media/gpu/vaapi:decode_test",
          "//media/gpu/vaapi:vaapi_unittest",
        ]
      }
      if (use_v4l2_codec) {
        data_deps += [ "//media/gpu/v4l2:v4l2_stateless_decoder" ]
      }
    }
```
Then media/gpu/v4l2/BUILD.gn states:
```
executable("v4l2_stateless_decoder") {
  testonly = true
  sources = [
    "test/av1_decoder.cc",
    "test/av1_decoder.h",
    "test/v4l2_ioctl_shim.cc",
    "test/v4l2_ioctl_shim.h",
    "test/v4l2_stateless_decoder.cc",
    "test/video_decoder.cc",
    "test/video_decoder.h",
    "test/vp9_decoder.cc",
    "test/vp9_decoder.h",
  ]

  deps = [
    "//base",
    "//media",
    "//media:test_support",
    "//media/gpu:common",
    "//third_party/libgav1:libgav1",
  ]
}
```
So the library seems to be always included when ```use_v4l2_codec``` is set.

MastaG, 2022-08-23:
Ah I think I fixed it.
It's because I'm building for arm32 bit.

The library will get included as a build dependency when either ```enable_libgav1_decoder``` or ```use_libgav1_parser``` is being set.
See: third_party/libgav1/options.gni
```
# Copyright 2019 The Chromium Authors. All rights reserved.
# Use of this source code is governed by a BSD-style license that can be
# found in the LICENSE file.

import("//build/config/chromeos/ui_mode.gni")
import("//build/config/gclient_args.gni")

declare_args() {
  # Enable libgav1 decoder.
  # TODO(b/179825724): enable on LaCrOS.
  enable_libgav1_decoder =
      is_chromeos_ash && (target_cpu == "arm" || target_cpu == "arm64")
  use_libgav1_parser =
      (is_chromeos || is_linux || is_win) &&
      (target_cpu == "x86" || target_cpu == "x64" ||
      target_cpu == "arm64" || target_cpu == "ppc64")
}
```
Now ```use_libgav1_parser``` won't be set for arm 32bit and ```enable_libgav1_decoder``` is only set when building for ```is_chromeos_ash```.

So I just added the ```use_libgav1_parser``` flag and see what happens.

JeffyCN, 2022-08-23:
guessing newer chrome would have patch for it

MastaG, 2022-08-23:
yeah I will.
Setting ```use_libgav1_parser``` seems to do the trick for now.
Thanks bro!