JeffyCN/meta-rockchip #37: No rule to make target 'rk3288-evb-rk808-linux.img'

weiqj, 2022-11-28:
Yocto Kirkstone, linux-rockchip_5.10
MACHINE = "rockchip-rk3288-evb"
KERNEL_IMAGETYPE = "zImage"

*** No rule to make target 'rk3288-evb-rk808-linux.img'.  Stop.

Can any one share some insight?

JeffyCN, 2022-11-28:
only BSP kernel 4.4 supports 3288 now

closed 2022-11-28 by JeffyCN

weiqj, 2022-11-28:
> only BSP kernel 4.4 supports 3288 now

I know 4.19 works with RK3288 & Dunfell. Is there any plan to bump up the version?

JeffyCN, 2022-11-29:
> > only BSP kernel 4.4 supports 3288 now
> 
> I know 4.19 works with RK3288 & Dunfell. Is there any plan to bump up the version?

only 4.4 3288 sdk, others are not maintained. so even it happens to work, might have more issues.

i've saw some commits in internal 5.10 branch, people might starting to port it, but 1/ not sure about the progress 2/ i've been warned that 5.10 sdks are not allowed for the community, needs to buy official sdk license :(

weiqj, 2022-11-29:
@JeffyCN Thank you very much for your information. The decision to charge extra fees for "SDK license" sucks! I will think twice when I choose CPU vendors in the future.