JeffyCN/meta-rockchip #41: build failure with latest bsp u-boot/bootloader (ERROR: No python2)

tmm1, 2022-12-10:
I am trying to build for rk3566 android demo box. I was having issues with ethernet and dmc clock, but stock android image works fine. I discovered the android image is using newer bootloader firmwares which fixes the problems.

So I tried to build with newer u-boot:

```diff
diff --git a/recipes-bsp/u-boot/u-boot_%.bbappend b/recipes-bsp/u-boot/u-boot_%.bbappend
index 696913a..83a6bb0 100644
--- a/recipes-bsp/u-boot/u-boot_%.bbappend
+++ b/recipes-bsp/u-boot/u-boot_%.bbappend
@@ -12,8 +12,8 @@ LIC_FILES_CHKSUM = "file://Licenses/README;md5=a2c678cfd4a4d97135585cad908541c6"
 
 inherit freeze-rev local-git
 
-SRCREV = "e3ca3c3805cc60cc9e2fe2a4d78694907b49ee46"
-SRCREV_rkbin = "104659686b734ab041ef958c0abece1a250f48a4"
+SRCREV = "84a9a3d8d9206465a616a1e78deb73786b3146fb"
+SRCREV_rkbin = "5ed54e688429a29c3d427768a3e0d463605eea60"
 SRC_URI = " \
        git://github.com/JeffyCN/mirrors.git;protocol=https;branch=u-boot; \
        git://github.com/JeffyCN/mirrors.git;protocol=https;branch=rkbin;name=rkbin;destsuffix=rkbin; \
```

but I get build failures:

```
| make -f /build/tmp/work/rk3566_t95plus-poky-linux/u-boot/1_2017.09-r0/git/scripts/Makefile.build obj=lib/rsa| make -f /build/tmp/work/rk3566_t95plus-poky-linux/u-boot/1_2017.09-r0/git/scripts/Makefile.build obj=lib/zlib| make -f /build/tmp/work/rk3566_t95plus-poky-linux/u-boot/1_2017.09-r0/git/scripts/Makefile.build obj=examples| make -f /build/tmp/work/rk3566_t95plus-poky-linux/u-boot/1_2017.09-r0/git/scripts/Makefile.build obj=examples/standalone
| start=$(aarch64-poky-linux-nm u-boot | grep __rel_dyn_start | cut -f 1 -d ' '); end=$(aarch64-poky-linux-nm u-boot | grep __rel_dyn_end | cut -f 1 -d ' '); tools/relocate-rela u-boot-nodtb.bin 0x00a00000 $start $end
|   cat u-boot-nodtb.bin dts/dt.dtb > u-boot-dtb.bin
|   cp u-boot-dtb.bin u-boot.bin
|   truncate -s "%8" u-boot.bin
|   aarch64-poky-linux-objcopy --dump-section .rodata.default_environment=u-boot-initial-env env/common.o; sed --in-place -e 's/\x00/\x0A/g' u-boot-initial-env
| make[1]: Leaving directory '/build/tmp/work/rk3566_t95plus-poky-linux/u-boot/1_2017.09-r0/build'| make: Leaving directory '/build/tmp/work/rk3566_t95plus-poky-linux/u-boot/1_2017.09-r0/git'| ERROR: No python2| WARNING: exit code 1 from a shell command.ERROR: Task (/vagrant/poky/meta/recipes-bsp/u-boot/u-boot_2022.01.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 2868 tasks of which 2862 didn't need to be rerun and 1 failed.
Summary: 1 task failed:  /vagrant/poky/meta/recipes-bsp/u-boot/u-boot_2022.01.bb:do_compile
Summary: There was 1 ERROR message, returning a non-zero exit code.
```

tmm1, 2022-12-10:
This can fix it. I can send a PR if you would like:

```diff
@@ -40,7 +40,7 @@ do_configure:prepend() {
                ${S}/arch/arm/mach-rockchip/make_fit_atf.py
 
        # Remove unneeded stages from make.sh
-       sed -i -e '/^select_tool/d' -e '/^clean/d' -e '/^\t*make/d' ${S}/make.sh
+       sed -i -e '/^select_tool/d' -e '/^clean/d' -e '/^\t*make/d' -e '/which python2/{n;n;s/exit 1/true/}' ${S}/make.sh
 
        if [ "x${RK_ALLOW_PREBUILT_UBOOT}" = "x1" ]; then
                # Copy prebuilt images
```

The result is:

```diff
--- a/make.sh                                                                                                              
+++ b/make.sh                                                                                                              
@@ -723,7 +723,7 @@ function pack_fit_image()                                                                              
        elif [ "${ARM64_TRUSTZONE}" == "y" ]; then                                                                         
                if ! which python2 >/dev/null 2>&1 ; then                                                                  
                        echo "ERROR: No python2"                                                                           
-                       exit 1                                                                                             
+                       true                                                                                               
                fi                                                                                                         
        fi                                                                                                                 
  
```

JeffyCN, 2022-12-11:
thanks for the patches:)

closed 2022-12-11 by JeffyCN

tmm1, 2022-12-11:
Thanks for merging!

BTW, on the stock android image for the rk3566 box I have, it is using u-boot 2017.10

Is there 2017.10 BSP u-boot tree somewhere? Or maybe it's custom for this android image.

JeffyCN, 2022-12-11:
i think rockchip only use 2017.09 bsp u-boot(for both android and linux)

so if you got a 2017.10, could be patched for some customers only.

PR #42 (2022-12-11): u-boot: fixup make.sh python2 check
fixes #41