JeffyCN/meta-rockchip #43: trust.img support?

tmm1, 2022-12-12:
In my yocto wic image, I notice there is no trust.img partition

During boot I see this error:

```
[    0.916639] rockchip-dmc dmc: Looking up center-supply from device tree
[    0.916705] rockchip-dmc dmc: Linked as a consumer to regulator.6
[    0.916735] rockchip-dmc dmc: current ATF version 0x101 (a0=0x0)!
[    0.916756] ERROR: could not get clock /dmc:dmc_clk(0)
[    0.916767] rockchip-dmc dmc: Cannot get the clk dmc_clk. If using SCMI, trusted firmware need update to V1.01 and above.
[    0.916822] rockchip-dmc dmc: Dropping the link to regulator.6
[    0.916860] rockchip-dmc: probe of dmc failed with error -2
```

Is there a way to select between TPL/SPL and idbLoader boot modes?

JeffyCN, 2022-12-13:
i think that is defined somewhere in uboot's ini and defconfig.

according to the log, your trust firmware is too old, that means the rkbin needs to update (check the mirror repo)

tmm1, 2022-12-13:
I did update rkbin yesterday with https://github.com/JeffyCN/meta-rockchip/issues/41#issue-1488970101, by changing `SRCREV_rkbin`

Now I see "current ATF version 0x102 (a0=0x0)!" with my logging. But `dmc_clk` is still not available when `rockchip_dmcfreq_probe` calls `rockchip_dmcfreq_power_control`. I thought this is related to trust.img but I'm not sure.

```patch
diff --git a/drivers/devfreq/rockchip_dmc.c b/drivers/devfreq/rockchip_dmc.c
index d75c21e51854..ab3efe46ea69 100644
--- a/drivers/devfreq/rockchip_dmc.c
+++ b/drivers/devfreq/rockchip_dmc.c
@@ -2839,6 +2839,7 @@ static int rockchip_dmcfreq_get_event(struct rockchip_dmcfreq *dmcfreq)
 
 static int rockchip_dmcfreq_power_control(struct rockchip_dmcfreq *dmcfreq)
 {
+	struct arm_smccc_res res;
 	struct device *dev = dmcfreq->dev;
 
 	dmcfreq->vdd_center = devm_regulator_get_optional(dev, "center");
@@ -2847,6 +2848,10 @@ static int rockchip_dmcfreq_power_control(struct rockchip_dmcfreq *dmcfreq)
 		return PTR_ERR(dmcfreq->vdd_center);
 	}
 
+	res = sip_smc_dram(0, 0,
+			   ROCKCHIP_SIP_CONFIG_DRAM_GET_VERSION);
+	dev_notice(dev, "current ATF version 0x%lx (a0=0x%lx)!\n", res.a1, res.a0);
+
 	dmcfreq->dmc_clk = devm_clk_get(dev, "dmc_clk");
 	if (IS_ERR(dmcfreq->dmc_clk)) {
 		dev_err(dev, "Cannot get the clk dmc_clk. If using SCMI, trusted firmware need update to V1.01 and above.\n");
```

tmm1, 2022-12-13:
I used:

```
UBOOT_MACHINE = "rk3568_defconfig"
```

So maybe that is the problem? I can try to remove `CONFIG_ANDROID_*` and maybe it also needs custom DTS instead of `CONFIG_DEFAULT_DEVICE_TREE="rk3568-evb"`

JeffyCN, 2022-12-13:
3568 is using scmi_clk to provide dmc_clk in rk3568.dtsi, so that would related to trusted firmware and arm,scmi-smc driver.
if trusted firmware is correct, that must be something wrong in the kernel side ?

drivers/firmware/arm_scmi/clock.c

my evb has these:
```
root@RK356X:/sys/bus/scmi_protocol/drivers/scmi-clocks/scmi_dev.1# cat uevent
DRIVER=scmi-clocks
OF_NAME=protocol
OF_FULLNAME=/firmware/scmi/protocol@14
OF_COMPATIBLE_N=0
```

and rk356x supports   bsp kernel 4.19 & 5.10, maybe try 4.19 kernel?

tmm1, 2022-12-15:
Thanks. I am using 4.19.232 and I see the same scmi uevent details as you.

I see in u-boot there are a lot of options available for different types of images, but I'm not sure which one is most easy. Do you recommend I switch to RKFW mode for make.sh?

```
# in make.sh

PLAT_TYPE="RKFW" # default

    if grep  -q '^CONFIG_ROCKCHIP_FIT_IMAGE_PACK=y' .config ; then
        PLAT_TYPE="FIT"
    elif grep  -q '^CONFIG_SPL_DECOMP_HEADER=y' .config ; then
        PLAT_TYPE="DECOMP"
    fi

function pack_images()
{
    if [ "${ARG_RAW_COMPILE}" != "y" ]; then
        if [ "${PLAT_TYPE}" == "FIT" ]; then
            pack_fit_image ${ARG_LIST_FIT}
        elif [ "${PLAT_TYPE}" == "DECOMP" ]; then
            ${SCRIPT_DECOMP} ${ARG_LIST_FIT} --chip ${RKCHIP_LABEL}
        else
            pack_uboot_image
            pack_trust_image
            pack_loader_image
```

Currently in the defconfig it forces FIT mode:

```
configs/rk3568_defconfig:CONFIG_ROCKCHIP_FIT_IMAGE=y
```

It seems to make trust work in FIT mode, I need to add variables like:

```
CONFIG_TRUST_NUM=
CONFIG_TRUST_RSA_MODE=
CONFIG_TRUST_SHA_MODE=
CONFIG_TRUST_SIZE_KB=
CONFIG_UBOOT_NUM=
CONFIG_UBOOT_SIZE_KB=
```

But I'm not sure what are the correct values or what these variables mean.

tmm1, 2022-12-15:
Thank you, I really appreciate your help.

I checked the DTS more and it seemed the clock is not setup correctly. When I fixed the dmc_clk, it started working as expected.

So trust.img is not required and that was just a red herring.

closed 2022-12-15 by tmm1