JeffyCN/meta-rockchip #45: RK3568 and kernel 5.10

chrisiberg, 2022-12-27:
What is needed to raise the kernel version vor RK3568 to 5.10?
I tried it with just changing in the rk356x.inc, bitbake is finishing the image, but this does not boot, with 4.19 it does.
It stops at the Rockchip logo (wihout kernel text on it).
I want to use yocto with NanoPi R5C, it has RTL 8125 LAN chipset which is not supported in 4.19 but 5.10.
Can you give me a hint, what has to be changed too?

JeffyCN, 2022-12-27:
the internal bsp 5.10 kernel already support 356x, but i've been warned not to provide newer version for 5.10 here(due to company policy).

so maybe try to contact rockchip FAE for newest kernel...

tmm1, 2023-01-01:
I was able to build working kernel with this layer:

```
$ cat meta-local/recipes-kernel/linux/linux-rockchip_5.10.bbappend
SRC_URI = " \
    git://github.com/rockchip-linux/kernel.git;protocol=https;nobranch=1;branch=develop-5.10; \
"
SRCREV = "${AUTOREV}"
```

chrisiberg, 2023-01-02:
It does not work for me.
Is it possible, that only the 4.19 <IMAGE>.wic is booting from SD-Card?
@tmm1 did you use your own fork and was the "linux-rockchip_5.10.bbappend" the only mod?

JeffyCN, 2023-01-05:
those are u-boot logs...
the u-boot's vop display driver might use kernel's dts, so maybe:
1/ something wrong in your u-boot
or
2/ something wrong in your kernel 5.10's dts(related to vop display driver)

Shaggy013, 2023-01-08:
All that kernel naming ;)  , and so many flavors now 4.19, 5.10 and  mainline (without npu) .

Nanopi has already a vendor rockchip 5.10 kernel for their rk356x boards ;)  only indeed not a yocto in their sdfuse-rk3568 , ah well here is a start ;) 

meta-local/recipes-kernel/linux/linux-rockchip_5.10.bbappend

SRC_URI = " \
    git://github.com/friendlyarm/kernel-rockchip.git;protocol=https;nobranch=1;branch=nanopi5-v5.10.y_opt; \
"
SRCREV = "${AUTOREV}"

and  conf/machine/include/rk356x.inc

PREFERRED_VERSION_linux-rockchip ?= "5.10%"
LINUXLIBCVERSION ?= "5.10-custom%"

and when building in SDk rename kernel folder so it wont use local for building kernel 

and watch wich mali version is needed for kernel 

ow and choose the right dts .

rk3568-nanopi5-rev01.dts is for r5s
rk3568-nanopi5-rev02.dts and rk3568-nanopi5-rev03.dts is for r5c

and if uboot keeps giving trouble you can replace it with the vendor rockchip uboot

https://github.com/friendlyarm/uboot-rockchip.git
SRCREV= "8c91d15ff7d106d0da37c23727b387c2d6883e8b"

ow and uboot defconfig then nanopi5_defconfig

chrisiberg, 2023-01-09:
@Shaggy013 
Thnx a lot!

I only had to patch` rk3568-nanopi5-common.dtsi` for the correct rootfs

`bootargs = "earlycon=uart8250,mmio32,0xfe660000 console=ttyFIQ0 coherent_pool=1m root=PARTUUID=614e0000-0000 rw rootwait";`

Now I stuck at the correct mali version vor 5.10
g2p0 seems not to work, how can I find out, and where are the correct sources?

chrisiberg, 2023-01-10:
@JeffyCN 

What is all needed for a hw accelerated chromium in my local.conf?

JeffyCN, 2023-01-10:
check:
https://github.com/JeffyCN/yocto-conf
https://github.com/JeffyCN/yocto-manifests

JeffyCN, 2023-01-11:
try:
https://github.com/glmark2/glmark2/pull/177

chrisiberg, 2023-01-13:
Chromium is unstable, closing randomly without any useable error, does anybody have an idea?

JeffyCN, 2023-02-17:
chromium on wayland is unstable for years, but it's getting better:)

closed 2023-02-17 by JeffyCN