JeffyCN/meta-rockchip #46: rk3328-evb u-boot Build break

jwinarske, 2023-01-19:
```
|   aarch64-poky-linux-gcc  --sysroot=/__w/meta-flutter/kirkstone-rockchip-rk3328/rockchip-rk3328-evb/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot/1_2017.09-r0/recipe-sysroot -Wp,-MD,spl/common/.image-android-dt.o.d -nostdinc -isystem /__w/meta-flutter/kirkstone-rockchip-rk3328/rockchip-rk3328-evb/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot/1_2017.09-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../lib/aarch64-poky-linux/gcc/aarch64-poky-linux/11.3.0/include -Iinclude  -I/__w/meta-flutter/kirkstone-rockchip-rk3328/rockchip-rk3328-evb/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot/1_2017.09-r0/git/include  -I/__w/meta-flutter/kirkstone-rockchip-rk3328/rockchip-rk3328-evb/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot/1_2017.09-r0/git/arch/arm/include -include /__w/meta-flutter/kirkstone-rockchip-rk3328/rockchip-rk3328-evb/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot/1_2017.09-r0/git/include/linux/kconfig.h  -I/__w/meta-flutter/kirkstone-rockchip-rk3328/rockchip-rk3328-evb/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot/1_2017.09-r0/git/spl/common -Ispl/common -D__KERNEL__ -D__UBOOT__ -DCONFIG_SPL_BUILD -Wall -Wstrict-prototypes -Wno-format-security -fno-builtin -ffreestanding -fshort-wchar -Os -fno-stack-protector -fno-delete-null-pointer-checks -g -fstack-usage -Wno-format-nonliteral -Wno-address-of-packed-member -Werror=date-time -ffunction-sections -fdata-sections -D__ARM__ -mstrict-align -ffunction-sections -fdata-sections -fno-common -ffixed-r9 -fno-common -ffixed-x18 -pipe -march=armv8-a+nosimd -D__LINUX_ARM_ARCH__=8   -I/__w/meta-flutter/kirkstone-rockchip-rk3328/rockchip-rk3328-evb/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot/1_2017.09-r0/git/arch/arm/mach-rockchip/include    -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(image_android_dt)"  -D"KBUILD_MODNAME=KBUILD_STR(image_android_dt)" -c -o spl/common/image-android-dt.o /__w/meta-flutter/kirkstone-rockchip-rk3328/rockchip-rk3328-evb/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot/1_2017.09-r0/git/common/image-android-dt.c
|   File "/__w/meta-flutter/kirkstone-rockchip-rk3328/rockchip-rk3328-evb/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot/1_2017.09-r0/git/tools/dtoc/dtoc", line 50
|     print result
|     ^^^^^^^^^^^^
| SyntaxError: Missing parentheses in call to 'print'. Did you mean print(...)?
```

jwinarske, 2023-01-19:
Looks like this machine is behind a bit.  Per the wiki aarch64 can be used, as well as upstream u-boot.

https://opensource.rock-chips.com/wiki_U-Boot

I'll take a look at this.

JeffyCN, 2023-01-24:
maybe try to hack it like:
--- a/recipes-bsp/u-boot/u-boot_%.bbappend
+++ b/recipes-bsp/u-boot/u-boot_%.bbappend
@@ -37,7 +37,8 @@ do_configure:prepend() {
        sed -i -e 's/\(open(.*[^"]\))/\1, "rb")/' -e 's/,$//' \
                -e 's/print >> \([^,]*\), *\(.*\)$/print(\2, file=\1)/' \
                -e 's/print \(.*\)$/print(\1)/' \
-               ${S}/arch/arm/mach-rockchip/make_fit_atf.py
+               ${S}/arch/arm/mach-rockchip/make_fit_atf.py \
+               ${S}/tools/dtoc/dtoc

or use u-boot-dummy

ballestrerod, 2023-01-25:
hi, I've got the same problem. after your patch build fails on this:

```
|   File "/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/tools/dtoc/dtoc", line 60
|     parser.add_option('-d', '--dtb-file', action='store'
|                                                  ^^^^^^^
| SyntaxError: invalid syntax. Perhaps you forgot a comma?

```

JeffyCN, 2023-01-25:
this should work:
```
--- a/recipes-bsp/u-boot/u-boot_%.bbappend
+++ b/recipes-bsp/u-boot/u-boot_%.bbappend
@@ -34,10 +34,11 @@ do_configure:prepend() {
        done

        # Support python3
-       sed -i -e 's/\(open(.*[^"]\))/\1, "rb")/' -e 's/,$//' \
-               -e 's/print >> \([^,]*\), *\(.*\)$/print(\2, file=\1)/' \
+       sed -i -e 's/\(open(.*[^"]\))/\1, "rb")/' \
+               -e 's/print >> \([^,]*\), *\(.*\),*$/print(\2, file=\1)/' \
                -e 's/print \(.*\)$/print(\1)/' \
-               ${S}/arch/arm/mach-rockchip/make_fit_atf.py
+               ${S}/arch/arm/mach-rockchip/make_fit_atf.py \
+               ${S}/tools/dtoc/dtoc
```

ballestrerod, 2023-01-25:
thank you very much. now compilation goes forward but stops on:

```
| Traceback (most recent call last):
|   File "/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/tools/dtoc/dtoc", line 75, in <module>
|     dtb_platdata.run_steps(args, options.dtb_file, options.include_disabled,
|   File "/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/tools/dtoc/dtb_platdata.py", line 558, in run_steps
|     plat.scan_dtb()
|   File "/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/tools/dtoc/dtb_platdata.py", line 246, in scan_dtb
|     self._fdt = fdt.FdtScan(self._dtb_fname)
|   File "/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/tools/dtoc/fdt.py", line 413, in FdtScan
|     dtb = Fdt(fname)
|   File "/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/tools/dtoc/fdt.py", line 275, in __init__
|     self._fdt = bytearray(fd.read())
|   File "/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/recipe-sysroot-native/usr/lib/python3.10/codecs.py", line 322, in decode
|     (result, consumed) = self._buffer_decode(data, self.errors, final)
| UnicodeDecodeError: 'utf-8' codec can't decode byte 0xd0 in position 0: invalid continuation byte
| make -f /home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/scripts/Makefile.build obj=spl/drivers/serial
| make -f /home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/scripts/Makefile.build obj=spl/lib/avb/rk_avb_user
| make[2]: *** [/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/scripts/Makefile.spl:278: include/generated/dt-structs-gen.h] Error 1
| make[2]: *** Waiting for unfinished jobs....

```

JeffyCN, 2023-01-26:
so this chip is building spl, which needs to parse dtb using dtoc.

all scripts of it need to hack
```
--- a/recipes-bsp/u-boot/u-boot_%.bbappend
+++ b/recipes-bsp/u-boot/u-boot_%.bbappend
@@ -34,10 +34,11 @@ do_configure:prepend() {
        done

        # Support python3
-       sed -i -e 's/\(open(.*[^"]\))/\1, "rb")/' -e 's/,$//' \
-               -e 's/print >> \([^,]*\), *\(.*\)$/print(\2, file=\1)/' \
+       sed -i -e 's/\(open([^,]*\))/\1, "rb")/' \
+               -e 's/print >> \([^,]*\), *\(.*\),*$/print(\2, file=\1)/' \
                -e 's/print \(.*\)$/print(\1)/' \
-               ${S}/arch/arm/mach-rockchip/make_fit_atf.py
+               ${S}/arch/arm/mach-rockchip/make_fit_atf.py \
+               ${S}/tools/dtoc/*.py
```

ballestrerod, 2023-01-26:
no way. sorry if I'm bothering you but that build keep failing. 
maybe is the wrong u-boot version or too old for kirkstone? I'm trying to build for a Rockchip RK3328 on an Olimex SOM board.

```
| Traceback (most recent call last):
|   File "/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/tools/dtoc/dtoc", line 75, in <module>
|     dtb_platdata.run_steps(args, options.dtb_file, options.include_disabled,
|   File "/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/tools/dtoc/dtb_platdata.py", line 558, in run_steps
|     plat.scan_dtb()
|   File "/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/tools/dtoc/dtb_platdata.py", line 246, in scan_dtb
|     self._fdt = fdt.FdtScan(self._dtb_fname)
|   File "/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/tools/dtoc/fdt.py", line 414, in FdtScan
|     dtb.Scan()
|   File "/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/tools/dtoc/fdt.py", line 289, in Scan
|     self._root.Scan()
|   File "/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/tools/dtoc/fdt.py", line 228, in Scan
|     node.Scan()
|   File "/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/tools/dtoc/fdt.py", line 217, in Scan
|     val = fdt_util.fdt32_to_cpu(phandle.value)
|   File "/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/tools/dtoc/fdt_util.py", line 29, in fdt32_to_cpu
|     val = val.encode('raw_unicode_escape')
| AttributeError: 'list' object has no attribute 'encode'
| make[2]: *** [/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/scripts/Makefile.spl:278: include/generated/dt-structs-gen.h] Error 1
| make[1]: *** [/home/ballest/rockhevc_workspace/oses/rockhevc-kirkstone/build_rockhevc/tmp/work/rockhevc-poky-linux/u-boot/1_2017.09-r0/git/Makefile:1446: tpl/u-boot-tpl.bin] Error 2
| make[1]: *** Waiting for unfinished jobs....

```

jwinarske, 2023-01-26:
@ballestrerod that was also my impression; old u-boot.

@JeffyCN Any reason you aren't keeping u-boot the same for all the machine types?  Also what is the point of using the mirror repo, opposed to using origin repo?

JeffyCN, 2023-01-27:
the u-boot is bsp's 2017, which is using python2. but yocto switched to python3, so we need to hack it's scripts to python3, or use other u-boot.

3328's u-boot is different with other chips, it has spl stage, so calls more scripts.

for this error, need this u-boot commit:
```
commit a90df2b17282c43600284a92bf034d81d0c49444
Author: Simon Glass <sjg@chromium.org>
Date:   Thu Oct 31 07:43:04 2019 -0600
    dtoc: Convert fdt.py to Python 3
```

or use dummy u-boot and flash your old u-boot images instead of the generated ones:
PREFERRED_PROVIDER_virtual/bootloader ?= "u-boot-dummy"
PREFERRED_PROVIDER_u-boot ?= "u-boot-dummy

JeffyCN, 2023-01-31:
FYI, i've uploaded a patch for it

ballestrerod, 2023-02-02:
great job! everything works now.

closed 2023-02-17 by JeffyCN

jwinarske, 2023-02-18:
@JeffyCN Thanks!

ogkita, 2024-12-26:
HI @JeffyCN , Thank you very much for the work you do.

I am trying to generate an image for my 3328 from your git and I am getting this error when generating u-boot:

Traceback (most recent call last):
  File "/datos/dusun/sdk/jeffyCN.scarthgap/build/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot-rockchip/2017.09/git/scripts/dtc/pylibfdt/setup.py", line 115, in <module>
    setup(
  File "/datos/dusun/sdk/jeffyCN.scarthgap/build/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot-rockchip/2017.09/recipe-sysroot-native/usr/lib/python3.12/site-packages/setuptools/_distutils/core.py", line 146, in setup
    _setup_distribution = dist = klass(attrs)

                                 ^^^^^^^^^^^^
  File "/datos/dusun/sdk/jeffyCN.scarthgap/build/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot-rockchip/2017.09/recipe-sysroot-native/usr/lib/python3.12/site-packages/setuptools/dist.py", line 293, in __init__

    self.metadata.version = self._normalize_version(self.metadata.version)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/datos/dusun/sdk/jeffyCN.scarthgap/build/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot-rockchip/2017.09/recipe-sysroot-native/usr/lib/python3.12/site-packages/setuptools/dist.py", line 329, in _normalize_version

    normalized = str(Version(version))

                     ^^^^^^^^^^^^^^^^
  File "/datos/dusun/sdk/jeffyCN.scarthgap/build/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot-rockchip/2017.09/recipe-sysroot-native/usr/lib/python3.12/site-packages/packaging/version.py", line 211, in __init__

    epoch=int(match.group("epoch")) if match.group("epoch") else 0,
                                       ^^^^^^^^^^^

AttributeError: 'NoneType' object has no attribute 'group'

make[4]: *** [/datos/dusun/sdk/jeffyCN.scarthgap/build/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot-rockchip/2017.09/git/scripts/dtc/pylibfdt/Makefile:27: scripts/dtc/pylibfdt/_libfdt.so] Error 1

make[3]: *** [/datos/dusun/sdk/jeffyCN.scarthgap/build/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot-rockchip/2017.09/git/scripts/Makefile.build:425: scripts/dtc/pylibfdt] Error 2

make[2]: *** [/datos/dusun/sdk/jeffyCN.scarthgap/build/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot-rockchip/2017.09/git/scripts/Makefile.build:425: scripts/dtc] Error 2

make[1]: *** [/datos/dusun/sdk/jeffyCN.scarthgap/build/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot-rockchip/2017.09/git/Makefile:498: scripts] Error 2

make: *** [Makefile:157: sub-make] Error 2

make: Leaving directory '/datos/dusun/sdk/jeffyCN.scarthgap/build/tmp/work/rockchip_rk3328_evb-poky-linux/u-boot-rockchip/2017.09/git'

+ die oe_runmake failed
+ bbfatal_log oe_runmake failed
+
ERROR: u-boot-rockchip-1_2017.09-r0 do_compile: oe_runmake failed
WARNING: exit code 1 from a shell command.

I think there is a problem in parsing the uboot version “u-boot-2017.09”.

Could you please help me with it?

JeffyCN, 2024-12-27:
hi,

i've uploaded a patch for this by syncing related codes with u-boot upstream'v v2025.01

ogkita, 2024-12-27:
Thank you very much @JeffyCN for your quick response,

I will try the image generation and let you know the result.

, Oscar