JeffyCN/meta-rockchip #48: glmark2-es2-drm cannot run glmark2-es2-drm on rk3566 (rockchip-rk3566-evb.conf)

namtrino, 2023-02-16:
I change recipes-graphics/rockchip-libmali/rockchip-libmali.bb with the patch below (also added opengl to DISTRO_FEATURES).
```
diff --git a/recipes-graphics/rockchip-libmali/rockchip-libmali.bb b/recipes-graphics/rockchip-libmali/rockchip-libmali.bb
index 0c569f0..6e3a7cf 100644
--- a/recipes-graphics/rockchip-libmali/rockchip-libmali.bb
+++ b/recipes-graphics/rockchip-libmali/rockchip-libmali.bb
@@ -24,6 +24,8 @@ MALI_VERSION ??= "r18p0"
 MALI_SUBVERSION ??= "none"
 MALI_PLATFORM ??= "${@bb.utils.contains('DISTRO_FEATURES', 'wayland', 'wayland', bb.utils.contains('DISTRO_FEATURES', 'x11', 'x11', 'gbm', d), d)}"

+MALI_PLATFORM = "gbm"
+
 # The utgard DDK and 'without-cl' subversion are not providing OpenCL.
 # The ICD OpenCL implementation should work with opencl-icd-loader.
 PROVIDES:append = " ${@ 'virtual/opencl-icd' if not d.getVar('MALI_GPU').startswith('utgard') and d.getVar('MALI_SUBVERSION') != 'without-cl' else ''}"
```
When run glmark2-es2-drm on rk3566 with failed with the error below. 
```
arm_release_ver of this libmali is 'g2p0-01eac0', rk_so_ver is '3'.Error: eglCreateWindowSurface failed with error: 0x3009
Error: eglCreateWindowSurface failed with error: 0x3009
Error: CanvasGeneric: Invalid EGL state
Error: main: Could not initialize canvas
```
Noted, on rk3399 platform both  kmscube and glmark2-es2-drm work well.

JeffyCN, 2023-02-16:
try:
https://github.com/glmark2/glmark2/pull/177

namtrino, 2023-02-17:
This [patch](https://github.com/glmark2/glmark2/pull/177) solve the issue. Thank you.

closed 2023-02-17 by namtrino