JeffyCN/meta-rockchip #49: GBM application failed with error EGL_BAD_DISPLAY

namtrino, 2023-02-17:
I test ozone-gbm on rk3566 but it failed with error below.
```
[ERROR:gl_surface_egl.cc(1382)] eglInitialize Default failed with error EGL_BAD_DISPLAY
[ERROR:gl_ozone_egl.cc(20)] GLSurfaceEGL::InitializeOneOff failed.
[ERROR:viz_main_impl.cc(161)] Exiting GPU process due to errors during initialization
```
But kmscube and glmark2-es2-drm work ok.

Is this a gbm driver issue?

JeffyCN, 2023-02-17:
??? the yocto sdk with chromium worked well for years...

JeffyCN, 2023-02-17:
i was using this to build images:
https://github.com/JeffyCN/yocto-manifests

namtrino, 2023-02-18:
The Chromium Wayland from https://github.com/JeffyCN/yocto-manifests can run.
But I was faced with two issues, poor video playback performance and it crashed randomly.

JeffyCN, 2023-02-18:
video playback is at the best performance, it has hw dec and dma importing and gpu rendering.

crash issue should related to chromium-wayland or chromium framework, you can report to chromium issue system.

chromium-x11 should work too.

gbm should be used in chromiumos, it requires more patches.

fydeos has chromiumos-like sdk for rk3399 and rk3588.

JeffyCN, 2023-02-18:
one guess would be using egl get platform display api to get egl gbm display, instead of the legacy egl get display api

namtrino, 2023-02-21:
Hi, chromium gbm require EGL version 1.5 (current version is  1.4) . Is it possible to upload version 1.5 to public?

JeffyCN, 2023-02-21:
it's almost impossible to upgrade it.

and i don't think it requires that, because the google do have chromebook with chromeos(chrome + gbm) for 3288 and 3399, and fydeos has chromiumos(chromium + gbm) for 3399/3588.

namtrino, 2023-02-21:
I already tested with RK3399 with [yocto-manifest](https://github.com/JeffyCN/yocto-manifests).
It also failed EGL_BAD_DISPLAY.

JeffyCN, 2023-02-21:
it's been tested by lots of people for years...

JeffyCN, 2023-02-24:
FYI, i've heard they use mali with dummy platform + google's minigbm for chrome os/chromium os's ozone gbm

namtrino, 2023-02-26:
Thank you for an information. I think I should study FydeOS with rockchip platform first.
Because look like they use mpp (https://github.com/openFyde/overlay-rock5b-openfyde/blob/main/sys-kernel/chromeos-kernel-5_10/files/012-add-virtual-v4l2-devices.patch).

BTW, I will closed this issue because meta-rockchip focus on wayland with linux gbm.

closed 2023-02-26 by namtrino