JeffyCN/meta-rockchip #50: [Chromium-Ozone-Wayland][Honister] crash when finish video playback

chaiwat-screencloud, 2023-02-28:
Hi,
When I play 1080P mp4 video on FullHD monitor, the chromium crashed every time with the error below.

```
[991.980] [RKMPP] [889] rkmpp_new_frame(81): DEC FPS:   60.2 || Frames: 18240
[993.989] [RKMPP] [889] rkmpp_new_frame(81): DEC FPS:   59.7 || Frames: 18360
[995.911] [RKMPP] [889] rkmpp_new_frame(81): DEC FPS:   62.4 || Frames: 18480
[997.960] [RKMPP] [889] rkmpp_new_frame(81): DEC FPS:   58.6 || Frames: 18600
[999.969] [RKMPP] [889] rkmpp_new_frame(81): DEC FPS:   59.7 || Frames: 18720
[721:721:0228/142327.653018:FATAL:wayland_screen.cc(193)] Check failed: iter != display_list_.displays().end().
#0 0x0055724d41dc (/usr/lib/chromium/chromium-bin+0x79791db)
#1 0x005572401da8 (/usr/lib/chromium/chromium-bin+0x78a6da7)
#2 0x005572419b24 (/usr/lib/chromium/chromium-bin+0x78beb23)
#3 0x00557241a494 (/usr/lib/chromium/chromium-bin+0x78bf493)
#4 0x00556ea5408c (/usr/lib/chromium/chromium-bin+0x3ef908b)
#5 0x005575468518 (/usr/lib/chromium/chromium-bin+0xa90d517)
#6 0x005572f7aa70 (/usr/lib/chromium/chromium-bin+0x841fa6f)
#7 0x005574fb3950 (/usr/lib/chromium/chromium-bin+0xa45894f)
#8 0x005574fb3818 (/usr/lib/chromium/chromium-bin+0xa458817)
#9 0x005573b80644 (/usr/lib/chromium/chromium-bin+0x9025643)
#10 0x005573b87da0 (/usr/lib/chromium/chromium-bin+0x902cd9f)
#11 0x00556fc0399c (/usr/lib/chromium/chromium-bin+0x50a899b)
#12 0x00556fd5a494 (/usr/lib/chromium/chromium-bin+0x51ff493)
#13 0x00556fc25a30 (/usr/lib/chromium/chromium-bin+0x50caa2f)
#14 0x00556fd46bbc (/usr/lib/chromium/chromium-bin+0x51ebbbb)
#15 0x00556fc3984c (/usr/lib/chromium/chromium-bin+0x50de84b)
#16 0x005572f79eb8 (/usr/lib/chromium/chromium-bin+0x841eeb7)
#17 0x00556ea53d74 (/usr/lib/chromium/chromium-bin+0x3ef8d73)
#18 0x00556ea518fc (/usr/lib/chromium/chromium-bin+0x3ef68fb)
#19 0x007f9c59b030 (/usr/lib/libffi.so.8.1.0+0x702f)
#20 0x007f9c59a588 (/usr/lib/libffi.so.8.1.0+0x6587)
#21 0x007f9c405524 (/usr/lib/libwayland-client.so.0.3.0+0x9523)
#22 0x007f9c401aa4 (/usr/lib/libwayland-client.so.0.3.0+0x5aa3)
#23 0x007f9c4030a4 wl_display_dispatch_queue_pending
#24 0x005572472d08 (/usr/lib/chromium/chromium-bin+0x7917d07)
#25 0x005572492ae4 (/usr/lib/chromium/chromium-bin+0x7937ae3)
#26 0x0055724923c4 (/usr/lib/chromium/chromium-bin+0x79373c3)
#27 0x00557242b7f0 (/usr/lib/chromium/chromium-bin+0x78d07ef)
#28 0x007f9fdccc00 g_main_context_dispatch
#29 0x007f9fdccea0 (/usr/lib/libglib-2.0.so.0.6800.4+0x56e9f)
#30 0x007f9fdccf6c g_main_context_iteration
#31 0x00557242b28c (/usr/lib/chromium/chromium-bin+0x78d028b)
#32 0x0055724934d8 (/usr/lib/chromium/chromium-bin+0x79384d7)
#33 0x005572454198 (/usr/lib/chromium/chromium-bin+0x78f9197)
#34 0x00556f76f8bc (/usr/lib/chromium/chromium-bin+0x4c148bb)
#35 0x00556f77160c (/usr/lib/chromium/chromium-bin+0x4c1660b)
#36 0x00556f76d0b0 (/usr/lib/chromium/chromium-bin+0x4c120af)
#37 0x005571f0eb64 (/usr/lib/chromium/chromium-bin+0x73b3b63)
#38 0x005571f0e580 (/usr/lib/chromium/chromium-bin+0x73b357f)
#39 0x005571f0c188 (/usr/lib/chromium/chromium-bin+0x73b1187)
#40 0x005571f0cb50 (/usr/lib/chromium/chromium-bin+0x73b1b4f)
#41 0x00556e1ca9d8 ChromeMain
#42 0x007f9c5d9030 (/lib/libc.so.6+0x2b02f)
#43 0x007f9c5d9108 __libc_start_main
#44 0x00556e1ca7f0 _start
Task trace:
#0 0x00556ea4bb50 (/usr/lib/chromium/chromium-bin+0x3ef0b4f)
Crash keys:
  "total-discardable-memory-allocated" = "4199178"
  "gpu-gl-renderer" = "Mali-G52"
  "gpu-gl-vendor" = "ARM"
  "gpu-generation-intel" = "0"
  "gpu-vsver" = "3.20"
  "gpu-psver" = "3.20"
  "gpu-driver" = "v1.g2p0"
  "gpu-devid" = "0x0000"
  "gpu-venid" = "0x0000"
  "ui_scheduler_async_stack" = "0x556EA4BB50 0x0"
  "num-extensions" = "0"
  "io_scheduler_async_stack" = "0x5572827374 0x0"
  "variations" = "b56a9ab2-1410f10,9481ce98-3d47f4f4,c559031-3d47f4f4,4b935545-3d47f4f4,9a38bae3-3d47f4f4,6f3a6be-3f4a17df,e43d4487-96beea8e,a206fd93-3d47f4f4,138c1dc-3d47f4f4,c1405ec8-fb0c8ff1,330546f3-463d4c5f,d69d967d-3695c92e,"
  "num-experiments" = "12"
  "switch-11" = "--enable-crashpad"
  "switch-10" = "--enable-features=VaapiVideoDecoder,VaapiVideoEncoder"
  "switch-9" = "--enable-accelerated-video-decode"
  "switch-8" = "--ignore-gpu-blocklist"
  "switch-7" = "--ignore-gpu-blacklist"
  "switch-6" = "--gpu-sandbox-start-early"
  "switch-5" = "--no-sandbox"
  "switch-4" = "--ozone-platform=wayland"
  "switch-3" = "--use-gl=egl"
  "switch-2" = "--data-path=/tmp/data"
  "switch-1" = "--http-disk-cache-size=5"
  "num-switches" = "13"
  "osarch" = "aarch64"
  "pid" = "721"
  "ptype" = "browser"

Received signal 6
```

JeffyCN, 2023-02-28:
wayland_screen.cc(193)] Check failed: iter != display_list_.displays().end()
it looks like something wrong in the chromium's wayland screen management, please try other chromium version or report to meta-browser or report to google(crbug.com)

one possible reason would be hot-plugging screens triggered racing issue

chaiwat-screencloud, 2023-02-28:
OK, I will check it.

closed 2023-03-02 by JeffyCN