JeffyCN/meta-rockchip #51: [Chromium-Ozone-Wayland][Honister] mpp_rkvdec2 reset repeatedly with mp4 4K60 video on 4K monitor

chaiwat-screencloud, 2023-02-28:
Kernel log via dmesg

```
[  295.905081] mpp_rkvdec2 fdf80200.rkvdec: frame not complete
[  295.905148] mpp_rkvdec2 fdf80200.rkvdec: resetting...
[  295.905421] mpp_rkvdec2 fdf80200.rkvdec: reset done
[  298.188488] mpp_rkvdec2 fdf80200.rkvdec: process task timeout
[  298.188535] rkvdec_link_reg_dump:185: dump link config status from timeout
[  298.188538] rkvdec_link_reg_dump:186: reg 0 00000300 - irq status
[  298.188542] rkvdec_link_reg_dump:187: reg 1 ffff8000 - cfg addr
[  298.188545] rkvdec_link_reg_dump:188: reg 2 80000002 - link mode
[  298.188549] rkvdec_link_reg_dump:189: reg 4 0000004f - decoded num
[  298.188551] rkvdec_link_reg_dump:190: reg 5 00000054 - total num
[  298.188554] rkvdec_link_reg_dump:191: reg 6 00000001 - link mode en
[  298.188556] rkvdec_link_reg_dump:192: reg 6 ffff8000 - next ltb addr
[  298.188560] mpp_rkvdec2 fdf80200.rkvdec: frame not complete
[  298.188621] mpp_rkvdec2 fdf80200.rkvdec: resetting...
[  298.188922] mpp_rkvdec2 fdf80200.rkvdec: reset done
[  302.868318] mpp_rkvdec2 fdf80200.rkvdec: process task timeout
[  302.868365] rkvdec_link_reg_dump:185: dump link config status from timeout
[  302.868370] rkvdec_link_reg_dump:186: reg 0 00000300 - irq status
[  302.868374] rkvdec_link_reg_dump:187: reg 1 ffff8000 - cfg addr
[  302.868377] rkvdec_link_reg_dump:188: reg 2 80000002 - link mode
[  302.868379] rkvdec_link_reg_dump:189: reg 4 0000014b - decoded num
[  302.868381] rkvdec_link_reg_dump:190: reg 5 00000150 - total num
[  302.868383] rkvdec_link_reg_dump:191: reg 6 00000001 - link mode en
[  302.868385] rkvdec_link_reg_dump:192: reg 6 ffffb000 - next ltb addr
[  302.868389] mpp_rkvdec2 fdf80200.rkvdec: frame not complete
[  302.868442] mpp_rkvdec2 fdf80200.rkvdec: resetting...
[  302.868792] mpp_rkvdec2 fdf80200.rkvdec: reset done
```

RKMPP
```
[431.897] [RKMPP] [928] decoder_thread_fn(305): ERR: frame err or discard
[431.897] [RKMPP] [928] decoder_thread_fn(305): ERR: frame err or discard
[431.905] [RKMPP] [928] decoder_thread_fn(305): ERR: frame err or discard
[431.905] [RKMPP] [928] decoder_thread_fn(305): ERR: frame err or discard
[431.905] [RKMPP] [928] decoder_thread_fn(305): ERR: frame err or discard
[431.905] [RKMPP] [928] decoder_thread_fn(305): ERR: frame err or discard
[431.916] [RKMPP] [928] decoder_thread_fn(305): ERR: frame err or discard
```

JeffyCN, 2023-02-28:
sorry, this doesn't look like a meta-rockchip related issue.

"rkvdec: frame not complete" means the video dec driver(or hardware) think the input frame is incomplete, and it cannot handle that.

maybe report to the mpp team(maintains mpp framework and video dec driver): https://github.com/rockchip-linux/mpp/issues

chaiwat-screencloud, 2023-02-28:
OK, thank you. Already report to mpp team.  https://github.com/rockchip-linux/mpp/issues/368

closed 2023-02-28 by chaiwat-screencloud