JeffyCN/meta-rockchip #52: Upstream Source URI

edrose-tewke, 2023-03-08:
Are you able to add some comments to the recipe files that indicate where the upstream sources are? Since you've mirrored everything into a single repository, I can't find any way of obtaining the original upstream source URL. Presumably it's only visible on the machine that is updating the mirrors.

Some additional comments for this information would be really helpful.

JeffyCN, 2023-03-09:
you need to buy rk's sdk to get access to rk's official internal repo

closed 2023-03-09 by JeffyCN

JeffyCN, 2023-03-09:
that is of cause true for kernel and uboot, and actually they are highlighted not allowed to upload to github...

edrose-tewke, 2023-03-09:
Who do I have to contact to learn more about the SDK?

And how does this fit in with the GPL? Surely anyone who has paid for the SDK will have to release the Rockchip internal sources the moment they ship a device

JeffyCN, 2023-03-09:
you can google rk's website to find the mail address.

and i don't think the gpl means people would have to setup a github repo for others to fork the newest version, the company would provide release version for the one who buy the sdk, the newest develop version of code would be in the internal repo.

JeffyCN, 2023-03-09:
actually rk has 2 kinds of internal repo, one is release version(for customers who buy the sdk), one is develop version(for rk developers).

the release version is much older than develop one, but well tested.

JeffyCN, 2023-03-09:
and my mirrors are from the internal develop version, might not always up-to-date, i've been warned multiple times not to provide kernel and uboot that newer than the release version

JeffyCN, 2023-03-09:
FYI, there're other mirrors, for example:
https://github.com/Fruit-Pi
https://gitee.com/caesar-wang/u-boot

edrose-tewke, 2023-03-10:
Thank you, this is all really good information!