JeffyCN/meta-rockchip #55: Build chromium (only)

redfast00, 2023-03-11:
We'd like to build the latest chromium recipe, with the patches for hardware acceleration. I've made a yocto directory, with the following repo's in it:

- https://github.com/OSSystems/meta-browser.git on the master branch
- https://github.com/kraj/meta-clang on the kirkstone branch
- git://git.openembedded.org/meta-openembedded.git on the kirkstone branch
- https://github.com/radxa/meta-rockchip.git on the kirkstone-radxa branch
- git://git.yoctoproject.org/poky on the kirkstone branch

In the build directory, I executed `bitbake chromium-x11`, but this resulted in the following (truncated) error, leading me to think that some dependencies aren't fulfilled yet:

```
/armv8a-poky-linux/chromium-x11/110.0.5481.77-r0/recipe-sysroot-native= -g1 -fvisibility-inlines-hidden -stdlib=libc++ -c ../../media/gpu/v4l2/v4l2_slice_video_decode_accelerator.cc -o obj/media/gpu/v4l2/v4l2/v4l2_slice_video_decode_accelerator.o
| ../../media/gpu/v4l2/v4l2_slice_video_decode_accelerator.cc:86:5: error: use of undeclared identifier 'V4L2_PIX_FMT_H264_SLICE'
|     V4L2_PIX_FMT_H264_SLICE, V4L2_PIX_FMT_VP8_FRAME, V4L2_PIX_FMT_VP9_FRAME,
|     ^
| ../../media/gpu/v4l2/v4l2_slice_video_decode_accelerator.cc:86:30: error: use of undeclared identifier 'V4L2_PIX_FMT_VP8_FRAME'
|     V4L2_PIX_FMT_H264_SLICE, V4L2_PIX_FMT_VP8_FRAME, V4L2_PIX_FMT_VP9_FRAME,
|                              ^
| ../../media/gpu/v4l2/v4l2_slice_video_decode_accelerator.cc:86:54: error: use of undeclared identifier 'V4L2_PIX_FMT_VP9_FRAME'
|     V4L2_PIX_FMT_H264_SLICE, V4L2_PIX_FMT_VP8_FRAME, V4L2_PIX_FMT_VP9_FRAME,
|                                                      ^
| ../../media/gpu/v4l2/v4l2_slice_video_decode_accelerator.cc:2168:45: error: no matching function for call to 'size'
|   return device->GetSupportedDecodeProfiles(std::size(supported_input_fourccs_),
|                                             ^~~~~~~~~
| /home/ubuntu/yocto/build/tmp/work/armv8a-poky-linux/chromium-x11/110.0.5481.77-r0/recipe-sysroot/usr/include/c++/v1/__iterator/size.h:27:16: note: candidate template ignored: substitution failure [with _Cont = unsigned int[]]: member reference base type 'const unsigned int[]' is not a structure or union
| constexpr auto size(const _Cont& __c)
|                ^
| /home/ubuntu/yocto/build/tmp/work/armv8a-poky-linux/chromium-x11/110.0.5481.77-r0/recipe-sysroot/usr/include/c++/v1/__iterator/size.h:34:18: note: candidate template ignored: could not match 'const _Tp[_Sz]' against 'const uint32_t[]' (aka 'const unsigned int[]')
| constexpr size_t size(const _Tp (&)[_Sz]) noexcept { return _Sz; }
|                  ^
| 4 errors generated.
```

JeffyCN, 2023-03-11:
the 110 is not yet supported, the latest ported patchset is 109

maybe try to checkout meta-browser to r109

JeffyCN, 2023-03-21:
r110 should be supported now.

closed 2023-03-21 by JeffyCN