JeffyCN/meta-rockchip #57: gstreamer1.0 plugins gl* is broken

vsitdikov, 2023-03-11:
The patch: https://github.com/JeffyCN/meta-rockchip/blob/master/recipes-multimedia/gstreamer/gstreamer1.0_1.20/0003-HACK-caps-Consider-dmabuf-subset-of-system-memory.patch breaks OpenGL/ES   support

JeffyCN, 2023-03-12:
it works on my side.

closed 2023-03-12 by JeffyCN

vsitdikov, 2023-03-12:
may be it is an issue of libmali or rk3588 ?
I used quite new BR from skyworth as basis and your layer.
libmali: commit 8f46cc7b9777975f40323b0971f3483fa2aefcd3
For me gst-launch-1.0 gltestsrc ! glimagesink only works without this patch
Otherwise  gst-launch-1.0 gltestsrc ! glimagesink fails to link pipeline

vsitdikov, 2023-03-12:
Could you please describe, what is the main idea behind this patch, what is the benefit of using it ?

JeffyCN, 2023-03-13:
"gst-launch-1.0 gltestsrc ! glimagesink" works well too, and i don't think this would related to libmali, more like related to gst version (i'm using 1.20.3) and patch version (this patch been updated recently, fixed a few issues)

main idea is that when the dmabuf is mmap-able, it can be used as system memory (after mmap), so that the other plugins in the pipeline would not need to support dmabuf feature.

JeffyCN, 2023-03-13:
the gl plugins should use GLMemory in that pipeline, not dmabuf, so there's no reason to be affected by that patch, unless you are using an old version of patch.

```
commit 72cf48198d6d45ec3bc7e53796674693fae6140a
Author: Jeffy Chen <jeffy.chen@rock-chips.com>
Date:   Thu Nov 3 10:46:17 2022 +0800

    gstreamer: Update patches

    Major changes:
    1/ gstreamer1: gst-launch: Fix random hang when EOS
    2/ gstreamer1: Fix NV12_10LE40 error with waylandsink <--- the newest version of this patch came from here
    3/ gst1-plugins-base: riff: Fix bps caculation error for ADPCM
```