JeffyCN/meta-rockchip #58: 单独构建chromium报错

iyangsheng, 2023-03-15:
执行步骤

iyangsheng, 2023-03-17:
> 尚不支持 r110，请尝试将元浏览器回滚到 r109

回滚到109后构建成功，在build/tmp/deploy/rpm/armv8a/目录下生成了rpm格式的安装包，如何在ubuntu系统中安装
[image]

JeffyCN, 2023-03-17:
this repo is for yocto only, not for ubuntu. and it's dangerous to install packages acrossing systems.

if you really want to do so, please make sure that:
1/ the yocto's glibc should not be newer than the ubuntu's
2/ unpack the yocto's chromium package into ubuntu manually
3/ if the chromium complains about depended libraries mismatching when running, find those libraries in yocto and put them into ubuntu's /usr/lib/chromium/ manually

JeffyCN, 2023-03-21:
r110 should be supported now.

closed 2023-03-21 by JeffyCN