JeffyCN/meta-rockchip #59: 4.0.8 kirkstone xserver-xorg_21.1.7.bb error

chrisiberg, 2023-03-16:
Since update to 4.0.8 I get this error?

WARNING: /home/chris/yocto/poky/meta/recipes-graphics/xorg-xserver/xserver-xorg_21.1.7.bb: Exception during build_dependencies for AUTOREV
WARNING: /home/chris/yocto/poky/meta/recipes-graphics/xorg-xserver/xserver-xorg_21.1.7.bb: Error during finalise of /home/chris/yocto/poky/meta/recipes-graphics/xorg-xserver/xserver-xorg_21.1.7.bb
ERROR: ExpansionError during parsing /home/chris/yocto/poky/meta/recipes-graphics/xorg-xserver/xserver-xorg_21.1.7.bb
Traceback (most recent call last):
  File "/home/chris/yocto/poky/bitbake/lib/bb/fetch2/git.py", line 250, in Git.urldata_init(ud=<bb.fetch2.FetchData object at 0x7f4eae649be0>, d=<bb.data_smart.DataSmart object at 0x7f4eae636160>):

    >        ud.setup_revisions(d)

  File "/home/chris/yocto/poky/bitbake/lib/bb/fetch2/__init__.py", line 1349, in FetchData.setup_revisions(d=<bb.data_smart.DataSmart object at 0x7f4eae636160>):
             for name in self.names:
    >            self.revisions[name] = srcrev_internal_helper(self, d, name)

  File "/home/chris/yocto/poky/bitbake/lib/bb/fetch2/__init__.py", line 1214, in srcrev_internal_helper(ud=<bb.fetch2.FetchData object at 0x7f4eae649be0>, d=<bb.data_smart.DataSmart object at 0x7f4eae636160>, name='default'):
         if srcrev == "AUTOINC":
    >        srcrev = ud.method.latest_revision(ud, d, name)

             except KeyError:                               >            revs[key] = rev = self._latest_revision(ud, d, name)
                 return rev
  File "/home/chris/yocto/poky/bitbake/lib/bb/fetch2/git.py", line 757, in Git._latest_revision(ud=<bb.fetch2.FetchData object at 0             except KeyError:
    >            revs[key] = rev = self._latest_revision(ud, d, name)                 return rev  File "/home/chris/yocto/poky/bitbake/lib/bb/fetch2/git.py", line 757, in Git._latest_revision(ud=<bb.fetch2.FetchData object at 0x7f4eae649be0>, d=<bb.data_smart.DataSmart object at 0x7f4eae636160>, name='default'):
                         return sha1    >        raise bb.fetch2.FetchError("Unable to resolve '%s' in upstream git repository in git ls-remote output for %s" % \                      (ud.unresolvedrev[name], ud.host+ud.path))
bb.data_smart.ExpansionError: Failure expanding variable SRCPV, expression was ${@bb.fetch2.get_srcrev(d)} which triggered exception FetchError: Fetcher failure: Unable to resolve '21.1.7_2022_11_03' in upstream git repository in git ls-remote output for github.com/JeffyCN/xorg-xserver
The variable dependency chain for the failure is: SRCPV -> AUTOREV[vardepvalue]

ERROR: Parsing halted due to errors, see error messages above
Summary: There were 2 WARNING messages.
Summary: There were 2 ERROR messages, returning a non-zero exit code.

JeffyCN, 2023-03-16:
again, not supported that version.

the newest xserver patches are for 21.1.4

JeffyCN, 2023-03-16:
i might try to port all those packages next week.

chrisiberg, 2023-03-16:
thnx!

JeffyCN, 2023-03-20:
uploaded 21.1.7

closed 2023-03-20 by JeffyCN