JeffyCN/meta-rockchip #61: Mender support

nandra, 2023-04-04:
Hello,

it's just question not bug. Does anybody tried to integrate meta-mender layer with this layer? https://github.com/mendersoftware/meta-mender

As rockchip partitions are predefined integration of mender seems to be little challenging as it require u-boot changes with evn + redundant env on some predefined offsets. Would be possible to fit it somehow with required partitions setup? Thanks.

JeffyCN, 2023-04-07:
don't know much about that, it looks like you want an original u-boot instead of the RK BSP u-boot?

if so, try to delete meta-rockchip/recipes-bsp/u-boot, and add patches to make the original u-boot work.
after that, the rk style update.img and partition table file(parameter.txt) would not be generated.

nandra, 2023-04-12:
can I somehow combine SPL + u-boot to put it in first partition? Or GPT partition table layout is mandatory for all RK devices pls? Thanks.

JeffyCN, 2023-04-12:
don't know much about it, since the upstream u-boot has already support rockchip platform, maybe ask in u-boot's maillist?

from what i know, the only rockchip special thing would be:
the bootrom(hard-coded in the chip) would try to load the spl(miniloader) from a fixed offset in the storage into sram, the spl needs to be packed in rockchip special format and has size limits(check u-boot's tools/rk*.c for more details).

JeffyCN, 2023-04-18:
you can try to modify the wks.in to add your new partition, just don't move the idblock partition

closed 2023-07-24 by JeffyCN

medemi68, 2023-08-26:
@nandra If you're willing to use RAUC instead, I am currently working on a guide for this repo that allows you to set it up. RAUC is a lot more flexible in it's implementation and doesn't require modifying the bootloader code. You'll also find that a lot of the things in u-boot-rockchip "just work" as opposed to the regular u-boot. 

It may be possible to modify my guide for RAUC (that I am currently working on) to work with Mender but afaik Mender is a bit more stringent with certain things and will require you to modify the bootloader code.