JeffyCN/meta-rockchip #62: Kernel Panic - not available partitions

MaciejBarzowski, 2023-04-09:
Hello,

I have a problem with partitions what's occurs kernel panic. 
Link to full log: [https://files.fm/u/58w9cerqj](url)

[    1.837604] VFS: Cannot open root device "(null)" or unknown-block(0,0): error -6
[    1.838269] Please append a correct "root=" boot option; here are the available partitions:
[    1.839011] 0100            4096 ram0
[    1.839013]  (driver?)
[    1.839566] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(0,0)

meta-rockchip hash: 558b8ea8089cc504aa0e219cc810de521763d5a8
I only added device tree related the board which I used -> Khadas Edge 1 [https://www.khadas.com/edge1](url)
u-boot -> [https://github.com/khadas/u-boot/blob/khadas-edge-v2017.09/arch/arm/dts/rk3399-kedge.dts](url)
kernel -> [https://github.com/khadas/linux/blob/khadas-edge-4.4.y/arch/arm64/boot/dts/rockchip/rk3399-khadas-edge-linux.dts](url)

Could you give me any advice, what is wrong?

Thanks,
Maciej

chrisiberg, 2023-04-09:
You can patch rk3399-u-boot.dtsi for the correct root device, but I think there is a more elegant way.

JeffyCN, 2023-04-10:
the Rockchip BSP SDK is using UUID to specify rootdev in parameter.txt and kernel's dts:
```
commit 6f92c6ee1ec282a04c75ef230a10c52df31f5fe1
Author: Caesar <wxt@rock-chips.com>
Date:   Tue Apr 3 20:33:27 2018 +0800

    arm64: dts: rockchip: force the bootargs for rk3399 linux

+++ b/arch/arm64/boot/dts/rockchip/rk3399-linux.dtsi
@@ -46,6 +46,10 @@
 / {
        compatible = "rockchip,linux", "rockchip,rk3399";

+       chosen {
+               bootargs = "earlycon=uart8250,mmio32,0xff1a0000 swiotlb=1 console=ttyFIQ0 root=PARTUUID=614e0000-0000 rootfstype=ext4 rootwait";
+       };
```

MaciejBarzowski, 2023-04-10:
Thank you JeffCN. Now it's working :)

closed 2023-04-19 by JeffyCN