JeffyCN/meta-rockchip #64: Compiling chromium alone does not support hevc decoding?

LiuFeiChen, 2023-05-12:
repo init -u https://github.com/JeffyCN/yocto-manifests -b chromium-kirkstone-next
repo sync
source ./oe-init-build-env
bitbake chromium-ozone-wayland

https://test-videos.co.uk/vids/bigbuckbunny/mp4/h265/1080/Big_Buck_Bunny_1080_10s_30MB.mp4 can't play.

chrome://gpu only h264

Is my compilation step wrong? Or is there a switch not turned on?

JeffyCN, 2023-05-12:
it's been tested by lots of people, please test it in yocto rootfs.

likely you are not using the right environment, it depends on libv4l-rkmpp v4l-utils rockchip-mpp and special startup script and command args

LiuFeiChen, 2023-05-12:
Ok, I try to use yocto to build.

closed 2023-05-12 by LiuFeiChen

JeffyCN, 2023-05-12:
if the h264 dec works, and the mpp supports h265 dec, maybe you just need to upgrade your libv4l-rkmpp:
https://github.com/JeffyCN/libv4l-rkmpp/commits/master