JeffyCN/meta-rockchip #65: build linux-rockchip error

LiuFeiChen, 2023-05-12:
https://github.com/JeffyCN/yocto-manifests  
branch: honister-next

| make: Entering directory '/home/clf/workspace/rockchip/honister-next/build/tmp/work/rockchip_rk3588_evb-poky-linux/linux-rockchip/5.10-r0/linux-rockchip_rk3588_evb-standard-build'
|   GEN     Makefile
|   CALL    /home/clf/workspace/rockchip/honister-next/build/tmp/work-shared/rockchip-rk3588-evb/kernel-source/scripts/atomic/check-atomics.sh
|   CALL    /home/clf/workspace/rockchip/honister-next/build/tmp/work-shared/rockchip-rk3588-evb/kernel-source/scripts/checksyscalls.sh
|   CC [M]  drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/aiutils.o
|   CC [M]  drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/siutils.o
| /home/clf/workspace/rockchip/honister-next/build/tmp/work-shared/rockchip-rk3588-evb/kernel-source/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/aiutils.c:25:10: fatal error: typedefs.h: No such file or directory
|    25 | #include <typedefs.h>
|       |          ^~~~~~~~~~~~
| compilation terminated.
| make[7]: *** [/home/clf/workspace/rockchip/honister-next/build/tmp/work-shared/rockchip-rk3588-evb/kernel-source/scripts/Makefile.build:273: drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/aiutils.o] Error 1
| make[7]: *** Waiting for unfinished jobs....
| /home/clf/workspace/rockchip/honister-next/build/tmp/work-shared/rockchip-rk3588-evb/kernel-source/drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd/siutils.c:25:10: fatal error: typedefs.h: No such file or directory
|    25 | #include <typedefs.h>
|       |          ^~~~~~~~~~~~
| compilation terminated.

JeffyCN, 2023-05-12:
i'm not allowed to update kernel 5.10 here, so the github' kernel 5.10 is a little old.

try to revert:
https://github.com/JeffyCN/meta-rockchip/commit/7a87695f013cb17d0747bebbff3c17109064708a

JeffyCN, 2023-05-12:
try:

+++ b/recipes-bsp/u-boot/u-boot-rockchip.bb
@@ -55,6 +55,8 @@ do_configure_prepend() {
        fi

        [ ! -e "${S}/.config" ] || make -C ${S} mrproper
+
+       sed -i 's/ found;/ found = NULL;' ${S}/lib/avb/libavb/avb_slot_verify.c
 }

LiuFeiChen, 2023-05-15:
> 

The sed substitution is missing a slash, it should look something like this
sed -i 's/ found;/ found = NULL;/' ${S}/lib/avb/libavb/avb_slot_verify.c

closed 2023-05-15 by LiuFeiChen