JeffyCN/meta-rockchip #68: Unable to resolve '21.1.8_2022_11_03' in upstream git repository

youkabeng, 2023-05-21:
Hi, I'm trying to build the project by following README, but I met an error.  It seems a branch is missing in github.com/JeffyCN/xorg-xserver repo.

Please take a look, thanks!

The output is as follows

bitbake core-image-minimal
Loading cache: 100% |################################################################################################################################| Time: 0:00:00
Loaded 2539 entries from dependency cache.
WARNING: /home/hp/Projects/yocto/poky/meta/recipes-graphics/xorg-xserver/xserver-xorg_21.1.8.bb: Exception during build_dependencies for AUTOREV     | ETA:  0:00:00
WARNING: /home/hp/Projects/yocto/poky/meta/recipes-graphics/xorg-xserver/xserver-xorg_21.1.8.bb: Error during finalise of /home/hp/Projects/yocto/poky/meta/recipes-graphics/xorg-xserver/xserver-xorg_21.1.8.bb
ERROR: ExpansionError during parsing /home/hp/Projects/yocto/poky/meta/recipes-graphics/xorg-xserver/xserver-xorg_21.1.8.bb#####                     | ETA:  0:00:00
Traceback (most recent call last):
  File "/home/hp/Projects/yocto/poky/bitbake/lib/bb/fetch2/git.py", line 250, in Git.urldata_init(ud=<bb.fetch2.FetchData object at 0x7f22b979f250>, d=<bb.data_smart.DataSmart object at 0x7f22ba3a3a60>):
     
    >        ud.setup_revisions(d)
     
  File "/home/hp/Projects/yocto/poky/bitbake/lib/bb/fetch2/__init__.py", line 1349, in FetchData.setup_revisions(d=<bb.data_smart.DataSmart object at 0x7f22ba3a3a60>):
             for name in self.names:
    >            self.revisions[name] = srcrev_internal_helper(self, d, name)
     
  File "/home/hp/Projects/yocto/poky/bitbake/lib/bb/fetch2/__init__.py", line 1214, in srcrev_internal_helper(ud=<bb.fetch2.FetchData object at 0x7f22b979f250>, d=<bb.data_smart.DataSmart object at 0x7f22ba3a3a60>, name='default'):
         if srcrev == "AUTOINC":
    >        srcrev = ud.method.latest_revision(ud, d, name)
     
  File "/home/hp/Projects/yocto/poky/bitbake/lib/bb/fetch2/__init__.py", line 1626, in Git.latest_revision(ud=<bb.fetch2.FetchData object at 0x7f22b979f250>, d=<bb.data_smart.DataSmart object at 0x7f22ba3a3a60>, name='default'):
             except KeyError:
    >            revs[key] = rev = self._latest_revision(ud, d, name)
                 return rev
  File "/home/hp/Projects/yocto/poky/bitbake/lib/bb/fetch2/git.py", line 761, in Git._latest_revision(ud=<bb.fetch2.FetchData object at 0x7f22b979f250>, d=<bb.data_smart.DataSmart object at 0x7f22ba3a3a60>, name='default'):
                         return sha1
    >        raise bb.fetch2.FetchError("Unable to resolve '%s' in upstream git repository in git ls-remote output for %s" % \
                 (ud.unresolvedrev[name], ud.host+ud.path))
bb.data_smart.ExpansionError: Failure expanding variable SRCPV, expression was ${@bb.fetch2.get_srcrev(d)} which triggered exception FetchError: Fetcher failure: Unable to resolve '21.1.8_2022_11_03' in upstream git repository in git ls-remote output for github.com/JeffyCN/xorg-xserver
The variable dependency chain for the failure is: SRCPV -> AUTOREV[vardepvalue]

ERROR: Parsing halted due to errors, see error messages above

Summary: There were 2 WARNING messages.
Summary: There were 2 ERROR messages, returning a non-zero exit code.

JeffyCN, 2023-05-21:
hi, that should due to a recent version bump in poky i guess.

please try to delete meta-rockchip's recipes-graphics/xorg-xserver for now, i'll port the new version in the next week.

youkabeng, 2023-05-22:
Thanks! By deleting the xorg-xserver recipe, I can run the build, but I met a new error later.

hp@powerplant:~/Projects/yocto/build$ bitbake core-image-minimal
Loading cache: 100% |############################################| Time: 0:00:00
Loaded 2818 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "rockchip-rk3326-evb"
DISTRO               = "poky"
DISTRO_VERSION       = "4.0.10"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta-rockchip        = "kirkstone:0d9ebcee2ae0e3a157178b789cf060513cd12982"
meta                 
meta-poky            
meta-yocto-bsp       = "kirkstone:f53ab3a2ff206a130cdc843839dd0ea5ec4ad02f"
meta-oe              = "kirkstone:9e539aaed608a03b1128a7361426703d8f9ba705"

Initialising tasks: 100% |##########################################################################################################################################################################| Time: 0:00:02
Sstate summary: Wanted 477 Local 0 Mirrors 0 Missed 477 Current 670 (0% match, 58% complete)
NOTE: Executing Tasks
ERROR: linux-rockchip-4.4-r0 do_kernel_metadata: A KBUILD_DEFCONFIG 'rk3326_linux_defconfig' was specified, but not present in the source tree (/home/hp/Projects/yocto/build/tmp/work-shared/rockchip-rk3326-evb/kernel-source/arch/arm64/configs/)
ERROR: linux-rockchip-4.4-r0 do_kernel_metadata: ExecutionError('/home/hp/Projects/yocto/build/tmp/work/rockchip_rk3326_evb-poky-linux/linux-rockchip/4.4-r0/temp/run.do_kernel_metadata.5403', 1, None, None)
ERROR: Logfile of failure stored in: /home/hp/Projects/yocto/build/tmp/work/rockchip_rk3326_evb-poky-linux/linux-rockchip/4.4-r0/temp/log.do_kernel_metadata.5403
ERROR: Task (/home/hp/Projects/yocto/meta-rockchip/recipes-kernel/linux/linux-rockchip_4.4.bb:do_kernel_metadata) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1559 tasks of which 1557 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/hp/Projects/yocto/meta-rockchip/recipes-kernel/linux/linux-rockchip_4.4.bb:do_kernel_metadata
Summary: There were 2 ERROR messages, returning a non-zero exit code.

JeffyCN, 2023-05-22:
it looks like the newest BSP kernel splitted that defconfig into rockchip_linux_defconfig and rk3326_linux.config.

please try:
```
index 57dc957..4893e48 100644
--- a/recipes-kernel/linux/linux-rockchip.inc+++ b/recipes-kernel/linux/linux-rockchip.inc@@ -34,6 +34,13 @@ python () {
 # Force regenerating defconfig
 do_kernel_metadata_prepend() {                      rm -f ${WORKDIR}/defconfig           +
+       CONFIG_DIR="${S}/arch/${ARCH}/configs"
+       if [ ! -f "${CONFIG_DIR}/${KBUILD_DEFCONFIG}" ]; then
+               cat "${CONFIG_DIR}/rockchip_linux_defconfig" \
+               "${CONFIG_DIR}/${KBUILD_DEFCONFIG/_defconfig/.config}" > \
+               "${CONFIG_DIR}/${KBUILD_DEFCONFIG}"
+       fi                                    }
 # Link rockchip style images
```

JeffyCN, 2023-05-22:
that patch doesn't work...
i've upload the fix, please update your meta-rockchip

phphxh, 2023-05-22:
Now it works, thanks a lot.

closed 2023-07-13 by JeffyCN