JeffyCN/meta-rockchip #71: The uboot/trust/boot partitions are not recognized

phphxh, 2023-06-18:
Hi, I wrote the built wic image to a sd card, but the uboot/trust/boot partitions are not recognized. They're marked microsoft basic data. Is it expected?

Device     Start    End Sectors   Size Type
/dev/sda1  16384  24575    8192     4M Microsoft basic data
/dev/sda2  24576  32767    8192     4M Microsoft basic data
/dev/sda3  32768  71987   39220  19.2M Microsoft basic data
/dev/sda4  81920 382805  300886 146.9M Linux filesystem

I'm new to this area, sorry if any dump questions. Currently, I own some rk3326/3566 handhelds instead of a rockchip dev board. What I'm gonna do is to create custom firmwares with yocto for the handhelds I own. I installed the burned sdcard to a rk3326 device(Anbernic RG351v), it shows black screen. The image I chose is core-image-full-cmdline. There must be some gaps I don't know. 
I would appreciate it if you could redirect me some materials for my scenario. Thanks a lot.

JeffyCN, 2023-06-18:
1/ those partitions don't contain valid filesystems, u-boot is raw binary, trust is  custom format, boot is android boot image format, so it's expected to be unknow format(basic data)

2/ it's not easy to boot from sdcard, you may need to check the debug serial port's messages to debug it.

i don't know much about that, but i think you may need to make sure that:
1/ the sdcard is workable by default in early boot(hardware related)
2/ the u-boot would prefer to boot the sdcard's kernel
3/ the sdcard's kernel prefer to boot into the sdcard's rootfs.
4/ the sdcard's u-boot and kernel are workable in your device.

phphxh, 2023-06-18:
Thanks, will dig into it. Closing this thread.

closed 2023-06-18 by phphxh