JeffyCN/meta-rockchip #75: u-boot-rockchip compilation error. Patch is not implemented

przemoch-dev, 2023-08-07:
Hi, I'm trying to build image for Rock 4 using latest kirkstone branch and I face an error while u-boot compilation.

```
| /home/przemoch/yocto/build/tmp/work/rockchip_rk3399_sapphire_excavator-poky-linux/u-boot-rockchip/1_2017.09-r0/git/common/lrz.c: In function 'zmodem_rx':
| /home/przemoch/yocto/build/tmp/work/rockchip_rk3399_sapphire_excavator-poky-linux/u-boot-rockchip/1_2017.09-r0/git/common/lrz.c:2431:47: error: 'now' may be used uninitialized [-Werror=maybe-uninitialized]
|  2431 |                                 if (stop_time && now>=stop_time) {
|       |                                     ~~~~~~~~~~^~~~~~~~~~~~~~~~~
| /home/przemoch/yocto/build/tmp/work/rockchip_rk3399_sapphire_excavator-poky-linux/u-boot-rockchip/1_2017.09-r0/git/common/lrz.c:2401:40: note: 'now' was declared here
|  2401 |                                 time_t now;
|       |                                        ^~~
```

and then 

```
| cc1: all warnings being treated as errors
| make[2]: *** [/home/przemoch/yocto/build/tmp/work/rockchip_rk3399_sapphire_excavator-poky-linux/u-boot-rockchip/1_2017.09-r0/git/scripts/Makefile.build:280: common/lrz.o] Error 1
```

then bitbake fails.

I resolved it by creating `u-boot-rockchip.bbappend` containing

```
SRC_URI += " \
    file://0003-Revert-Makefile-enable-Werror-option.patch \
"
```

I'm using `rockchip-rk3399-sapphire-excavator` machine (btw, I'm not sure if it is the right one for Rock 4SE) with following configuration

```
Build Configuration:
BB_VERSION           = "2.0.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "rockchip-rk3399-sapphire-excavator"
DISTRO               = "poky"
DISTRO_VERSION       = "4.0.11"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta-rockchip        = "kirkstone:e59dbd13941798de755d992529df940fe43c63bd"
meta
meta-poky
meta-yocto-bsp       = "kirkstone:7fdfb828fd7586a2e9b93b174e48246fc2a90f3e"
meta-oe              = "kirkstone:4da92ed9be41734f6ced46b981958e2e868cbff2"
```

JeffyCN, 2023-08-07:
good idea! could you upload your patch(or PR?)

przemoch-dev, 2023-08-07:
The patch is already there. I just added it to `SRC_URI`, because it seems to be not in use at all

JeffyCN, 2023-08-08:
should be fixed now

closed 2023-08-08 by JeffyCN