JeffyCN/meta-rockchip #79: OTA Update / Additional partition

medemi68, 2023-08-15:
Hello,

I am trying to build an over-the-air update mechanism for my RK3288-EVB based board. There are a couple of things I want to do. The first is I want to have a separate partition to store configuration files that will persist between updates. And the second is that I want to be able to replace the rootfs/boot/uboot partitions on the system without having to reflash the entire system using update.img. I was thinking of creating a script that would do this, check for an update (which I can easily do), download a zip containing the rootfs.img, boot.img, uboot.img etc. and then use dd (I think) to write to the partitions in /dev/mmcblk as a way to overwrite them. 

Is there a better way to do this? How can I add an additional partition to my image?

JeffyCN, 2023-08-15:
the rockchip official sdk would have an ota flow(similar with android's)

for adding new partitions, you can try to modify the .wic file, or hack the rockchip-common.inc's gen_rkparameter to use your custom parameter.txt(rockchip's legacy partition table file)

medemi68, 2023-08-16:
I have been looking into RAUC and trying to see if I can use that. Part of it involves creating a uboot script.

medemi68, 2023-08-16:
I know that mender has integration for rockchip.

medemi68, 2023-08-16:
I am trying to use mender right now and I am getting the following error:

```
| install: cannot stat '/home/miguel/build/tmp/deploy/images/rockchip-rk3288-evb/uboot.env': No such file or directory
| + bb_sh_exit_handler
| + ret=1
| + [ 1 != 0 ]
| + echo WARNING: exit code 1 from a shell command.
| WARNING: exit code 1 from a shell command.
| + exit 1
```

It seems like the uboot.env file is not generated by meta-rockchip.

JeffyCN, 2023-08-16:
there's no plan to support mender.

there should be another meta-rockchip with upstream kernel and u-boot, that might be easier to work with mender

medemi68, 2023-08-17:
Does the upstream kernel and u-boot support rockusb mode?

medemi68, 2023-08-17:
Because of your unwillingness to even take a look at whether or not it is possible to get it to output a uboot.env file, I went ahead and used the other branches' u-boot and ended up bricking my device. The community version of meta-rockchip sucks, and you've provided no solutions for updates.

medemi68, 2023-08-17:
Sorry for being upset, thought I had completely bricked it beyond repair for a second. I was able to find two tiny pads located on two separate spots on the board (this board is a random chinese board, not a standard evb or a firefly) and I was able to short them and enter maskrom mode. Still, I do think that I can modify your u-boot to support mender. The mender docs make it clear enough as to what needs to be changed, I'm just not seeing a uboot.env file in the output. I think it is the same as u-boot-initial-env but I am not sure.

Your version of meta-rockchip is the only one that has worked right out of the get-go and seems to be better supported than the other meta-rockchip.

JeffyCN, 2023-08-17:
maybe, i can create a new branch if your patches break other cases

medemi68, 2023-08-18:
So the missing u-boot.env file turns out might be the same as u-boot-rockchip-initial-env. I haven't tried modifying your source yet (I might just do the whole thing in another layer) (it's pretty late where I am so i'll try tomorrow). See https://hub.mender.io/t/compulab-iot-gate-imx8-does-not-generate-uboot-env-file-anymore/5948/8

medemi68, 2023-08-20:
I think I am going to give up on using mender as it seems to do a lot of proprietary stuff (requiring major changes of the bootloader). I'm probably going to stick with RAUC. I'll open a new issue with that.