JeffyCN/meta-rockchip #80: RAUC Integration

medemi68, 2023-08-20:
Hello,

I was wondering if it would be possible to add integration for RAUC. It doesn't require patching the bootloader like Mender does, and works with a boot.scr script and userspace configuration of the boot environment (using fw_setenv and fw_getenv).

It seems the only real things that are required for it are using a custom boot.scr file, and of course whatever implementation is required to use fw_setenv and fw_getenv.

Miguel

medemi68, 2023-08-20:
Do I just need to add a UBOOT_ENV and UBOOT_ENV_SUFFIX to my custom board config?

medemi68, 2023-08-20:
Okay, so after doing some looking around, I realized that CONFIG_IS_IN_NOWHERE=y is set in the .config file for u-boot-rockchip somehow. I am assuming that means it is relying on compiled in versions which won't work with this. I'm going to try and change it so that it uses the emmc and then add the config directly to the emmc using the wks file. Will get back to this later.

medemi68, 2023-08-20:
OKAY! So this was very difficult to figure out but I finally managed to get the fw_printenv and fw_setenv working and to get u-boot to use the uboot in mmc. I will try to get the rest of the RAUC implementation done later.

### Preparation
First things first, we need to add the `u-boot-fw-utils` tools to your image. You can add that to an IMAGE_INSTALL:append wherever you normally define those. You might also want to add `util-linux-lsblk` so we can use `lsblk` later.

### Board Config

You need to edit the defconfig for your board in u-boot. In my case, that was ```configs/rk3288_defconfig```. You will need to either use ```devtool modify u-boot-rockchip``` or some other method of editing your board config. Add the following lines to it:
```
# CONFIG_IS_IN_NOWHERE is not set
CONFIG_ENV_IS_IN_MMC=y
CONFIG_ENV_OFFSET=0x1800000
CONFIG_ENV_SIZE=0x20000
```

This switches u-boot from using the compiled env file (compiled in the binary) to one that is located on the eMMC flash. We are essentially giving the env space 128KB of storage (CONFIG_ENV_SIZE) and an offset of 24MB (CONFIG_ENV_OFFSET). We set it to 24MB to give us ample room for uboot.img, trust.img, and boot.img that needs to come before it. Of course, the more space between, the better. I just happen to know that those three files from the output take a total of 23.7MB for my compilation.

### Modify WIC
I am sure you could do this in your own layer but for simplicity sake I modified meta-rockchip's `generic-gptdisk.wks.in` in the `meta-rockchip/wic` folder.

Here is what the file contents should look like for now (we won't add the dual partition yet, this is just to test to see if we can get fw_printenv working):

```
# Copyright (c) 2019, Fuzhou Rockchip Electronics Co., Ltd
# Released under the MIT license (see COPYING.MIT for the terms)
#
# long-description: Creates a GPT disk image for Rockchip boards

# 0~32K: gpt
bootloader --ptable gpt
part --source rawcopy --sourceparams="file=idblock.img" --align 32 --no-table
part --source rawcopy --sourceparams="file=uboot.img" --part-name uboot --align 8192

part --source rawcopy --sourceparams="file=trust.img" --part-name trust
part --source rawcopy --sourceparams="file=boot.img" --part-name boot

part env --source rawcopy --sourceparams="file=env.img" --part-name env --offset 24576 --fixed-size 128K
part / --source rootfs --fstype ${RK_ROOTFS_TYPE} --part-name rootfs --uuid ${RK_ROOTDEV_UUID} --align 8192
```
You'll notice there is a new part there called env, and it compiles a file called env.img. We need to create that env.img file using a special tool, and for that we are going to use a bbappend file in our own layer.

### Adding env.img
In your own u-boot-rockchip.bbappend file in your own layer, add the following (you'll notice that I used devtool to make a patch for my rk3288_defconfig from earlier):

```
FILESEXTRAPATHS:prepend := "${THISDIR}/${PN}:"

SRC_URI += "file://0001-Changed-env-location-to-be-in-mmc-at-offset.patch"

DEPENDS += "u-boot-tools-native"

do_compile:append() {

    if [ -f "${B}/u-boot-initial-env" ]; then
        mkenvimage -s 0x20000 -o env.img "${B}/u-boot-initial-env"
    else
        bbwarn "u-boot-initial-env not found"
    fi

}

do_deploy:append() {

    if [ -f "env.img" ]; then
        install env.img "${DEPLOYDIR}/env.img-${PV}"
        ln -sf "env.img-${PV}" "${DEPLOYDIR}/env.img"
    else
        bbwarn "env.img not found during deploy"
    fi

}

```
What we are doing here is adding the u-boot-tools-native package which will give us the tool `mkenvimage` that we will need to generate the env.img file. On line 10, you'll notice that we set the size to 0x20000 (128KB) and we use an output file of env.img.  We are also using the u-boot-initial-env that is automatically generated by u-boot-rockchip.

In do_deploy:append, we add the env.img file to our deploy directory so it can be used later during WIC creation.

### Build and Flash
Build your image like you normally would, but we need to do something special for flashing:

For whatever reason, *** we are unable to use the update.img*** when flashing. I am not sure why yet but I will figure it out eventually. Instead, we can use the wic file and flash that instead.

If your device is in MASKROM mode, which it probably is, you will need to flash the loader.bin first into memory (I happen to have the rkbin tools at a different directory but you'll need to have access to the rkbin tools first):

```
sudo ./upgrade_tool DB <latest_build_image_location>/loader.bin
```
Of course, replace `<latest_build_image_location>` with the location of your loader.bin file, which is typically in your `build/latest` directory.

After that is complete, the device will now be in rockusb mode, and you can flash the WIC
```
sudo ./upgrade_tool wl 0 <latest_build_image_location>/<your_wic_file_name>.wic
```
Of course, replace both the latest_build_image_location and your_wic_file_name with their appropriate locations and names.

### Setting /etc/fw_env.config
I haven't done it yet, but you can do this in a recipe. You'll need to create a file at /etc/fw_env.config in order to get this working, but your configuration might vary depending on the name of the block device and the partition that the env.img gets put on. You can use lsblk (make sure you include that in your image) to get the name of the partition with a size of 128K. Here is the /etc/fw_env.config file I used:
```
/dev/mmcblk1p4 0x0000 0x20000
```

And thats it! Test it with fw_printenv

medemi68, 2023-08-23:
Okay, so I've made some major progress with the bootloader and integrating RAUC and I will have an update soon.

medemi68, 2023-08-26:
Got RAUC mostly working. Just need to generate a system.conf file for it and add it to the guide I'm working on to post here. Any chance we could maybe add a markdown file to the main source?

JeffyCN, 2023-08-26:
good to hear that, you can create a pull request for adding that guide, or using the github wiki

ursfassler, 2024-05-08:
@medemi68 Did you had any success with rauc? Do you have the code somewhere so I can use it?