JeffyCN/meta-rockchip #82: rk3588 weston cannot start

LiuFeiChen, 2023-08-24:
weston --backend=drm-backend.so --idle-time=100
Date: 2021-01-01 UTC
[12:18:31.052] weston 10.0.2
               https://wayland.freedesktop.org
               Bug reports to: https://gitlab.freedesktop.org/wayland/weston/issues/
               Build: 10.0.2
[12:18:31.053] Command line: weston --backend=drm-backend.so --idle-time=100
[12:18:31.054] OS: Linux, 5.10.160-rockchip-standard, #1 SMP Fri Aug 4 02:10:04 UTC 2023, aarch64
[12:18:31.054] Flight recorder: enabled
[12:18:31.055] warning: XDG_RUNTIME_DIR "/tmp" is not configured
correctly.  Unix access mode must be 0700 (current mode is 0777),
and must be owned by the user UID 0 (current owner is UID 0).
Refer to your distribution on how to get it, or
http://www.freedesktop.org/wiki/Specifications/basedir-spec
on how to implement it.
[12:18:31.057] Using config file '/etc/xdg/weston/weston.ini'
[12:18:31.058] Output repaint window is 7 ms maximum.
[12:18:31.058] Loading module '/usr/lib/libweston-10/drm-backend.so'
[12:18:31.059] initializing drm backend
[12:18:31.059] Trying libseat launcher...
[12:18:31.059] libseat: session control granted
[12:18:31.060] using /dev/dri/card0
[12:18:31.060] DRM: supports atomic modesetting
[12:18:31.060] DRM: does not support GBM modifiers
[12:18:31.061] DRM: supports picture aspect ratio
[12:18:31.062] Loading module '/usr/lib/libweston-10/gl-renderer.so'
[12:18:31.062] EGL client extensions: EGL_EXT_client_extensions
               EGL_EXT_platform_base EGL_KHR_client_get_all_proc_addresses
               EGL_KHR_platform_gbm EGL_KHR_platform_wayland
               EGL_EXT_platform_wayland
arm_release_ver of this libmali is 'g6p0-01eac0', rk_so_ver is '6'.
Failed creating base context during opening of kernel driver.
Kernel module m[y not have been loaded
[12:18  :31.066] failed to initialize1 display
[12:18:31.066] EGL e4rror state: EGL_NOT_INITIALIZED (0x3001)
[12:18:31.066] fai3led to initialize egl
.062089] mali fb000000.gpu: Firmware header version 0.2 not understood
[  143.062125] mali fb000000.gpu: Firmware initialization failed
[  143.063739] dwhdmi-rockchip fdea0000.hdmi: use tmds mode
[  143.06[12:18:31.125] fatal: failed t90o create compositor backend
12] dwhdmi-rockchip fdea0000.hdmi: use tmds mode
[  143.085838] dwhdmi-rockchip fdea0000.hdmi: use tmds mode
Internal warning: debug scope 'drm-backend' has not been destroyed.

JeffyCN, 2023-08-24:
Failed creating base context during opening of kernel driver.
Kernel module m[y not have been loaded

something wrong ib your kernel mali driver.

LiuFeiChen, 2023-08-24:
root@rockchip-rk3588-rock-5b:/dev# cat mali0 
[ 1131.958287] mali fb000000.gpu: Firmware header version 0.2 not understood
[ 1131.958601] mali fb00000cat: can't open 'mali0': Invalid argument
0.gpu: Firmware initialization failed

So how do I configure the kernel?

JeffyCN, 2023-08-24:
that log means the mali firmware(/lib/firmware/mali_csffw.bin) is too new for the kernel driver.

rk3588 is using Rockchip's BSP kernel 5.10, the mirror on my github for it is a little old(i'm not allowed to mirror the newer version here).

so maybe you can:
1/ find other RK 5.10 kernel source on github(radxa maybe)
or
2/ downgrade the libmali's firmware(firmware/g610/mali_csffw.bin):
https://github.com/JeffyCN/mirrors/commits/libmali/firmware/g610/mali_csffw.bin

this one might work:
https://github.com/JeffyCN/mirrors/commit/7809b2a97f2254e268d35f868b124099aea3462c

LiuFeiChen, 2023-08-24:
Thank you very much, this commit version work fine for me.

https://github.com/JeffyCN/mirrors/commit/ca33693a03b2782edc237d1d3b786f94849bed7d

LiuFeiChen, 2023-08-24:
md5sum mali_csffw.bin
8cacdb33ddb8ccaa2281d318e60c1376  mali_csffw.bin

closed 2023-08-24 by LiuFeiChen