JeffyCN/meta-rockchip #83: What is the best way to enable uart?

yrik, 2023-08-29:
I'm building the board where I want to use uart/serial for communication with an external device.
I have two options. Either to free the default uart from system communications (linux is using this serial to show boot logs) or to enable another uart. Is it possible to configure it on yocto level? Are there any configuration variables I could use?

I see there is `SERIAL_CONSOLES` defined in this repo, would it help if I redefine it?
What is the best way to include a custom device tree overlay like the one mentioned here https://forum.radxa.com/t/enable-serial-port-uart2/15467?

JeffyCN, 2023-08-29:
it's easy to enable uart in kernel dts, something like
```
&uartx {
status="okay"
};
```

so you might need to:
1/ use git clone to download the kernel source
2/ modify your kernel dts and git commit it
3/ set the kernel to use this local source(follow the readme.md of this repo's last section)

yrik, 2023-08-30:
Thank you for the answer. 
I was looking for a way to do it on the level of yocto configuration.
For example, for raspberry pi I can do following settings in `conf/machine/rpi4.conf`:

```
 # prevent using serial for linux output
CMDLINE_SERIAL = ""
 # modify boot commands
CMDLINE:append = "..."
 # modify device overlays 
RPI_KERNEL_DEVICETREE_OVERLAYS:append = "overlays/tc358743-overlay.dtbo"
```

closed 2023-09-04 by yrik