JeffyCN/meta-rockchip #85: WPEwebkit on rk3288

woutervanh, 2023-09-12:
I'm having difficulties running wpewebkit on rk3288. It doesn't work with drm or with wayland backend.
Weston itself is running. glmark2 on top of it too. glimagesink too. Only cog (wpewebkit launcher) has difficulties.
Relevant error I think is:
running without wayland:
# cog -P drm http://80.0.0.63
```
 failed to import a wl_buffer_resource into gbm_bo
arm_release_ver of this libmali is 'r18p0-01rel0, rk_so_ver is '4'.
```
running with wayland:
# cog -P wl http://80.0.0.63
```
EGLDisplay Initialization failed: EGL_NOT_INITILAIZED
Cannot create EGL sharing context: invalid display (last error: EGL_SUCCESS)
Cannot create EGL context: invalid display (last error: EGL_SUCCESS)
```
Any insights or anyone has experience with wpewebkit?

JeffyCN, 2023-09-13:
haven't try that.
but it's abnormal that it tries to import wayland buffer in drm backend. (the mali doesn't support importing wl buffer)

for egl init error, normally because it cannot communicate with display server(weston)

woutervanh, 2023-09-13:
Myeah, I had to pach wpewebkit becaus symbols were in libmali and libmali_hook, but didn't work anyway.
Restarted from scratch with a mainline kernel + panfrost + mesa  + drm and now everything works well.
Only thing I'm missing is video right now. Do you happen to know how I can add a gstreamer plugin for the decoder? I'm on mainline, so not sure how to continue or what to do for it

closed 2023-09-13 by woutervanh

simonchen007, 2023-10-02:
If you using wpewebkit to 2.38.6, it would works with libmali in drm mode. 
```log
[root@Nscreen64:~]# cog -P drm
arm_release_ver: g13p0-01eac0, rk_so_ver: 9

(cog:13780): Cog-DRM-WARNING **: 22:34:34.554: Renderer 'modeset' does not support rotation 0 (0 degrees).

(cog:13780): GLib-GIO-WARNING **: 22:34:34.583: Your application does not implement g_application_activate() and has no handlers connected to the 'activate' signal.  It should do one of these.
arm_release_ver: g13p0-01eac0, rk_so_ver: 9
Cog-Core-Message: 22:34:34.967: <http://127.0.0.1/index.html> Load started.
Cog-Core-Message: 22:34:35.024: <http://127.0.0.1/index.html> Loading...
Cog-Core-Message: 22:34:38.713: <http://127.0.0.1/index.html#/v-main> Loaded successfully.
```
but not works in wayland mode:
```log
[root@Nscreen64:~]# export WAYLAND_DISPLAY=wayland-0
[root@Nscreen64:~]# export XDG_RUNTIME_DIR=/tmp/.xdg;
[root@Nscreen64:~]# export WESTON_DRM_MIRROR=1;
[root@Nscreen64:~]# export WESTON_DISABLE_ATOMIC=1;
[root@Nscreen64:~]# export WESTON_DRM_KEEP_RATIO=1;
[root@Nscreen64:~]# export WESTON_FREEZE_DISPLAY=/tmp/.xdg/.freeze_weston;
[root@Nscreen64:~]# weston &
[root@Nscreen64:~]# cog --platform=wl 
arm_release_ver: g13p0-01eac0, rk_so_ver: 9
xkbcommon: ERROR: couldn't find a Compose file for locale "C" (mapped to "C")

(cog:17083): GLib-GIO-WARNING **: 22:36:12.872: Your application does not implement g_application_activate() and has no handlers connected to the 'activate' signal.  It should do one of these.
arm_release_ver: g13p0-01eac0, rk_so_ver: 9
Cog-Core-Message: 22:36:13.257: <http://127.0.0.1/index.html> Load started.
**
Cog-Wayland:ERROR:../platform/wayland/cog-platform-wl.c:1705:on_export_wl_egl_image: assertion failed: (s_eglCreateWaylandBufferFromImageWL)
Bail out! Cog-Wayland:ERROR:../platform/wayland/cog-platform-wl.c:1705:on_export_wl_egl_image: assertion failed: (s_eglCreateWaylandBufferFromImageWL)
Aborted
[root@Nscreen64:~]# warning: queue 0x7f20001290 destroyed while proxies still attached:
  wpe_dmabuf_pool@11 still attached

[root@Nscreen64:~]# 
```
Any idea for that, Jeffy ? @JeffyCN

JeffyCN, 2023-10-11:
hi,
the mali doesn't support 
eglCreateWaylandBufferFromImageWL

simonchen007, 2023-10-12:
> hi, the mali doesn't support eglCreateWaylandBufferFromImageWL

Is possible to add  "importing wl buffer" via libmali-hook ?

JeffyCN, 2023-10-12:
sorry, i don't think it's possible.

i would suggest to use panfrost in this case, or modify the wpewebkit somehow, to use dma buf fd or gbm bo to pass buffers

simonchen007, 2023-10-12:
> sorry, i don't think it's possible.
> 
> i would suggest to use panfrost in this case, or modify the wpewebkit somehow, to use dma buf fd or gbm bo to pass buffers

Panfrost is much slower then the rockchip libmali. 
Eg: From the glmark2-es2-wayland testing panfrost just got 209 sources, but  rockchip libmali got 402 sources.

Does RK have a plan to porting the wpewebkit ?

JeffyCN, 2023-10-12:
that might due to gpu clock freq? or not sure would wpewebkit support vulkan with panfrost?

there's no plan in rockchip i think(mainly use chromium for wayland browser), it's a big job, arm might trying to support more wayland related features in newer ddk, but might not for rk3326.

simonchen007, 2023-10-12:
> that might due to gpu clock freq? or not sure would wpewebkit support vulkan with panfrost?
> 
> there's no plan in rockchip i think(mainly use chromium for wayland browser), it's a big job, arm might trying to support more wayland related features in newer ddk, but might not for rk3326.

I have already run the fllowing scripts before glmark2 testing.
```bash
echo performance > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor
echo userspace > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor
echo 1514000 > /sys/devices/system/cpu/cpufreq/policy0/scaling_setspeed
cat /sys/devices/system/cpu/cpufreq/policy0/cpuinfo_cur_freq
echo performance > /sys/devices/platform/ff400000.gpu/devfreq/ff400000.gpu/governor
echo performance >  /sys/class/devfreq/ff400000.gpu/governor
```
Panfrost always have low perforcemance(then the rk-libmali). So, I think it must some ptimization in the  rk-libmali.
The web looks stuck when using touching. 

We tried RK3566 with panfrost + wpewebkit, the GPU of RK3566 having higher perforcemance(not stuck in the web). But the customer not accept to upgrade from RK3326s to RK3566 in their oven(although they are selling expensive https://item.jd.com/1480602775.html).   That's forcing us to try wpewebkit + libmali in RK3326s.

Actually, We like to followed the RK SDK (libmali + weston + chromium-ozone-wayland ), but the last SDK having the a scaling issue since Chromium 104.

Yes, I know this is a big job if porting the wpewebkit.
If rockchip have no plan, We have to go back to libmali + weston + chromium-ozone-wayland.
I have submited a redmine issue (ID:442659) , hope  RK can help us for resolving this issue.

Thanks