JeffyCN/meta-rockchip #86: Update patches for newer version of chromium

amazingfate, 2023-10-18:
https://github.com/JeffyCN/meta-rockchip/blob/master/dynamic-layers/recipes-browser/chromium/chromium_114.0.5735/0001-media-gpu-v4l2-Support-V4L2-VDA-with-libv4l2-on-Linu.patch is outdated and it's hard to maintain as chromium updates their code. We can use build flags `use_v4l2_codec=true use_v4lplugin=true enable_platform_hevc=true enable_hevc_parser_and_hw_decoder=true` without this patch to  build chromium with mainline linux-libc-dev for v4l2-request APIs.
Here is the pathes I use to do hardware decoding on snapdragon 865 platform, which has v4l2m2m decoding API:
https://github.com/amazingfate/chromium-libv4l2-patches/tree/main/v114.0.5735.35
I also dropped other patches, but the biggest difference is that I removed the patch adding flag `use_linux_v4l2_only`.

JeffyCN, 2023-10-18:
that should depends on mainline kernel(with mainline drivers)

this repo is mainly for rockchip's bsp kernel and related packages(including mpp), so i cannot do that.

for mainline kernel, there's another meta-rockchip, but i'm not sure would they willing to maintain chromium patches in it.

amazingfate, 2023-10-18:
Maybe we can add v4l2-request uAPI headers in chromium, just to make build success.

JeffyCN, 2023-10-18:
that is doable, but i think removing the v4l2-request codes seems more easiler for me when porting patches to new chromium ;)

amazingfate, 2023-10-18:
I'm packaging chromium for linux distros and linux uAPI headers are not big problems for me. Anyway many thanks for your work~

closed 2023-10-18 by amazingfate