JeffyCN/meta-rockchip #91: RK3399 reset/reboot every 10min

chrisiberg, 2023-11-27:
I have an iot-3399e board with Kernel 4.19 the board is rebooting after about 10min, the CPU Temp is at ~65C. The DTS is from a running ubuntu image with the same kernel version.
There is no error on the serial console, just rebooting without reason. Does anybody have a hint for me where to start?

chrisiberg, 2023-11-28:
same with kernel 4.4, reboot after 10min.

JeffyCN, 2023-11-29:
you csn try to use the working loader & kernel with this yocto's rootfs.

if there's no restart log in serial console, likely something wrong in the kernel/hardware side

chrisiberg, 2023-12-04:
I have the same problem with a iot-3568A now (same company) and I ran in the same problem?
If I only exchange the rootfs, "everything" is working.
I tried to disable u-boot watchdog, event if I think the timeout must be shorter.

The restart ist initiated on u-boot level, if I stop u-boot with SCTRG+C it does the restart after 10 minutes too.

How would it behave if the trusted firmware is wrong, would it even boot, or is there a time limitation?
Is 600sec there a number?
Is there even a company difference or is for every rk3568 the same trusted firmware? 

What can I enable in u-boot or kernel to se why the reboot is initiated, tried many things already but no usable output in the console.

What about this message?

```
TEEC: Waring: Could not find security partition
**lib/avb/libavb_user/avb_ops_user.c: trusty_read_lock_state failed**
Error determining whether device is unlocked.
Device is: UNLOCKED
```

maybe the log is saying something?

```
DDR V1.18 f366f69a7d typ 23/07/17-15:48:58
In
LP4/4x derate en, other dram:1x trefi
ddrconfig:0
LP4 MR14:0x4d
LPDDR4, 324MHz
BW=32 Col=10 Bk=8 CS0 Row=17 CS=1 Die BW=16 Size=4096MB
tdqss: cs0 dqs0: 24ps, dqs1: -72ps, dqs2: -48ps, dqs3: -144ps,

change to: 324MHz
clk skew:0x62

change to: 528MHz
clk skew:0x58

change to: 780MHz
clk skew:0x58

change to: 1560MHz(final freq)
PHY drv:clk:38,ca:38,DQ:30,odt:60
vrefinner:16%, vrefout:29%
dram drv:40,odt:80
vref_ca:00000068
clk skew:0x1c
cs 0:
the read training result:
DQS0:0x34, DQS1:0x35, DQS2:0x38, DQS3:0x31,
min  : 0xe  0xf 0x12  0xf  0x3  0x5  0x8  0x3 , 0xa  0x8  0x2  0x2  0xf  0xb  0xd  0x9 ,
      0x13 0x14  0xd  0xa  0x2  0x1  0x3  0x5 , 0xd  0x8  0x9  0x1 0x12 0x10  0xd  0xf ,
mid  :0x2a 0x2b 0x2d 0x2b 0x1e 0x22 0x24 0x1f ,0x28 0x26 0x1f 0x1e 0x2c 0x28 0x2a 0x26 ,
      0x30 0x31 0x2a 0x27 0x20 0x1e 0x20 0x22 ,0x28 0x23 0x24 0x1d 0x2c 0x2b 0x28 0x2b ,
max  :0x47 0x47 0x49 0x47 0x3a 0x3f 0x40 0x3c ,0x46 0x45 0x3c 0x3b 0x4a 0x45 0x47 0x43 ,
      0x4d 0x4e 0x47 0x45 0x3f 0x3c 0x3e 0x3f ,0x44 0x3f 0x3f 0x39 0x47 0x47 0x43 0x47 ,
range:0x39 0x38 0x37 0x38 0x37 0x3a 0x38 0x39 ,0x3c 0x3d 0x3a 0x39 0x3b 0x3a 0x3a 0x3a ,
      0x3a 0x3a 0x3a 0x3b 0x3d 0x3b 0x3b 0x3a ,0x37 0x37 0x36 0x38 0x35 0x37 0x36 0x38 ,
the write training result:
DQS0:0x20, DQS1:0xe, DQS2:0x13, DQS3:0x0,
min  :0x68 0x6d 0x6f 0x6a 0x5c 0x60 0x65 0x65 0x62 ,0x52 0x4f 0x4b 0x4a 0x57 0x53 0x56 0x54 0x4d ,
      0x5e 0x5f 0x59 0x56 0x50 0x4c 0x4f 0x54 0x54 ,0x4a 0x4a 0x4a 0x43 0x51 0x50 0x4d 0x53 0x45 ,
mid  :0x85 0x89 0x8b 0x85 0x77 0x7b 0x80 0x7e 0x7d ,0x6f 0x6c 0x65 0x64 0x72 0x6e 0x71 0x6d 0x68 ,
      0x7b 0x7b 0x74 0x72 0x6c 0x68 0x6b 0x6e 0x70 ,0x67 0x66 0x63 0x5d 0x6b 0x6b 0x67 0x6e 0x60 ,
max  :0xa3 0xa5 0xa7 0xa1 0x93 0x97 0x9b 0x98 0x98 ,0x8c 0x89 0x7f 0x7f 0x8d 0x89 0x8c 0x87 0x84 ,
      0x99 0x98 0x8f 0x8e 0x88 0x84 0x87 0x88 0x8c ,0x85 0x82 0x7c 0x78 0x86 0x87 0x82 0x89 0x7b ,
range:0x3b 0x38 0x38 0x37 0x37 0x37 0x36 0x33 0x36 ,0x3a 0x3a 0x34 0x35 0x36 0x36 0x36 0x33 0x37 ,
      0x3b 0x39 0x36 0x38 0x38 0x38 0x38 0x34 0x38 ,0x3b 0x38 0x32 0x35 0x35 0x37 0x35 0x36 0x36 ,
CA Training result:
[... 144 lines cut ...]
## Booting Android Image at 0x0407f800 ...
Kernel: 0x00280000 - 0x00cc0331 (10497 KiB)
## Flattened Device Tree blob at 0x08300000
   Booting using the fdt blob at 0x08300000
   Uncompressing LZ4 Kernel Image from 0x04080000 to 0x00280000 ... with 0157b008 bytes OK
   kernel loaded at 0x00280000, end = 0x017fb008
   Using Device Tree in place at 0000000008300000, end 00000000083245be
vp0, plane_mask:0x15, primary-id:4, curser-id:0
vp1, plane_mask:0x2a, primary-id:5, curser-id:1
vp2, plane_mask:0x0, primary-id:0, curser-id:-1
WARNING: could not set reg FDT_ERR_BADOFFSET.
## reserved-memory:
  drm-logo@00000000: addr=edf00000 size=b8000
  drm-cubic-lut@00000000: addr=eff00000 size=8000
  ramoops@110000: addr=110000 size=f0000
Adding bank: 0x00200000 - 0x08400000 (size: 0x08200000)
Adding bank: 0x09400000 - 0xf0000000 (size: 0xe6c00000)
Adding bank: 0x1f0000000 - 0x200000000 (size: 0x10000000)
Total: 736.800/779.549 ms

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]

```

chrisiberg, 2024-01-15:
I have identified a solution; there is a chip on all SMDT boards referred as MCU in the device tree. This chip is connected to the I2C bus and needs to be managed as a watchdog. Initially, I attempted desoldering the VDD of this chip, which halted the 10-minute reset. However, modifying the hardware for each board was not a feasible solution for me. Instead, I used a logic analyzer to read out the init and watchdog strings, and fortunately, this approach proved successful. Now, I employ an init script along with a watchdog script that triggers every 10 seconds.

closed 2024-01-15 by chrisiberg

robinZhao, 2024-10-28:
> I have identified a solution; there is a chip on all SMDT boards referred as MCU in the device tree. This chip is connected to the I2C bus and needs to be managed as a watchdog. Initially, I [...]

can you share the watchdog script?