JeffyCN/meta-rockchip #96: rock3c

ozkansen, 2024-03-12:
Hi,
Which machine settings should I use on yocto for model "Rock3 Model C" ?

Thanks.

Ref: https://wiki.radxa.com/Rock3/3c

JeffyCN, 2024-03-12:
sorry, this layer only support a few rockchip evb boards.

please check the readme.md for more details.

chrisiberg, 2024-03-12:
You have to get the devicetree for this board, look here.
https://github.com/radxa/kernel/tree/stable-4.19-rock3/arch/arm64/boot/dts/rockchip

closed 2024-03-12 by JeffyCN

JeffyCN, 2024-03-12:
https://github.com/JeffyCN/meta-rockchip/blob/941f4102d94c8943bcbd6a12257605316fba7638/README.md#v-supporting-new-machine

chrisiberg, 2024-03-12:
@JeffyCN just wanted to give ozkansen a hint 😉

ozkansen, 2024-03-12:
thanks, I have no idea how to do it but I will try :)

JeffyCN, 2024-03-12:
one simple way is to keep your original working kernel and loader, and replace the rootfs image only(might need to install gpu lib(and maybe firmware too) and wifi/bt modules/firmware manually).