JeffyCN/meta-rockchip #100: rknn not working.

ck167493, 2024-04-01:


ck167493, 2024-04-01:
# we run rknn-toolkit2 on host
# test.py source: https://github.com/rockchip-linux/rknn-toolkit2/blob/master/rknn-toolkit2/examples/onnx/yolov5/test.py
$ python3 /rknn-toolkit2/examples/onnx/yolov5/test.py
but line 271 result in error:    line271  ret = rknn.init_runtime()
 
we have set in layer:
USB_DEBUGGING_ENABLED = "1"
IMAGE_INSTALL:append = " android-tools android-tools-adbd android-tools-conf-rockchip "
and we have follow the [rknn-toolkit2](https://github.com/airockchip/rknn-toolkit2) to move librarys in board. and use rknn_server script to enable rknn_server runtime.
adb is working. but still rknn occurs error

D RKNN: [09:55:45.576] ----------------------------------------
D RKNN: [09:55:45.576] Total Internal Memory Size: 7600KB
D RKNN: [09:55:45.576] Total Weight Memory Size: 7132.25KB
D RKNN: [09:55:45.576] ----------------------------------------
D RKNN: [09:55:45.576] <<<<<<<< end: rknn::RKNNMemStatisticsPass
I NPUTransfer: Starting NPU Transfer Client, Transfer version 2.1.0 (b5861e7@2020-11-23T11:50:36)
D NPUTransfer: Transfer spec = local:transfer_proxy
D NPUTransfer: ERROR: socket read fd = 3, n = -1: Connection reset by peer
D NPUTransfer: Transfer client closed, fd = 3
E RKNNAPI: rknn_init,  server connect fail!  ret = -9(ERROR_PIPE)!
[1;31mE[0m [1;31minit_runtime: The rknn_server on the concected device is abnormal, please start the rknn_server on the device according to:
               https://github.com/airockchip/rknpu2/blob/master/rknn_server_proxy.md[0m
[1;33mW[0m [1;33minit_runtime: ===================== WARN(7) =====================[0m
[1;31mE[0m [1;31mrknn-toolkit2 version: 1.6.0+81f21f4d[0m
[1;31mE[0m [1;31minit_runtime: Catch exception when init runtime![0m
[1;31mE[0m [1;31minit_runtime: Traceback (most recent call last):[0m
[1;31mE[0m [1;31minit_runtime:   File "rknn/api/rknn_base.py", line 2502, in rknn.api.rknn_base.RKNNBase.init_runtime[0m
[1;31mE[0m [1;31minit_runtime:   File "rknn/api/rknn_runtime.py", line 389, in rknn.api.rknn_runtime.RKNNRuntime.build_graph[0m
[1;31mE[0m [1;31minit_runtime:   File "rknn/api/rknn_log.py", line 92, in rknn.api.rknn_log.RKNNLog.e[0m
[1;31mE[0m [1;31minit_runtime: ValueError: The rknn_server on the concected device is abnormal, please start the rknn_server on the device according to:[0m
[1;31mE[0m [1;31minit_runtime:                https://github.com/airockchip/rknpu2/blob/master/rknn_server_proxy.md[0m
[1;33mW[0m [1;33mIf you can't handle this error, please try updating to the latest version of the toolkit2 and runtime from:
  https://console.zbox.filez.com/l/I00fc3 (Pwd: rknn)  Path: RKNPU2_SDK / 1.X.X / develop /
  If the error still exists in the latest version, please collect the corresponding error logs and the model,
  convert script, and input data that can reproduce the problem, and then submit an issue on:
  https://redmine.rock-chips.com (Please consult our sales or FAE for the redmine account)[0m
done
--> Export rknn model
done
--> Init runtime environment
Init runtime environment failed!

ck167493, 2024-04-01:
here is the full log
$ python3 /rknn-toolkit2/examples/onnx/yolov5/test.py
[1;33mW[0m [1;33m__init__: rknn-toolkit2 version: 1.6.0+81f21f4d[0m
[1;33mW[0m [1;33mload_onnx: It is recommended onnx opset 19, but your onnx model opset is 12![0m
[1;33mW[0m [1;33mload_onnx: Model converted from pytorch, 'opset_version' should be set 19 in torch.onnx.export for successful convert![0m
--> Config model
done
--> Loading model

Loading :   0%|                                                             | 0/125 [00:00<?, ?it/s]
Loading : 100%|████████████████████████████████████████████████| 125/125 [00:00<00:00, 32279.77it/s]
I base_optimize ...
I base_optimize done.
I 
I fold_constant ...
I fold_constant done.
I 
I correct_ops ...
I correct_ops done.
I 
I fuse_ops ...
I fuse_ops done.
I 
I sparse_weight ...
I sparse_weight done.
I 
done
--> Building model

GraphPreparing :   0%|                                                      | 0/149 [00:00<?, ?it/s]
GraphPreparing : 100%|██████████████████████████████████████████| 149/149 [00:00<00:00, 9789.64it/s]

Quantizating :   0%|                                                        | 0/149 [00:00<?, ?it/s]
Quantizating :  44%|████████████████████                          | 65/149 [00:00<00:00, 648.85it/s]
Quantizating : 100%|█████████████████████████████████████████████| 149/149 [00:00<00:00, 836.08it/s]
I 
I quant_optimizer ...
I quant_optimizer results:
I     adjust_tanh_sigmoid: ['Sigmoid_146', 'Sigmoid_148', 'Sigmoid_150']
I     adjust_relu: ['Relu_144', 'Relu_141', 'Relu_139', 'Relu_137', 'Relu_135', 'Relu_132', 'Relu_130', 'Relu_127', 'Relu_125', 'Relu_123', 'Relu_121', 'Relu_118', 'Relu_116', 'Relu_113', 'Relu_111', 'Relu_109', 'Relu_107', 'Relu_102', 'Relu_100', 'Relu_97', 'Relu_95', 'Relu_93', 'Relu_91', 'Relu_86', 'Relu_84', 'Relu_75', 'Relu_73', 'Relu_70', 'Relu_67', 'Relu_65', 'Relu_63', 'Relu_61', 'Relu_59', 'Relu_56', 'Relu_53', 'Relu_51', 'Relu_48', 'Relu_46', 'Relu_43', 'Relu_41', 'Relu_39', 'Relu_37', 'Relu_35', 'Relu_32', 'Relu_29', 'Relu_27', 'Relu_24', 'Relu_22', 'Relu_20', 'Relu_18', 'Relu_16', 'Relu_13', 'Relu_10', 'Relu_8', 'Relu_6', 'Relu_4', 'Relu_2']
I     adjust_no_change_node: ['MaxPool_81', 'MaxPool_80']
I quant_optimizer done.
I 
[1;33mW[0m [1;33mbuild: The default input dtype of 'images' is changed from 'float32' to 'int8' in rknn model for performance!
                       Please take care of this change when deploy rknn model with Runtime API![0m
[1;33mW[0m [1;33mbuild: The default output dtype of 'output' is changed from 'float32' to 'int8' in rknn model for performance!
                      Please take care of this change when deploy rknn model with Runtime API![0m
[1;33mW[0m [1;33mbuild: The default output dtype of '283' is changed from 'float32' to 'int8' in rknn model for performance!
                      Please take care of this change when deploy rknn model with Runtime API![0m
[1;33mW[0m [1;33mbuild: The default output dtype of '285' is changed from 'float32' to 'int8' in rknn model for performance!
                      Please take care of this change when deploy rknn model with Runtime API![0m
I rknn building ...
I RKNN: [09:55:45.121] compress = 0, conv_eltwise_activation_fuse = 1, global_fuse = 1, multi-core-model-mode = 7, output_optimize = 1,enable_argb_group=0 ,layout_match = 0, pipeline_fuse = 0
I RKNN: librknnc version: 1.6.0 (585b3edcf@2023-12-11T07:42:56)
D RKNN: [09:55:45.161] RKNN is invoked
W RKNN: [09:55:45.310] Model initializer tensor data is empty, name: 219
W RKNN: [09:55:45.310] Model initializer tensor data is empty, name: 238
D RKNN: [09:55:45.315] >>>>>> start: rknn::RKNNExtractCustomOpAttrs
D RKNN: [09:55:45.315] <<<<<<<< end: rknn::RKNNExtractCustomOpAttrs
D RKNN: [09:55:45.315] >>>>>> start: rknn::RKNNSetOpTargetPass
D RKNN: [09:55:45.315] <<<<<<<< end: rknn::RKNNSetOpTargetPass
D RKNN: [09:55:45.315] >>>>>> start: rknn::RKNNBindNorm
D RKNN: [09:55:45.315] <<<<<<<< end: rknn::RKNNBindNorm
D RKNN: [09:55:45.315] >>>>>> start: rknn::RKNNAddFirstConv
D RKNN: [09:55:45.315] <<<<<<<< end: rknn::RKNNAddFirstConv
D RKNN: [09:55:45.315] >>>>>> start: rknn::RKNNEliminateQATDataConvert
D RKNN: [09:55:45.316] <<<<<<<< end: rknn::RKNNEliminateQATDataConvert
D RKNN: [09:55:45.316] >>>>>> start: rknn::RKNNTileGroupConv
D RKNN: [09:55:45.316] <<<<<<<< end: rknn::RKNNTileGroupConv
D RKNN: [09:55:45.316] >>>>>> start: rknn::RKNNAddConvBias
[... 419 lines cut ...]
D NPUTransfer: Transfer spec = local:transfer_proxy
D NPUTransfer: ERROR: socket read fd = 3, n = -1: Connection reset by peer
D NPUTransfer: Transfer client closed, fd = 3
E RKNNAPI: rknn_init,  server connect fail!  ret = -9(ERROR_PIPE)!
[1;31mE[0m [1;31minit_runtime: The rknn_server on the concected device is abnormal, please start the rknn_server on the device according to:
               https://github.com/airockchip/rknpu2/blob/master/rknn_server_proxy.md[0m
[1;33mW[0m [1;33minit_runtime: ===================== WARN(7) =====================[0m
[1;31mE[0m [1;31mrknn-toolkit2 version: 1.6.0+81f21f4d[0m
[1;31mE[0m [1;31minit_runtime: Catch exception when init runtime![0m
[1;31mE[0m [1;31minit_runtime: Traceback (most recent call last):[0m
[1;31mE[0m [1;31minit_runtime:   File "rknn/api/rknn_base.py", line 2502, in rknn.api.rknn_base.RKNNBase.init_runtime[0m
[1;31mE[0m [1;31minit_runtime:   File "rknn/api/rknn_runtime.py", line 389, in rknn.api.rknn_runtime.RKNNRuntime.build_graph[0m
[... cut ...]
[1;33mW[0m [1;33mIf you can't handle this error, please try updating to the latest version of the toolkit2 and runtime from:
  https://console.zbox.filez.com/l/I00fc3 (Pwd: rknn)  Path: RKNPU2_SDK / 1.X.X / develop /
  If the error still exists in the latest version, please collect the corresponding error logs and the model,
  convert script, and input data that can reproduce the problem, and then submit an issue on:
  https://redmine.rock-chips.com (Please consult our sales or FAE for the redmine account)[0m
done
--> Export rknn model
done
--> Init runtime environment
Init runtime environment failed!

JeffyCN, 2024-04-01:
sorry, i know nothing about rknn, please ask the rknn maintainer(check the related code's committor)

closed 2024-04-01 by JeffyCN