JeffyCN/meta-rockchip #101: ADBD is not working .

ck167493, 2024-04-08:
I have set below configure in local.conf: 
  IMAGE_INSTALL:append = " android-tools-adbd android-tools-conf-rockchip "
  PREFERRED_PROVIDER_android-tools-conf = "android-tools-conf-rockchip"
  USB_DEBUGGING_ENABLED = "1"
It seems adbd can attach from pc,  but we check netstat -anp . adbd is not listening port 5037/5555.

ck167493, 2024-04-08:
result in Unable to connect Android ADB over TCP/IP

ck167493, 2024-04-09:
I have use the version provide by buildroot,and it worked! and it`s source from https://launchpad.net/ubuntu/

ck167493, 2024-04-09:
we found the patched from rockchip sdk produce by you. seems you haven`t integrate in yocto yet

ck167493, 2024-04-09:
workaround:
diff --git a/system/core/adb/adb.c b/system/core/adb/adb.c
index 10a1e0da2..6bfd68fca 100644
--- a/system/core/adb/adb.c
+++ b/system/core/adb/adb.c
@@ -1400,15 +1400,17 @@ int adb_main(int is_daemon, int server_port)
     // If one of these properties is set, also listen on that port
     // If one of the properties isn't set and we couldn't listen on usb,
     // listen on the default port.
-    property_get("service.adb.tcp.port", value, "");
-    if (!value[0]) {
-        property_get("persist.adb.tcp.port", value, "");
-    }
+    // property_get("service.adb.tcp.port", value, "");
+    // if (!value[0]) {
+    //     property_get("persist.adb.tcp.port", value, "");
+    // }
+    strcpy(value, getenv("ADB_TCP_PORT") ? : "");
     if (sscanf(value, "%d", &port) == 1 && port > 0) {
         printf("using port=%d\n", port);
-        // listen on TCP port specified by service.adb.tcp.port property
+    // listen on TCP port specified by service.adb.tcp.port property
         local_init(port);
     } else if (!usb) {
+        printf("Using USB\n");
         // listen on default port
         local_init(DEFAULT_ADB_LOCAL_TRANSPORT_PORT);
     }

and then change the env in 10-adbd-rockchip.conf
Environment="ADB_TCP_PORT=5555"

JeffyCN, 2024-04-10:
Hmm, actually i only ported a few necessary patches here, except for weston, xserver, chromium and gstreamer, since i have to maintain multiple custom versions for other BSP OS as well.

Having lots of patches in this layer would make it harder to upgrade and be compatible with multiple yocto versions.  In the Rockchip BSP yocto SDK, there's a post rootfs hook to install a prebuilt static adbd and related scripts.

closed 2024-04-10 by JeffyCN

JeffyCN, 2024-04-10:
[yocto_usb.tar.gz](https://github.com/JeffyCN/meta-rockchip/files/14926538/yocto_usb.tar.gz)

https://github.com/JeffyCN/rockchip_mirrors/tree/buildroot/package/android-adbd
this is based on:
https://salsa.debian.org/android-tools-team/android-platform-system-core  8.1.0+r23-8