JeffyCN/meta-rockchip #102: How to login into system?

DhimanAirao, 2024-05-02:
I built the image for rk3568 evb board. When I flashed the Image I was asked with login. But there is no mention of username and password. So How should I login into system?

chrisiberg, 2024-05-02:
The default image (core-image-base) that is generated by Yocto defaults to “root” without any password.

DhimanAirao, 2024-05-02:
Hii, Thanks for fast reply. But Unfortunately I tried root as password for "core-image-base" but it's showing Login incorrect. Also I tired "core-image-minimal" and it's not accepting root as user login.

Anyways I enabled "debug-tweaks" currently for testing purpose and with this I was able to login into system. I would request you to test this login issue at your end too.

JeffyCN, 2024-05-10:
this part of logic in unchanged, so maybe you can try to search the yocto official documents(poky/documentation/ref-manual/features.rst).

currently you can:
1/ just use root user in serial console
2/ use adb tool(no login required)
3/ if you are using ssh, need to enable debug-tweaks or allow-root-login

closed 2024-05-10 by JeffyCN