JeffyCN/meta-rockchip #103: fatal error: 'media/gpu/v4l2/v4l2_stubs.h' file not found

lilulu2018, 2024-05-09:
We used chromium_120.0.6099 patch to build chrome but received error as the title mentioned , more logs like this
```
clang -plugin-arg-find-bad-constructs -Xclang raw-ref-template-as-trivial-member -Xclang -plugin-arg-find-bad-constructs -Xclang check-stack-allocated -Xclang -plugin-arg-find-bad-constructs -Xclang check-raw-ptr-to-stack-allocated -Xclang -plugin-arg-find-bad-constructs -Xclang raw-ptr-fix-crbug-1449812 -Xclang -plugin-arg-find-bad-constructs -Xclang check-ipc -Wheader-hygiene -Wstring-conversion -Wtautological-overlap-compare -isystem../../build/linux/debian_bullseye_arm64-sysroot/usr/include/glib-2.0 -isystem../../build/linux/debian_bullseye_arm64-sysroot/usr/lib/aarch64-linux-gnu/glib-2.0/include -Wno-redundant-parens -Wno-redundant-parens -DPROTOBUF_ALLOW_DEPRECATED=1 -isystem../../build/linux/debian_bullseye_arm64-sysroot/usr/include/nss -isystem../../build/linux/debian_bullseye_arm64-sysroot/usr/include/nspr -std=c++20 -Wno-trigraphs -gsimple-template-names -fno-exceptions -fno-rtti -nostdinc++ -isystem../../third_party/libc++/src/include -isystem../../third_party/libc++abi/src/include --sysroot=../../build/linux/debian_bullseye_arm64-sysroot -fvisibility-inlines-hidden -c ../../media/capture/video/linux/v4l2_capture_device_impl.cc -o obj/media/capture/capture_lib/v4l2_capture_device_impl.o
../../media/capture/video/linux/v4l2_capture_device_impl.cc:14:10: fatal error: 'media/gpu/v4l2/v4l2_stubs.h' file not found
   14 | #include "media/gpu/v4l2/v4l2_stubs.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
[27/1152] CXX obj/media/gpu/command_buffer_helper/command_buffer_helper.o
ninja: build stopped: subcommand failed.
```
Any ideas ?

JeffyCN, 2024-05-09:
the rule of generating it is in the patch 0003*

lilulu2018, 2024-05-10:
@JeffyCN all patches are applied including patch 0006 not applicable so I manually modified files affected. use_v4l2_codec = true and use_vaapi = false added to args.gn, but still getting this error. Here is what exactly I have done:
system ubuntu 22.04
1.  git clone https://chromium.googlesource.com/chromium/tools/depot_tools.git
2. export PATH="${HOME}/depot_tools:$PATH"
3. mkdir ~/chromium && cd ~/chromium
4. fetch --nohooks chromium
5. cd src
6. git checkout -b v120 tags/120.0.6099.291
7. gclient sync
8. ./build/install-build-deps.sh
9. ./build/linux/sysroot_scripts/install-sysroot.py --arch=arm64
10. apply patches from dynamic-layers/recipes-browser/chromium/chromium_120.0.6099
11. gn gen out/Arm64
12. edit args.gn as below 
```
# Set build arguments here. See `gn help buildargs`.
target_cpu = "arm64"
enable_nacl = false
symbol_level = 0
blink_symbol_level = 0
v8_symbol_level = 0
is_official_build = true
chrome_pgo_phase = 0
use_v4l2_codec = true
use_vaapi = false
```
Then I run ```autoninja -C out/Arm64``` chrome from src directory and get this error. Is there anything I've done wrong or missing anything ？

JeffyCN, 2024-05-10:
check:
https://github.com/JeffyCN/meta-rockchip/blob/master/dynamic-layers/recipes-browser/chromium/chromium-%25.bbappend#L9
use_v4lplugin=true

lilulu2018, 2024-05-10:
added ``` use_v4lplugin = true ```resolved my problem

closed 2024-05-10 by lilulu2018